raspberrypi/linux #3443: [arm64] linux-5.5.y cma_alloc regression (possible fix included)

asavah, 2020-02-01:
**Describe the bug**
pi4b 1.1 4G(<-important)  arm64 kernel 5.5.y fails to initialize brcm-pcie
It worked fine on 5.4.y.
See details below
```
[    0.917301] cma: cma_alloc: alloc failed, req-size: 8192 pages, ret: -12
[    0.917374] ------------[ cut here ]------------
[    0.917404] WARNING: CPU: 0 PID: 1 at mm/page_alloc.c:4713 __alloc_pages_nodemask+0x2c8/0x320
[    0.917418] Modules linked in:
[    0.917437] CPU: 0 PID: 1 Comm: swapper/0 Not tainted 5.5.0-k5 #1
[    0.917450] Hardware name: Raspberry Pi 4 Model B Rev 1.1 (DT)
[    0.917465] pstate: 20000005 (nzCv daif -PAN -UAO)
[    0.917482] pc : __alloc_pages_nodemask+0x2c8/0x320
[    0.917500] lr : __dma_direct_alloc_pages+0x1a0/0x290
[    0.917512] sp : ffffffc01002b910
[    0.917523] x29: ffffffc01002b910 x28: ffffffc0111f0000
[    0.917539] x27: 000000000000000d x26: ffffffc010d74638
[    0.917554] x25: 000000000000000d x24: 0000000001ffffff
[    0.917568] x23: 00000000ffffffff x22: ffffff80f6d80000
[    0.917582] x21: ffffff80f6fe5010 x20: 0000000000001fff
[    0.917595] x19: 0000000000000cc4 x18: ffffffffffffffff
[    0.917609] x17: 00000000d9ad352c x16: 000000009e10a494
[    0.917622] x15: ffffffc01002b658 x14: 0720072007200720
[    0.917635] x13: 0720072007200720 x12: 0720072007200720
[    0.917648] x11: 0720072007200720 x10: 0720072007200720
[    0.917661] x9 : ffffffc010128630 x8 : 0000000000000085
[    0.917675] x7 : 0000000000000000 x6 : 0000000000000000
[    0.917688] x5 : ffffffc010d89000 x4 : 0000000000000001
[    0.917700] x3 : 0000000000000000 x2 : 0000000000000000
[    0.917713] x1 : 0000000000000000 x0 : 0000000000000cc4
[    0.917727] Call trace:
[    0.917743]  __alloc_pages_nodemask+0x2c8/0x320
[    0.917758]  __dma_direct_alloc_pages+0x1a0/0x290
[    0.917773]  dma_direct_alloc_pages+0x5c/0x1d0
[    0.917788]  dma_direct_alloc+0x14/0x20
[    0.917802]  dma_alloc_attrs+0x84/0xf0
[    0.917816]  brcm_pcie_bounce_init+0xb8/0x190
[    0.917829]  brcm_pcie_probe+0x2ac/0x960
[    0.917846]  platform_drv_probe+0x58/0xb0
[    0.917860]  really_probe+0xe0/0x300
[    0.917873]  driver_probe_device+0x5c/0xf0
[    0.917887]  device_driver_attach+0x74/0x80
[    0.917901]  __driver_attach+0x5c/0xe0
[    0.917914]  bus_for_each_dev+0x74/0xd0
[    0.917927]  driver_attach+0x28/0x30
[    0.917940]  bus_add_driver+0x14c/0x1f0
[    0.917953]  driver_register+0x6c/0x120
[    0.917968]  __platform_driver_register+0x50/0x60
[    0.917984]  brcm_pcie_driver_init+0x20/0x28
[    0.917998]  do_one_initcall+0x4c/0x230
[    0.918014]  kernel_init_freeable+0x1f8/0x268
[    0.918030]  kernel_init+0x18/0x108
[    0.918044]  ret_from_fork+0x10/0x24
[    0.918059] ---[ end trace 2a2538234edd38bb ]---
[    0.918079] brcm-pcie fd500000.pcie: dmabounce: could not allocate 33554432 byte DMA pool
[    0.918102] brcm-pcie fd500000.pcie: could not init bounce buffers: -12
[    0.918142] brcm-pcie: probe of fd500000.pcie failed with error -12
```

**To reproduce**
Boot pi4 4GB ram with arm64 linux-5.5.y

**Expected behaviour**
brcm-pcie and usb work

**Actual behaviour**
See backtrace above,pcie fails to initialize usb controller missing.

**System**
* Which model of Raspberry Pi? e.g. Pi3B+, PiZeroW
pi4 4GB
* Which OS and version (`cat /etc/rpi-issue`)?
own pet project
* Which firmware version (`vcgencmd version`)?
Jan 30 2020 13:51:58
Copyright (c) 2012 Broadcom
version 8f792e011c8a0ea3c5d47e7cc10172cc10b93c09 (clean) (release) (start_x)
* Which kernel version (`uname -a`)?
Linux rpi4 5.5.0-k6 #1 SMP Sat Feb 1 02:46:34 EET 2020 aarch64 GNU/Linux

**Logs**
Will add full dmesgs on request if needed, see detailed explanation below.

**Additional context**
So here starts the fun
adding `total_mem=3072` to config.txt fixes the issue, pcie gets initialized, usb works.
But trading 1G for USB is not fair.

`cma: cma_alloc: alloc failed, req-size: 8192 pages, ret: -12` thats ENOMEM right?
but how?

lets check cmdline
```
Kernel command line: coherent_pool=1M 8250.nr_uarts=1 cma=64M bcm2708_fb.fbwidth=1920 bcm2708_fb.fbheight=1280 bcm2708_fb.fbswap=1 smsc95xx.macaddr=SNIP vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  dwc_otg.lpm_enable=0 console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 fsck.repair=yes elevator=noop rootwait
```
cma=64M, all good

but lets check beginning of dmesg
```
[    0.000000] Reserved memory: created CMA memory pool at 0x000000003a000000, size 32 MiB
[    0.000000] Memory: 3869148K/4063232K available (9022K kernel code, 1018K rwdata, 3228K rodata, 1024K init, 640K bss, 161316K reserved, 32768K cma-reserved)
```
So it should be 64M, but apparently cmdline `cma=` is ignored and value from dts is used instead
https://github.com/raspberrypi/linux/blob/rpi-5.5.y/arch/arm/boot/dts/bcm2711.dtsi#L25

Quick patch:
```
diff --git a/arch/arm/boot/dts/bcm2711.dtsi b/arch/arm/boot/dts/bcm2711.dtsi
index e2f6ffb00aa9..e860906b9d35 100644
--- a/arch/arm/boot/dts/bcm2711.dtsi
+++ b/arch/arm/boot/dts/bcm2711.dtsi
@@ -24,7 +24,7 @@ reserved-memory {
                 */
                linux,cma {
                        compatible = "shared-dma-pool";
-                       size = <0x2000000>; /* 32MB */
+                       size = <0x4000000>; /* 64MB */
                        alloc-ranges = <0x0 0x00000000 0x40000000>;
                        reusable;
                        linux,cma-default;
```

Lets see dmesg now:
```
[    0.000000] Reserved memory: created CMA memory pool at 0x0000000038000000, size 64 MiB
SNIP
[    0.945569] brcm-pcie fd500000.pcie: dmabounce: initialised - 32768 kB, threshold 0x00000000c0000000 (0x00000001c0000000)
[    0.945619] brcm-pcie fd500000.pcie: could not get clock
[    0.945703] brcm-pcie fd500000.pcie: host bridge /scb/pcie@7d500000 ranges:
[    0.945767] brcm-pcie fd500000.pcie:   MEM 0x600000000..0x603ffffff -> 0xf8000000
[    0.987189] brcm-pcie fd500000.pcie: link up, 5.0 Gbps x1 (!SSC)
[    0.987532] brcm-pcie fd500000.pcie: PCI host bridge to bus 0000:00
[    0.987558] pci_bus 0000:00: root bus resource [bus 00-01]
[    0.987582] pci_bus 0000:00: root bus resource [mem 0x600000000-0x603ffffff] (bus address [0xf8000000-0xfbffffff])
[    0.987640] pci 0000:00:00.0: [14e4:2711] type 01 class 0x060400
[    0.987832] pci 0000:00:00.0: PME# supported from D0 D3hot
[    0.991320] pci 0000:00:00.0: bridge configuration invalid ([bus 00-00]), reconfiguring
[    0.991541] pci 0000:01:00.0: [1106:3483] type 00 class 0x0c0330
[    0.991676] pci 0000:01:00.0: reg 0x10: [mem 0x00000000-0x00000fff 64bit]
[    0.992048] pci 0000:01:00.0: PME# supported from D0 D3cold
[    0.998378] pci_bus 0000:01: busn_res: [bus 01] end is updated to 01
[    0.998425] pci 0000:00:00.0: BAR 8: assigned [mem 0x600000000-0x6000fffff]
[    0.998453] pci 0000:01:00.0: BAR 0: assigned [mem 0x600000000-0x600000fff 64bit]
[    0.998492] pci 0000:00:00.0: PCI bridge to [bus 01]
[    0.998516] pci 0000:00:00.0:   bridge window [mem 0x600000000-0x6000fffff]
[    0.998592] pci 0000:00:00.0: enabling device (0000 -> 0002)
[    0.998624] pci 0000:01:00.0: enabling device (0000 -> 0002)
```
```
root@rpi4:~# lspci
00:00.0 PCI bridge: Broadcom Inc. and subsidiaries Device 2711 (rev 10)
01:00.0 USB controller: VIA Technologies, Inc. VL805 USB 3.0 Host Controller (rev 01)
root@rpi4:~# lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
root@rpi4:~#free
              total        used        free      shared  buff/cache   available
Mem:        3902940      222764     3535636         7
```

Everything seems to be ok.

So it looks like that the actual problem is that cma= from cmdline is ignored and value from bcm2711.dtsi is used.

Hope I didn't make wrong conclusions and this helps hunt a bug or two )

If needed I can provide full dmesg outputs I kept while trying to get to the bottom of this.

pelwell, 2020-02-01:
Sorry about that - I chose the wrong Pi 4 to test on.

Can you try this patch? It ought to re-activate the old command line parameter until I patch the various overlays to use the DT property instead (or as well):
```diff
--- a/arch/arm/boot/dts/bcm2711-rpi-4-b.dts
+++ b/arch/arm/boot/dts/bcm2711-rpi-4-b.dts
@@ -167,6 +167,10 @@
        };
 
        /delete-node/ wifi-pwrseq;
+
+       reserved-memory {
+               /delete-node/ linux,cma;
+       };
 };
 
 &mmcnr {
```

asavah, 2020-02-01:
@pelwell Thanks, with your patch and without mine everything works nicely.

```
[    0.000000] cma: Reserved 64 MiB at 0x0000000038000000
[    0.000000] Memory: 3836380K/4063232K available (9022K kernel code, 1018K rwdata, 3228K rodata, 1024K init, 640K bss, 161316K reserved, 65536K cma-reserved)
[    0.944129] brcm-pcie fd500000.pcie: dmabounce: initialised - 32768 kB, threshold 0x00000000c0000000 (0x00000001c0000000)
[    0.944179] brcm-pcie fd500000.pcie: could not get clock
[    0.944264] brcm-pcie fd500000.pcie: host bridge /scb/pcie@7d500000 ranges:
[    0.944329] brcm-pcie fd500000.pcie:   MEM 0x600000000..0x603ffffff -> 0xf8000000
[    0.985531] brcm-pcie fd500000.pcie: link up, 5.0 Gbps x1 (!SSC)
[    0.985869] brcm-pcie fd500000.pcie: PCI host bridge to bus 0000:00
[    0.985895] pci_bus 0000:00: root bus resource [bus 00-01]
[    0.985918] pci_bus 0000:00: root bus resource [mem 0x600000000-0x603ffffff] (bus address [0xf8000000-0xfbffffff])
[    0.985977] pci 0000:00:00.0: [14e4:2711] type 01 class 0x060400
[    0.986170] pci 0000:00:00.0: PME# supported from D0 D3hot
[    0.989651] pci 0000:00:00.0: bridge configuration invalid ([bus 00-00]), reconfiguring
[    0.989871] pci 0000:01:00.0: [1106:3483] type 00 class 0x0c0330
[    0.990008] pci 0000:01:00.0: reg 0x10: [mem 0x00000000-0x00000fff 64bit]
[    0.990348] pci 0000:01:00.0: PME# supported from D0 D3cold
[    0.996736] pci_bus 0000:01: busn_res: [bus 01] end is updated to 01
[    0.996783] pci 0000:00:00.0: BAR 8: assigned [mem 0x600000000-0x6000fffff]
[    0.996810] pci 0000:01:00.0: BAR 0: assigned [mem 0x600000000-0x600000fff 64bit]
[    0.996848] pci 0000:00:00.0: PCI bridge to [bus 01]
[    0.996872] pci 0000:00:00.0:   bridge window [mem 0x600000000-0x6000fffff]
[    0.996948] pci 0000:00:00.0: enabling device (0000 -> 0002)
[    0.996979] pci 0000:01:00.0: enabling device (0000 -> 0002)
```

pelwell, 2020-02-01:
Thanks for the early report and quick feedback. I'll push the patch when I'm next near a keyboard.

asavah, 2020-02-01:
Thanks, feel free to close once the patch is in.

closed 2020-02-01 by pelwell

vianpl, 2020-02-03:
> Sorry about that - I chose the wrong Pi 4 to test on.
>
> Can you try this patch? It ought to re-activate the old command line parameter until I patch the various overlays to use the DT property [...]
For the record, this[1] will probably land in v5.6, which fixes the cmdline/dt CMA config conflict mentioned above.

[1] https://lkml.org/lkml/2020/1/10/674

pelwell, 2020-02-03:
Thanks, @vianpl - I should have remembered that patch.