raspberrypi/linux #3444: ‘IPT_ECN_OP_SET_ECE’ undeclared when building kernel

yangjunsss, 2020-02-01:
**Describe the bug**
Can't compile the kernel via guide https://www.raspberrypi.org/documentation/linux/kernel/building.md 

**To reproduce**
```sh
git clone https://github.com/raspberrypi/tools ~/tools

echo PATH=\$PATH:~/tools/arm-bcm2708/gcc-linaro-arm-linux-gnueabihf-raspbian-x64/bin >> ~/.bashrc
source ~/.bashrc

sudo apt install git bc bison flex libssl-dev make

git clone --depth=1 https://github.com/raspberrypi/linux

cd linux
KERNEL=kernel7l make ARCH=arm CROSS_COMPILE=arm-linux-gnueabihf- bcm2711_defconfig

make ARCH=arm CROSS_COMPILE=arm-linux-gnueabihf- zImage modules dtbs
```

**Expected behaviour**
Build successfully.

**Actual behaviour**
Get error below
```sh
  CC [M]  net/ipv4/netfilter/ipt_ECN.o
net/ipv4/netfilter/ipt_ECN.c:29:46: warning: ‘struct ipt_ECN_info’ declared inside parameter list [enabled by default]
 set_ect_ip(struct sk_buff *skb, const struct ipt_ECN_info *einfo)
                                              ^
net/ipv4/netfilter/ipt_ECN.c:29:46: warning: its scope is only this definition or declaration, which is probably not what you want [enabled by default]
net/ipv4/netfilter/ipt_ECN.c: In function ‘set_ect_ip’:
net/ipv4/netfilter/ipt_ECN.c:33:44: error: dereferencing pointer to incomplete type
  if ((iph->tos & IPT_ECN_IP_MASK) != (einfo->ip_ect & IPT_ECN_IP_MASK)) {
                                            ^
net/ipv4/netfilter/ipt_ECN.c:40:21: error: dereferencing pointer to incomplete type
   iph->tos |= (einfo->ip_ect & IPT_ECN_IP_MASK);
                     ^
net/ipv4/netfilter/ipt_ECN.c: At top level:
net/ipv4/netfilter/ipt_ECN.c:48:47: warning: ‘struct ipt_ECN_info’ declared inside parameter list [enabled by default]
 set_ect_tcp(struct sk_buff *skb, const struct ipt_ECN_info *einfo)
                                               ^
net/ipv4/netfilter/ipt_ECN.c: In function ‘set_ect_tcp’:
net/ipv4/netfilter/ipt_ECN.c:58:14: error: dereferencing pointer to incomplete type
  if ((!(einfo->operation & IPT_ECN_OP_SET_ECE) ||
              ^
net/ipv4/netfilter/ipt_ECN.c:58:28: error: ‘IPT_ECN_OP_SET_ECE’ undeclared (first use in this function)
  if ((!(einfo->operation & IPT_ECN_OP_SET_ECE) ||
```

**System**
root@yjvm:~/linux# uname -a
Linux yjvm 4.15.0-76-generic #86-Ubuntu SMP Fri Jan 17 17:24:28 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux

**Additional context**
Solution:
Copy the header from your host os and `make` again
```sh
root@yjvm:~/linux# cp /usr/include/linux/netfilter_ipv4/ipt_ECN.h include/uapi/linux/netfilter_ipv4/ipt_ECN.h
```

ED6E0F17, 2020-02-03:
This might happen if your virtual machine was running on a Windows file system that could confuse "ipt_ECN.h" with "ipt_ecn.h"

pelwell, 2020-02-03:
I agree with @ED6E0F17 - it looks like a problem caused by a lack of case-sensitivity.

What does `mount` report?

JamesH65, 2020-02-13:
@yangjunsss Have you been able to try the suggestion above?

yangjunsss, 2020-02-16:
Yes, u are right. I'm working on my MacBook. Already fixed it via `git reset --hard HEAD~1`. Thanks.

closed 2020-02-16 by yangjunsss