raspberrypi/linux #3445: Udev rule cannot send udp network pakets

Blackbox-git, 2020-02-01:
i have a problem with a udev rule. I use Raspbian Buster with Kernel 4.19.93-v7+ #1290. At first, if i execute the script manually it works. The udev rule also execute the script, but abort when to send a upd paket. (I will send a upd paket to localhost which a other programm register) Here is the working Script with debugging-Code:

```
#!/usr/bin/python
# -*- coding: utf-8 -*-

# Import der 
notwendigen Bibliotheken
import os
import sys
import getpass
import socket

f = open('/home/pi/scripts/keypad/client.txt','a')
f.write('Starting Script from Udev-Rule as user: ')
f.write(str(getpass.getuser()))
#f.write('\n')
f.close()

try:
    # Serververbindung öffnen, an Port binden und auf Lauschen stellen
    s = socket.socket(socket.AF_INET, socket.SOCK_DGRAM)
    s.sendto('1', ('127.0.0.1', 50000))
except:
    f=open('/home/pi/scripts/keypad/client.txt','a')
    f.write('\n')
    f.write(str(sys.exc_info()[0]))
    f.write(str(sys.exc_info()[1]))
    f.write(str(sys.exc_info()[2]))
    f.write('\n')
    f.close()

f = open('/home/pi/scripts/keypad/client.txt','a')
f.write('End of Script from Udev-Rule\n')
f.write('\n')
f.write('\n')
f.close()
```

If i but my device in the Rasbperry, the script will be excecuted and abort. Here you cann see what i am logging:

```
Starting Script from Udev-Rule as user: root
<class 'socket.error'>
[Errno 1] Operation not permitted
<traceback object at 0x76c36260>
End of Script from Udev-Rule
```

If i do this manually the script works perfect. I also tried to replace the code with a simply bash command "echo "1" | nc -w1 -u 127.0.0.1 50000" . But the result is the same. Manually works if udev execute its not working.

Has anyone an idea?

PS: If I use the Image from Homepage (September 2019) it works.

asavah, 2020-02-01:
This has nothing to do with the kernel, but with systemd/udev itself.
Don't execute such scripts directly from udev rule, udev has timeout limits too.
Write a oneshot service for systemd and trigger it from the udev rule.

Blackbox-git, 2020-02-02:
I know the timeout Limit. Therefore i wrote this Script. I should only send a udp network paket to the main program.

I dosen't know a netter way

pelwell, 2020-02-07:
I agree with @asavah - this is not a kernel problem.

closed 2020-02-07 by pelwell