raspberrypi/linux #3447: Correct g-rx-fifo-size in dwc2-overlay.dts

pavhofman, 2020-02-03:
Hi,

In co-operation with dwc2 module kernel maintainer we identified a problem in current setting of g-rx-fifo-size in RPI's dwc2-overlay.dts:

When testing the g_audio USB gadget I tried to reach maximum high-speed isochronous packet size allowed by the USB2 specs - 1024 bytes per packet. No packets over 900 bytes got through from USB host to the RPi gadget. 

The cause was identified in the current setting g-rx-fifo-size = <256>. 256 is not sufficient for the full packet size since additional information is stored in the RX FIFO, according to DWC2 specs. A value of 304 works OK for all isochronous packet sizes up to 1024.

The requested change in https://github.com/raspberrypi/linux/blob/4dc8a4f35e45ba669e39d56ac9e0114eb56057f1/arch/arm/boot/dts/overlays/dwc2-overlay.dts#L15

g-rx-fifo-size = <304>;

I cannot quote from the NDA docs screenshot by the Synopsys dwc2 developer, but the wording explaining the RX FIFO requirements is identical e.g. to https://www.silabs.com/documents/public/reference-manuals/ezr32hg-rm.pdf section 15.4.7.1.1.1 page 274.

Thanks a lot for including this change in the RPi kernel.

pelwell, 2020-02-03:
Can you explain how the number 304 was arrived at?

pavhofman, 2020-02-03:
The value 304 was recommended by the Synopsys team of Minas Harutyunyan, I did not make the calculation. The value is number of 32bits (i.e. x 4 in bytes) - look at section 15.6.15 USB_GRXFSIZ - Receive FIFO Size Register (page 306) of that manual, quoting

_This value is in terms of 32-bit words. Minimum value is 16. Maximum value is 512._

Clearly 256 x 4 = 1024 bytes of the current setup cannot accommodate for high-speed isochronous max packet size of 1024 bytes - quoting from that manual:

_* Status information is written to the FIFO along with each received packet. Therefore, a minimum space of (Largest Packet Size / 4) + 1 must be allotted to receive packets. If a high-bandwidth
endpoint is enabled, or multiple isochronous endpoints are enabled, then at least two (Largest Packet Size / 4) + 1 spaces must be allotted to receive back-to-back packets. Typically, two (Largest Packet Size / 4) + 1 spaces are recommended so that when the previous packet is being transferred to AHB, the USB can receive the subsequent packet. If AHB latency is high, you must allocate enough space to receive multiple packets. This is critical to prevent dropping any isochronous packets.
• Along with each endpoint's last packet, transfer complete status information is also pushed to the
FIFO. Typically, one location for each OUT endpoint is recommended._

Plus the original manual copied by Minas specified extra space requirements:

* An additional location for EPDisable for each endpoint is also required
* Typically, two locations for each OUT endpoint are recommended

I can ask Minas for exact details if it's necessary.

Thanks.

pelwell, 2020-02-03:
A change like this needs a commit message explaining how the new value was calculated, in case it needs to be changed again in the future - like showing your workings in an exam - so "exact details" would be appreciated.

ED6E0F17, 2020-02-03:
I use a config setting of:
````
dtparam=g-rx-fifo-size="768"
````

I think that you can go as high as 900 before you run out of unused fifo space (downstream), but if Minas says 304, that is what you use.

Upstream are still using 256, so they might want to change their settings.

pavhofman, 2020-02-04:
Details by Minas - each point is documentation quote followed by explanation for DWC2 core in the Broadcom SoC:

_* RAM for SETUP Packets: 4 * n + 6 locations in Scatter/Gather DMA mode and 5 * n+8 locations in
Slave and Buffer DMA mode must be reserved in the RxFIFO to receive up to n SETUP packets on
control endpoints, where n is the number of control endpoints the device controller supports. The
controller does not use these locations, which are Reserved for SETUP packets, to write any other
data._

RPi: 5*n+8. The Broadcom core has 1 control EP (n=1), so 13 locations

_* One location for Global OUT NAK_

RPi: 1 location

_* Status information is written to the FIFO along with each received packet. Therefore, a minimum
space of (Largest Packet Size / 4) + 1 must be allotted to receive packets. If a high-bandwidth
endpoint is enabled, or multiple isochronous endpoints are enabled, then at least two (Largest
Packet Size / 4) + 1 spaces must be allotted to receive back-to-back packets. Typically, two
(Largest Packet Size / 4) + 1 spaces are recommended so that when the previous packet is being
transferred to AHB, the USB can receive the subsequent packet. If AHB latency is high, you must
allocate enough space to receive multiple packets. This is critical to prevent dropping of any
isochronous packets._

RPi: (1024/4) +1 = 257 locations. For MC >1: 2 * 257 = 514 locations

_* Along with last packet of each endpoint, transfer complete status information is also pushed in to
the FIFO._

RPi: The core should have 5 IN and 5 OUT EP's. 1 location for each EP - 10 locations
It's for the case when all EPs are simultaneously completing transfers.

_* An additional location for EPDisable status for each endpoint is also required._
RPi: 1 location for each EP - 10 EP's  - 10 locations
It's for case if EP simultaneously disabled.

_* Typically, two locations for each OUT endpoint is recommended._

RPi: 5*2 = 10 locations

Total:  13 + 1 + 257 + 10 +10 + 10 = 301 locations

Ideally 301 + 257 (for MC>1) = 558 locations.

Location is dword = 4 bytes.
MC is number of packets per frame.

My value of 304 is slightly larger than required. I can test with 301. Better would be 558.

pelwell, 2020-02-04:
Thank you for the detailed explanation - we can link to that posting from the final commit.

@P33M Would you rather go for 301 or 558?

P33M, 2020-02-04:
There's no reason why the ARM11 on bcm2835 shouldn't be able to process inbound data at a rate where you need an isoc MC > 1 (roughly 130mbit/s), so I'd say 558.

pavhofman, 2020-02-04:
Fantastic. Guys, thanks a lot to all.

pelwell, 2020-02-04:
Thank you everyone - the overlay has been updated.

lategoodbye, 2020-02-04:
@pavhofman Would be nice if you can prepare a patch for the upstream files bcm283x-rpi-usb-otg.dtsi and bcm283x-rpi-usb-peripheral.dtsi .

pavhofman, 2020-02-04:
No problem, the same patch incl. description would make sense. Just the patch wording is Phill's, if he is ok with using his text...

We discussed with Minas in private mail, but the result is above.

pelwell, 2020-02-04:
I'm happy for you to take my commit message wording with or without my Signed-off-by: line.

pelwell, 2020-02-07:
Please close this issue if you think the matter is resolved.

pavhofman, 2020-02-07:
I will prepare the patch for upstream. Please which mailing list to send it? Thanks.

closed 2020-02-07 by pavhofman

pelwell, 2020-02-07:
The get_maintainer script says:
```
Rob Herring <robh+dt@kernel.org> (maintainer:OPEN FIRMWARE AND FLATTENED DEVICE TREE BINDINGS)
Mark Rutland <mark.rutland@arm.com> (maintainer:OPEN FIRMWARE AND FLATTENED DEVICE TREE BINDINGS)
Florian Fainelli <f.fainelli@gmail.com> (maintainer:BROADCOM BCM281XX/BCM11XXX/BCM216XX ARM ARCHITE...)
Ray Jui <rjui@broadcom.com> (maintainer:BROADCOM BCM281XX/BCM11XXX/BCM216XX ARM ARCHITE...)
Scott Branden <sbranden@broadcom.com> (maintainer:BROADCOM BCM281XX/BCM11XXX/BCM216XX ARM ARCHITE...)
bcm-kernel-feedback-list@broadcom.com (maintainer:BROADCOM BCM281XX/BCM11XXX/BCM216XX ARM ARCHITE...)
devicetree@vger.kernel.org (open list:OPEN FIRMWARE AND FLATTENED DEVICE TREE BINDINGS)
linux-kernel@vger.kernel.org (open list)
```
However, this is more of an issue with the DWC2 interface and driver, of which the MAINTAINERS file says:
```
DESIGNWARE USB2 DRD IP DRIVER
M:	Minas Harutyunyan <hminas@synopsys.com>
L:	linux-usb@vger.kernel.org
T:	git git://git.kernel.org/pub/scm/linux/kernel/git/balbi/usb.git
S:	Maintained
F:	drivers/usb/dwc2/
```
I suggest sending your report to:
```
Minas Harutyunyan <hminas@synopsys.com
Rob Herring <robh+dt@kernel.org>
Nicolas Saenz Julienne <nsaenzjulienne@suse.de>
linux-usb@vger.kernel.org
devicetree@vger.kernel.org
linux-rpi-kernel@lists.infradead.org
```

pavhofman, 2020-02-07:
:-) I knew I better ask :-) Thanks.

lategoodbye, 2020-02-07:
Yes, get_maintainer.pl isn't smart. @pelwell suggestion looks good.

Just a note: infradead mailing lists are moderated, so they require a subscription before submission.