raspberrypi/linux #3452: AX88179 USB 3.0 Gigabit Ethernet driver appends a two-byte trailer to every packet

fj604, 2020-02-04:
**Describe the bug**
AX88179 USB 3.0 Gigabit Ethernet driver appends a two-byte trailer to every packet.

**To reproduce**
Connect AX88179-based Gigabit Ethernet adapter like https://www.amazon.co.uk/dp/B00M77HMU0, perform packet capture with tcpdump, open pcap in Wireshark, observe a two-byte "VSS Monitoring Ethernet trailer" added to every packet

**Expected behaviour**
No trailer should be added to packets

**Actual behaviour**
A two-byte trailer is added to every packet.

**System**

* Which model of Raspberry Pi? e.g. Pi3B+, PiZeroW
Pi 4 4GB

* Which OS and version (`cat /etc/rpi-issue`)?
`Raspberry Pi reference 2019-09-26
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 80d486687ea77d31fc3fc13cf3a2f8b464e129be, stage5`

* Which firmware version (`vcgencmd version`)?
`Jan 10 2020 16:52:04
Copyright (c) 2012 Broadcom
version fdb5c37e330e7cb3027ac4fcc5b1cd5f244b351f (clean) (release) (start)`

* Which kernel version (`uname -a`)?
`4.19.93-v7l+ #1290 SMP Fri Jan 10 16:45:11 GMT 2020 armv7l GNU/Linux`

**Logs**
If applicable, add the relevant output from `dmesg` or similar.

**Additional context**
 Issue is fixed with the manufacturer-supplied driver: 
https://www.asix.com.tw/products.php?op=pItemdetail&PItemID=131;71;112

6by9, 2020-02-04:
Context https://www.raspberrypi.org/forums/viewtopic.php?f=107&t=264003

6by9, 2020-02-04:
I've done a quick compare between and the vendor supplied driver is vastly different from the mainline one. It may be possible to find a trivial difference between the two, otherwise this ought to be punted upstream.

fj604, 2020-02-05:
The driver in the latest mainline kernel still has this issue:
` 5.5.0-050500-generic #202001262030 SMP Mon Jan 27 01:33:36 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux`
I'll report this issue upstream.

6by9, 2020-02-05:
Comparing between the Asix and mainline drivers, ax88179_rx_fixup has three clauses that are
```
if (NET_IP_ALIGN == 0)
   skb_pull(skb, 2);
```
Seeing as you've built the vendor driver, could you add a lines
```
#if NET_IP_ALIGN == 2
	#warning IP align 2
#else
	#warning IP align != 2
#endif
```
somewhere after all the includes, and recompile. During compile it'll report the value as 2 or != 2.

The mainline kernel has NET_IP_ALIGN defined as 2, so I'm wondering if the vendor kernel matches.
As the vendor supplied driver only does those skb_pulls if NET_IP_ALIGN == 0, it implies that it is incorrect for the mainline driver to be doing them unconditionally. I'm not the maintainer of the driver though, and I'm just looking for odd things. (Networking skbs has always confused me).

fj604, 2020-02-05:
Result:
`pi@raspberrypi:~/AX88179_178A_LINUX_DRIVER_v1.19.0_SOURCE $ sudo make
make -C /lib/modules/4.19.93-v7l+/build SUBDIRS=/home/pi/AX88179_178A_LINUX_DRIVER_v1.19.0_SOURCE modules
make[1]: Entering directory '/usr/src/linux-headers-4.19.93-v7l+'
  CC [M]  /home/pi/AX88179_178A_LINUX_DRIVER_v1.19.0_SOURCE/ax88179_178a.o
/home/pi/AX88179_178A_LINUX_DRIVER_v1.19.0_SOURCE/ax88179_178a.c:54:3: warning: #warning IP align 2 [-Wcpp]
  #warning IP align 2
   ^~~~~~~
  Building modules, stage 2.
  MODPOST 1 modules
  CC      /home/pi/AX88179_178A_LINUX_DRIVER_v1.19.0_SOURCE/ax88179_178a.mod.o
  LD [M]  /home/pi/AX88179_178A_LINUX_DRIVER_v1.19.0_SOURCE/ax88179_178a.ko
make[1]: Leaving directory '/usr/src/linux-headers-4.19.93-v7l+'
`

6by9, 2020-02-05:
So seeing as NET_IP_ALIGN is 2, the 
```
if (NET_IP_ALIGN == 0)
   skb_pull(skb, 2);
```
clauses in ax88179_rx_fixup won't be executed, and we have 2 extra bytes in the buffer.
It really needs the datasheet to understand what is going on within the hardware to know whether this is the correct thing or not. We may be lucky that someone on the mailing list has the information to hand to confirm that.

fj604, 2020-02-05:
Now I'm confused. The snippet above is from the vendor driver which does not have the extra two bytes, and NET_IP_ALIGN is 2, it is never executed - but everything works?

fj604, 2020-02-05:
You can see my linux bug report here:
https://marc.info/?l=linux-netdev&m=158091806911389&w=2

6by9, 2020-02-05:
Apologies - getting confused by the diff. 

There is one ```skb_pull(skb,2)``` that has been made conditional such that it won't execute in the vendor driver 
https://github.com/raspberrypi/linux/blob/rpi-5.4.y/drivers/net/usb/ax88179_178a.c#L1391
Try removing it.

In the remaining diffs for that function we have another ifdef on RX_SKB_COPY. That appears NOT to be defined, and the remaining code is pretty close to the mainline driver.
Vendor driver:
```
		ax_skb = skb_clone(skb, GFP_ATOMIC);
		if (ax_skb) {
			ax_skb->len = pkt_len;
			skb_set_tail_pointer(ax_skb, ax_skb->len);
			ax_skb->truesize = ax_skb->len + sizeof(struct sk_buff);
			ax88179_rx_checksum(ax_skb, pkt_hdr);
			usbnet_skb_return(dev, ax_skb);
		} else {
			return 0;
		}
```
vs mainline
```
		ax_skb = skb_clone(skb, GFP_ATOMIC);
		if (ax_skb) {
			ax_skb->len = pkt_len;
			ax_skb->data = skb->data + 2;
			skb_set_tail_pointer(ax_skb, pkt_len);
			ax_skb->truesize = pkt_len + sizeof(struct sk_buff);
			ax88179_rx_checksum(ax_skb, pkt_hdr);
			usbnet_skb_return(dev, ax_skb);
		} else {
			return 0;
		}
```
That ```ax_skb->data = skb->data + 2``` also looks suspicious.

If you're happy building the kernel, then I'd try each in turn to see if they solve the problem under some/all circumstances.

Your post to the mailing list is fine. I can't make any further comment as I don't have the expertise in networking.

6by9, 2020-02-25:
Sorry to say that net-dev seem to be ignoring your request. Give it another couple of days and send a followup.

Could you try my suggestion above? I seem to recall I do have one of those adapters around, but if you have a setup that can easily replicate this then that's probably faster.

PAlbrechtHH, 2020-10-17:
Does this issue still exist in current stable Raspberry Pi OS?
I am thinking about getting a AX88179 based adapter for my Raspberry Pi 4b since it is the cheapest one that is available in my country.

hurda, 2020-10-19:
I guess this commit applies to this issue: https://github.com/raspberrypi/linux/commit/27b70214fc6995087c76a24cc3ff2742a23aff5d#diff-0ac966c0229e99d07523378c08151f97e24c8406d087714de6eb7248eb41f42e

6by9, 2020-10-19:
Yes, it looks like upstream finally woke up, and managed to get Asix assistance too.
https://lore.kernel.org/netdev/20200527060334.19441-1-jk@ozlabs.org/
https://lore.kernel.org/netdev/20200615025456.30219-1-jk@ozlabs.org/

ri-pa, 2020-10-19:
Thanks for the information hurda and 6by9.
So this issue could theoretically be set to resolved?

6by9, 2020-12-09:
Issue resolved by upstream, and backported to 5.4.

closed 2020-12-09 by 6by9