raspberrypi/linux #3457: Rpi4 official 7'' LCD vsync callback number is 67-68 and not 60. Why? (Pi3 at the same display makes 60 as I espect)

ArtArt0, 2020-02-08:


JamesH65, 2020-02-14:
Can you be a bit more specific in what you are reporting please? What do you mean by vsync callback number? How are you encountering the fault?

ArtArt0, 2020-02-16:
Yes, of course.

In Rpi4 I use the /dev/dri/cardX device to render to the screen via opengl es.

First I render, set the `page_flip_handler`, call `drmModePageFlip` and wait for device is readable by `select`.
If the select happens, I call the `drmHandleEvent` (to call back my callback function).
In the callback function I render, and call the `drmModePageFlip`.

The FPS is calling number to `page_flip_handler` or `drmHandleEvent`.

It seems to work normally, except the FPS what is ~78, but I expect 60.