raspberrypi/linux #3458: On Rpi4 the second call of drmModePageFlip blocks in dual monitor mode. The same on an native Ubuntu runs correctly.

ArtArt0, 2020-02-08:


JamesH65, 2020-02-14:
Is this a permanent block - everything stops working requiring shutting down the code/Pi? What specific circumstances and what versions of software are you using? (kernel, firmware etc). Any other debug messages available? Your report is a little short of details.

popcornmix, 2020-02-14:
Example code that triggers the issue would be helpful.

ArtArt0, 2020-02-16:
It is a little bit hard to simplify my code for this, so I try to write what I am doing:

I am using the official LCD (800x480) and a HDMI monitor (1366x768).

In Rpi4 I use the /dev/dri/cardX device to render to the screen via opengl es.

First I set the `page_flip_handler`, render, call `drmModePageFlip`, render2, call `drmModePageFlip` and I would waiting for device is readable by select.
But I can not, because the second call of `drmModePageFlip` blocks, and when it is released it is too late, the select does not block.
I think the `drmModePageFlip` must not block, and the select should even if I am using two display.

It is OK in this way on native Ubuntu, so the  second `drmModePageFlip` returns immadietly and I am wait in the `select`.