raspberrypi/linux #3459: On-board Bluetooth: loading firmware with devicetree

macmpi, 2020-02-10:
I'm trying to understand how to load Bluetooth firmware with device-tree rather than using [`pi-bluetooth`](https://github.com/RPi-Distro/pi-bluetooth) Raspbian package together with related `btuart` script and assorted downstream-patched depreciated `hciattach` (indeed `btattach` seems still unreliable for this).
Rationale is, for instance, to seamlessly setup Bluetooth on a PiZeroW with different distributions such as AlpineLinux, and enjoy proper Bluetooth operations even when audio streaming is involved (A2DP).

[bcm2835-rpi-zero-w.dts](https://github.com/raspberrypi/linux/blob/rpi-5.4.y/arch/arm/boot/dts/bcm2835-rpi-zero-w.dts#L134-L144) seems to have the pieces together to do the trick, but firmware does appear to load by default.
I noticed @lategoodbye and @pelwell discussed similar topic on [Pi4 context](https://github.com/raspberrypi/linux/issues/3260), but I'm not sure this now supported on all Pi device with onboard Bluetooth.

Do we need any statement in `config.txt` or some other extra setup to do the magic?

lategoodbye, 2020-02-10:
There are two essential parts which are:
- the necessary serdev driver ( config BT_HCIUART_BCM )
- the necessary bluetooth section with contains the compatible brcm,bcm43438-bt to trigger probing of the serdev driver

Both requirements should be fulfilled for all RPi variants with mainline kernel >= 5.5. Btw the RPi 4 required a patch in the serdev driver (include in Linux 5.4) to get working.

macmpi, 2020-02-10:
Thanks.
bcm27xx DT files for downstream kernel do not have the section indeed, but I thought with [bcmbt-overlay](https://github.com/RoEdAl/alarm-bluetooth-raspberrypi/blob/master/bcmbt-overlay.dts) it would work, but firmware does not happen to load on AlpineLinux unfortunately.

bcm28xx DT files have the section (even on 5.4), but I guess those can only be used with upstream kernel, right?

lategoodbye, 2020-02-11:
I think you should provide the following information:
- used Raspberry Pi board
- dmesg output
- full path of the bluetooth firmware

macmpi, 2020-02-11:
Typically, I'm on a raspberry PiZero W under Alpine 3.11 (distribution's stock linux-rpi kernel 5.4), firmware at /lib/firmware/brcm/BCM43430A1.hcd

I compare 2 setups:
1) with [btuart](https://github.com/RPi-Distro/pi-bluetooth/blob/master/usr/bin/btuart) script running during sysinit: console shows firmware gets loaded (but dmesg does not). dmesg & `hciconfig feature` are [here](https://pastebin.com/C4UPpk6z). All is (mostly) fine on Bluetooth side.

2) without that sysinit script, but with [bcmbt overlay](https://github.com/RoEdAl/alarm-bluetooth-raspberrypi) instead. dmesg & `hciconfig feature` are [here](https://pastebin.com/JMRpK3BZ). Firmware not loaded, and hci0 not available.

I'd like to load firmware with device-tree rather than with btuart/downstream hciattach, if at all reliably possible. I guess bcm27xx DT files will probably include this sometime soon?
Thanks for any hint.

(note, unrelated, but in all case I'm having wierd errors on my cdrom drive, I'm at lost with...usb driver issue?)
```
[   19.831006] blk_update_request: I/O error, dev sr0, sector 910720 op 0x0:(READ) flags 0x0 phys_seg 1 prio class 0
[   19.831025] Buffer I/O error on dev sr0, logical block 113840, async page read
```

lategoodbye, 2020-02-11:
Did you checked that serdev and BT_HCIUART_BCM is enabled in kernel config?

macmpi, 2020-02-11:
Yes, that part seems ok
```
cat /boot/config-rpi | grep BT_HCIUART_BCM
CONFIG_BT_HCIUART_BCM=y

lsmod | grep serdev
serdev                 20480  0
```

macmpi, 2020-02-13:
I see there's some recent significant work on DT files stuff (upstream/downstream), including `overlay_prefix` firmware support...

@pelwell is there any chance Bluetooth firmware loading get integrated in downstream DT files as part of this work?
Any suggested alternative method to get it working in the meantime?

macmpi, 2020-07-01:
closing as per https://github.com/raspberrypi/linux/pull/3682

closed 2020-07-01 by macmpi