raspberrypi/linux #3460: Dispatching compute shader costs us at least 30ms

notogawa, 2020-02-10:
**Describe the bug**

When we dispatch a shader program using `ioctl(SUBMIT_CSD)` on Raspberry Pi 4, it costs us 30ms. Calling `ioctl(SUBMIT_CSD)` 10 times before `ioctl(WAIT_BO)` means 300ms to complete all jobs. Even if a shader program is short enough and do-nothing, this cost is not discounted at all.

We checked trace and confirmed that `v3d_clean_caches`, called after each dispatched job, has this 30ms. https://github.com/raspberrypi/linux/commit/c1a40a21b2637d2ce7f71a71cc9c7fe101a14fef says this `v3d_clean_caches` is required to avoid MMU errors. Is this a performance problem of `v3d_clean_caches` ?

**To reproduce**

[This is an example program](https://gist.github.com/notogawa/36d0cc9168ae3236902729f26064281d#file-main-c) to reproduce. In this example, a shader is 8 `nop` instructions.

~~~console
git clone https://gist.github.com/notogawa/36d0cc9168ae3236902729f26064281d
cd 36d0cc9168ae3236902729f26064281d
gcc -o main main.c
./main
~~~

**Expected behaviour**

Shader jobs should be completed at a suitable time. I think 30ms is too long.

**Actual behaviour**

at least 30ms / dispatch job

[This is an example program output.](https://gist.github.com/notogawa/36d0cc9168ae3236902729f26064281d#file-main-log)

**System**

* Which model of Raspberry Pi? e.g. Pi3B+, PiZeroW

~~~
Hardware        : BCM2835
Revision        : c03112
Serial          : 100000006499218e
Model           : Raspberry Pi 4 Model B Rev 1.2
~~~

* Which OS and version (`cat /etc/rpi-issue`)?

~~~
Raspberry Pi reference 2019-09-26
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 80d486687ea77d31fc3fc13cf3a2f8b464e129be, stage2
~~~

* Which firmware version (`vcgencmd version`)?

~~~
Sep 24 2019 17:34:30
Copyright (c) 2012 Broadcom
version cd3add54955f8fa065b414d8fc07c525e7ddffc8 (clean) (release) (start)
~~~

* Which kernel version (`uname -a`)?

~~~
Linux pi4 4.19.93-v7l+ #1290 SMP Fri Jan 10 16:45:11 GMT 2020 armv7l GNU/Linux
~~~

popcornmix, 2020-02-10:
Thanks for detailed report. I can reproduce your results when running form console.

Interestingly running from desktop (after startx) gives rather more desirable results:
[code]
[submit x10]
  try 1: 0.000030 sec
  try 2: 0.000030 sec
  try 3: 0.000030 sec
  try 4: 0.000031 sec
  try 5: 0.000030 sec
[/code]

It is possible that having the 3d hardware inactive is increasing the time taken.
Perhaps clocks are lower, or something is required for submitted job to complete is missing when inactive and we're falling back to a (30ms) timeout.

notogawa, 2020-02-10:
Interesting. I certainly use raspbian lite without X.

popcornmix, 2020-02-10:
I don't think it's clock related. gpu_freq=500 gpu_freq_min=500 means
```
vcgencmd measure_clock v3d
```
is always at 500, but still see the same 30ms pre-desktop.

popcornmix, 2020-02-10:
Running kmscube in background does not help. Still see 30ms. That surprised me a little.

notogawa, 2020-02-12:
I try to run from raspbian desktop. still see 30ms.

popcornmix, 2020-02-12:
Actually the configuration that gave the low times was using the KMS driver and desktop enabled.
FKMS with desktop enabled was still 30ms.

If you have a pi3 you can test with kms enabled. On Pi4 kms is currently experimental and not supported by standard kernel.

@anholt does this ring any bells with yout? Any idea where a 30ms block may come from through the shader ioctl?

anholt, 2020-02-12:
If the delay is in clean_caches as the OP reports, then this is something we should probably have Igalia talk to Dave Emett about.  It definitely shouldn't take that long, and I don't believe there's any good way to avoid it (even if you're feeding CSD output to other GPU usage, you're still going to have to flush most of it).

The only thing I think should be checked before escalating is that our GPU clock measures at the expected rate while this program is running, since we've got some weird clock hacks.

popcornmix, 2020-02-12:
I tested v3d clock forced to 500Mhz and still get the 30ms.

It was 1000 times faster when running with desktop active on kms driver, so I don't think the cache flush inherently takes notable time.

The 30ms is pretty consistent across clock rates, so I'm suspecting it's a timeout of something.

popcornmix, 2020-02-12:
@notogawa how did you determine the delay was in v3d_clean_caches? Did you instrument the kernel code (exactly where) or rely on debug messages?

notogawa, 2020-02-12:
by `function_graph` tracer.

This is part of trace.
~~~
...
 3)   0.944 us    |    v3d_job_put [v3d]();
 3)   0.944 us    |    v3d_job_put [v3d]();
 3) + 33.092 us   |  }
 3)               |  v3d_submit_csd_ioctl [v3d]() {
 3)               |    v3d_job_init [v3d]() {
 3)   0.981 us    |      v3d_add_dep [v3d]();
 3)   3.297 us    |    }
 3)               |    v3d_job_init [v3d]() {
 3)   0.926 us    |      v3d_add_dep [v3d]();
 3)   2.963 us    |    }
 3)   2.000 us    |    v3d_lookup_bos [v3d]();
 3)               |    v3d_lock_bo_reservations [v3d]() {
 3)   1.129 us    |      v3d_add_dep [v3d]();
 3)   3.777 us    |    }
 3)   2.018 us    |    v3d_push_job [v3d]();
 3)   0.963 us    |    v3d_add_dep [v3d]();
 3)   1.592 us    |    v3d_push_job [v3d]();
 3)   1.222 us    |    v3d_attach_fences_and_unlock_reservation [v3d]();
 3)   0.926 us    |    v3d_job_put [v3d]();
 3)   0.945 us    |    v3d_job_put [v3d]();
 3) + 31.852 us   |  }
 3)               |  v3d_wait_bo_ioctl [v3d]() {
 ------------------------------------------
 0)  kworker-868   =>  v3d_cac-232  
 ------------------------------------------
 0) * 29277.25 us |    } /* v3d_clean_caches [v3d] */
 0) * 29280.40 us |  } /* v3d_cache_clean_job_run [v3d] */
 ------------------------------------------
 3)  nosetes-3402  =>  v3d_csd-231  
 ------------------------------------------
 3) * 29069.85 us |    } /* v3d_invalidate_caches [v3d] */
 3)   2.000 us    |    v3d_fence_create [v3d]();
 0)   1.370 us    |  v3d_job_dependency [v3d]();
 3) * 29075.98 us |  } /* v3d_csd_job_run [v3d] */
 0) + 24.130 us   |  v3d_irq [v3d]();
 0)   0.926 us    |  v3d_job_dependency [v3d]();
 0)   0.907 us    |  v3d_job_dependency [v3d]();
 3)   1.222 us    |  v3d_job_dependency [v3d]();
...
~~~

popcornmix, 2020-02-12:
Looks like there is just polling for the cache clean to complete and the cache_clean_lock mutex (which doesn't look like it could be held for long anywhere) inside v3d_clean_caches.

I assume the write combiner flush should be a trivial op, so the L2T is more likely to take time.

I still wonder if the hardware is disabled in some way when the flush starts and it only gets to complete when some other thread kicks it.

popcornmix, 2020-02-12:
@notogawa can you rebuild the kernel? Try this patch
```
diff --git a/drivers/gpu/drm/v3d/v3d_drv.h b/drivers/gpu/drm/v3d/v3d_drv.h
index 15bc4e28b6cc..f5f1400142aa 100644
--- a/drivers/gpu/drm/v3d/v3d_drv.h
+++ b/drivers/gpu/drm/v3d/v3d_drv.h
@@ -276,7 +276,7 @@ struct v3d_csd_job {
                                ret__ = -ETIMEDOUT;                     \
                        break;                                          \
                }                                                       \
-               msleep(1);                                      \
+               /*msleep(1);*/                                  \
        }                                                               \
        ret__;                                                          \
 })
```
@anholt I think we should be using usleep_range here.

notogawa, 2020-02-13:
I rebuild the kernel and get the result that I expected.

~~~
[submit x10]
  try 1: 0.000199 sec
  try 2: 0.000197 sec
  try 3: 0.000199 sec
  try 4: 0.000194 sec
  try 5: 0.000195 sec
~~~

anholt, 2020-02-14:
Wow, yeah, that should definitely not be an msleep.  Probably want to steal i915's wait_for helper.

JamesH65, 2020-02-14:
vc4_drv.h has a slightly different implementation, but still uses msleep (which in previous experience has a minimum wait of about 30ms on the Pi's).

JamesH65, 2020-02-17:
OK, tried the definition in intel_drv.h (from i915 tree) and tested with a number of 3D apps, all seemed to work fine. Results from the test app referenced above, all seem pretty good.
```

[submit x1]
  try 1: 0.000009 sec
  try 2: 0.000008 sec
  try 3: 0.000007 sec
  try 4: 0.000006 sec
  try 5: 0.000006 sec
[submit x2]
  try 1: 0.000009 sec
  try 2: 0.000008 sec
  try 3: 0.000008 sec
  try 4: 0.000009 sec
  try 5: 0.000008 sec
[submit x3]
  try 1: 0.000010 sec
  try 2: 0.000010 sec
  try 3: 0.000011 sec
  try 4: 0.000010 sec
  try 5: 0.000010 sec
[submit x4]
  try 1: 0.000012 sec
  try 2: 0.000013 sec
  try 3: 0.000012 sec
  try 4: 0.000012 sec
  try 5: 0.000012 sec
[submit x5]
  try 1: 0.000014 sec
  try 2: 0.000015 sec
  try 3: 0.000014 sec
  try 4: 0.000014 sec
  try 5: 0.000014 sec
[submit x6]
  try 1: 0.000016 sec
  try 2: 0.000016 sec
  try 3: 0.000016 sec
  try 4: 0.000016 sec
  try 5: 0.000016 sec
[submit x7]
  try 1: 0.000018 sec
  try 2: 0.000018 sec
  try 3: 0.000018 sec
  try 4: 0.000018 sec
  try 5: 0.000018 sec
[submit x8]
  try 1: 0.000020 sec
  try 2: 0.000020 sec
  try 3: 0.000020 sec
  try 4: 0.000020 sec
  try 5: 0.000020 sec
[submit x9]
  try 1: 0.000022 sec
  try 2: 0.000022 sec
  try 3: 0.000022 sec
  try 4: 0.000022 sec
  try 5: 0.000022 sec
[submit x10]
  try 1: 0.000024 sec
  try 2: 0.000024 sec
  try 3: 0.000024 sec
  try 4: 0.000024 sec
  try 5: 0.000024 sec

```

JamesH65, 2020-02-17:
Where do I need to submit the patch? Raspberry Pi first, or kernel somewhere?

popcornmix, 2020-02-17:
Upstream is the right place for the patch.
We can cherry pick it earlier to raspberry pi tree while waiting for it to be accepted.

JamesH65, 2020-02-17:
OK, have submitted upstream, probably wrongly in some obscure way.

popcornmix, 2020-02-17:
I'm sure they'll let you know...

JamesH65, 2020-02-19:
Made it to the LKM and DRM mailing list, but no responses.

notogawa, 2020-03-05:
How is the progress on  https://lkml.org/lkml/2020/2/20/241 ?

anholt, 2020-03-05:
Merged now that I sorted out the dim tooling.

JamesH65, 2020-03-05:
Cool. @notogawa Can you close once you are happy the issue has been fixed.

notogawa, 2020-03-05:
Thx. I'm waiting for raspbian kernel update.

closed 2020-03-05 by notogawa