raspberrypi/linux #3462: Latest kernel update omits CP-2102 driver?

IowaDave, 2020-02-11:
**Is this the right place for my bug report?**  Yes, most likely involves kernel

**Describe the bug**
 Recently did apt full-upgrade, Buster Debian version is now 10.3. After this upgrade my NodeMCU hardware boards no longer connect to a /dev/ttyusbX serial port. Perhaps the CP-2012 USB-to-UART driver was omitted?

**To reproduce**
Upgrade Buster to 10.3. 
Then plug in a development board (ESP8266 or ESP32 that uses the CP2102 device. 
Then run dmesg.
Notice the devise is detected and connected, correctly identified as Silicon Labs CP2102. However, it is not then joined to a /dev/ttyusbX port.

**Expected behaviour**
Expected the device to be joined to a port, as it has been done on previous versions of the kernel

**Actual behaviour**
See "to reporduce", above.

**System**
 Copy and paste the results of the raspinfo command in to this section. Alternatively, copy and paste a pastebin link, or add answers to the following questions:

Umm... "bash: raspinfo: command not found"  ??? So I am answering the questions below

* Which model of Raspberry Pi? e.g. Pi3B+, PiZeroW
  > Pi 4
* Which OS and version (`cat /etc/rpi-issue`)?
  > Raspberry Pi reference 2019-09-26
  > Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 
  > 80d486687ea77d31fc3fc13cf3a2f8b464e129be, stage5

* Which firmware version (`vcgencmd version`)?
  > Jan 10 2020 16:52:04 
  > Copyright (c) 2012 Broadcom
  > version fdb5c37e330e7cb3027ac4fcc5b1cd5f244b351f (clean) (release) (start)

* Which kernel version (`uname -a`)?
  > Linux raspberrypi 4.19.93-v7l+ #1290 SMP Fri Jan 10 16:45:11 GMT 2020 armv7l GNU/Linux

**Logs**
[ 4900.745986] usb 1-1.2: new full-speed USB device number 9 using xhci_hcd
[ 4900.924194] usb 1-1.2: New USB device found, idVendor=10c4, idProduct=ea60, bcdDevice= 1.00
[ 4900.924213] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 4900.924227] usb 1-1.2: Product: CP2102 USB to UART Bridge Controller
[ 4900.924240] usb 1-1.2: Manufacturer: Silicon Labs
[ 4900.924252] usb 1-1.2: SerialNumber: 0001
[ 5108.604885] usb 1-1.2: USB disconnect, device number 9
[ 5121.194158] usb 1-1.1: new full-speed USB device number 10 using xhci_hcd
[ 5121.330521] usb 1-1.1: New USB device found, idVendor=1a86, idProduct=7523, bcdDevice= 2.54
[ 5121.330539] usb 1-1.1: New USB device strings: Mfr=0, Product=2, SerialNumber=0
[ 5121.330552] usb 1-1.1: Product: USB2.0-Serial
[ 6082.314600] usb 1-1.1: USB disconnect, device number 10
[ 6093.622110] usb 1-1.2: new full-speed USB device number 11 using xhci_hcd
[ 6093.758540] usb 1-1.2: New USB device found, idVendor=1a86, idProduct=7523, bcdDevice= 2.54
[ 6093.758560] usb 1-1.2: New USB device strings: Mfr=0, Product=2, SerialNumber=0
[ 6093.758573] usb 1-1.2: Product: USB2.0-Serial

**Additional context**
This issue prevents using NodeMCU development boards with the Arduino IDE if they rely upon a Silicon Labs CP2102 usb interface..

IowaDave, 2020-02-12:
After further digging I found a work-around. 

Note the problem occurred when doing apt full upgrade to the Feb 2020 kernel from the Sept 2019 kernel. 

By a happy contrast, doing a fresh install of the Feb 2020 distro appears to bring the CP1202 driver into the system. I did this, then plugged a NodeMCU ESP32 into USB. Received the following report from dmesg:

  > cp210x converter now attached to ttyUSB0

Serial communication with the device functions as expected.

pelwell, 2020-02-12:
Nothing has changed in the kernel image recently with regard to the CP210X driver, so I'm going to put this down to an upgrade glitch.

closed 2020-02-12 by pelwell

fernandoarcevega, 2021-03-01:
IowaDave,
Can you show me how you fixed the problem? Please.

IowaDave, 2021-03-03:
Hi! Thanks for checking in. 

I'm happy to help and I will do my best to answer the question. Then I will try to explain it a little. 

Answer: The problem was fixed by installing a new Raspberry Pi OS image on my SD card. 

Explanation: My problem appeared when I ran the command  ```sudo apt full-upgrade```. The upgrade procedure left my system without the CP210X driver. 

I was happy to learn that the driver is still included in the standard Raspberry Pi OS image. Installing a fresh image makes the driver available again.

Detailed steps for my procedure:
1. Backup all of my files and make a list of user-installed applications that I added to my R-Pi.
2. Install a fresh Raspberry Pi OS on my SD card.
3. Re-install my applications and restore my backed-up files.

An alternate procedure might be:
1. Remove the SD card from the R-Pi and set it aside for the moment.
2. Install a fresh OS on a new SD card.
3. Boot the R-Pi up on that new card. Verify that the CP-1202 driver is present and functioning. (Plug in an 8266 then examine the output of the dmesg command in the terminal, for example.)
4. Treat the former SD card as the "backup" and copy user-created files to the new SD card from that older card. Re-install your user-selected set of additional applications.

I am happy to report that system upgrades that I have done since that time have not caused any problem with the CP-1202 driver.

Yes, the procedure took a little bit of time. However, the procedure definitely fixed the problem in a very straight-forward, non-technical way that a novice like me can easily understand and trust.

Thanks for reading this explanation. I hope it helps!

fernandoarcevega, 2021-03-03:
Hey Dave,

Thanks so much for your help, so helpful.

Best regards,
Fernando.