raspberrypi/linux #3463: Raspberry Pi 4 Fails to Reboot with USB to Ethernet Adapter

Omoeba, 2020-02-12:
**Describe the bug**
When a USB to Ethernet adapter is attached, the Pi sometimes fails to reboot properly. The adapter lights are off and the main ethernet lights are on. The original ethernet is still flashing but the system appears to be in a dead state and it is impossible to SSH into. The only way to temporarily resolve the bug is to disconnect and reconnect power.

**To reproduce**
Plug an AX88179 based USB to ethernet adapter into any of the USB 3.0 ports on the Pi. I tested with the AmazonBasics AE3101X1 and the Startech ST3300GU3B. Disconnect and reconnect power and issue the command ```sudo reboot```. Keep issuing ```sudo reboot``` as soon as the Pi is accessible via SSH until it doesn't come back up. This usually occurs within 10-15 reboots.

**Expected behaviour**
The Raspberry Pi reboots normally with the indicator lights on the USB to ethernet adapter turning off first. A few seconds later, the indicator lights on the main ethernet port turns off and then both the adapter and the main ethernet lights come back up. The Pi is accessible from SSH a few seconds after the lights come back up.

**Actual behaviour**
The Pi starts rebooting normally with the indicator lights on the USB to ethernet adapter turning off. However, the lights on the main ethernet port never turns off and the lights on the adapter never turns back on. The Pi is stuck in a "dead" state where the main ethernet port is blinking but it is impossible to access it via SSH. The only way to return the Pi to normal is cutting and restoring the power.

**System**
 Copy and paste the results of the raspinfo command in to this section. Alternatively, copy and paste a pastebin link, or add answers to the following questions:

* Which model of Raspberry Pi? e.g. Pi3B+, PiZeroW
Raspberry Pi 4 Model B 4GB RAM
* Which OS and version (`cat /etc/rpi-issue`)?
Raspberry Pi reference 2019-09-26
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 80d486687ea77d31fc3fc13cf3a2f8b464e129be, stage4
* Which firmware version (`vcgencmd version`)?
Copyright (c) 2012 Broadcom
version d21c2d51634b793e0b0ce2816c4e3d1cf0d87e6b (clean) (release) (start)
* Which kernel version (`uname -a`)?
Linux gateway 4.19.97-v7l+ #1294 SMP Thu Jan 30 13:21:14 GMT 2020 armv7l GNU/Linux

**Logs**
Here are the ```journalctl``` shutdown logs
```
Feb 11 19:28:42 gateway systemd[1]: Reached target Shutdown.
Feb 11 19:28:42 gateway systemd[1]: Reached target Final Step.
Feb 11 19:28:42 gateway systemd[1]: systemd-reboot.service: Succeeded.
Feb 11 19:28:42 gateway systemd[1]: Started Reboot.
Feb 11 19:28:42 gateway systemd[1]: Reached target Reboot.
Feb 11 19:28:42 gateway systemd[1]: Stopping Restore / save the current clock...
Feb 11 19:28:42 gateway systemd[1]: Shutting down.
Feb 11 19:28:43 gateway systemd[1]: Hardware watchdog 'Broadcom BCM2835 Watchdog timer', version 0
Feb 11 19:28:43 gateway systemd[1]: Failed to set timeout to 600s: Invalid argument
Feb 11 19:28:43 gateway kernel: systemd-shutdow: 39 output lines suppressed due to ratelimiting
Feb 11 19:28:43 gateway systemd-shutdown[1]: Syncing filesystems and block devices.
Feb 11 19:28:44 gateway systemd-shutdown[1]: Sending SIGTERM to remaining processes...
Feb 11 19:28:44 gateway systemd-journald[127]: Journal stopped
```

**Additional context**
The ```Invalid argument``` line occurs whether the reboot fails or completes successfully.

JamesH65, 2020-02-14:
I think that invalid argument is simply because 600s seems to be larger than the maximum that the watchdog timer can accept. (I think it's defined maximum according to the driver is 255). As for the issue itself, not sure. Anything in dmesg?

ablacksheep, 2020-02-19:
I have tried to reproduce this with a slightly different setup. I have been using 2x StarTech USB3 AX88179 (USB31000S) based adapters. I have not been able to reproduce it on Raspbian Buster Lite after approximate 40 retries.

With the testing above and the info provided from creator i would suggest this is not necessarily a bug related to the AX88179. As of the description it sounds important to issue the reboot command as fast as possible after ssh is up(?), and most likely before systemd has finished starting all its targets. If that actually is the case, maybe @Omoeba  could try to reproduce the bug with no usb adapter connected?

```
cat /etc/rpi-issue 
Raspberry Pi reference 2019-09-26
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 80d486687ea77d31fc3fc13cf3a2f8b464e129be, stage2
```

```
uname -a
Linux raspberrypi 4.19.97-v7l+ #1293 SMP Wed Jan 22 17:16:14 GMT 2020 armv7l GNU/Linux
```

```
vcgencmd version
Jan 22 2020 17:23:15 
Copyright (c) 2012 Broadcom
version acb03c7d18aa08f25f6419d696d0cc39761d3e62 (clean) (release) (start)
```

Omoeba, 2020-02-22:
The bug appears to be related to [this forum topic](https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=262649&hilit=usb+quality+control) and [fixed](https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=262649&hilit=usb+quality+control&start=125#p1610738).

closed 2020-02-22 by Omoeba