raspberrypi/linux #3466: Missing patches v4l2 patches upstream?

bkchr, 2020-02-13:
Hey,
I'm currently trying to get the v4l2 interface working on 5.5.3 with mainline. I'm now trying every kernel since 5.x started and something is broken. With `ffmpeg` I get the following output:
```
[h264 @ 0x1c989010] non-existing PPS 0 referenced
    Last message repeated 1 times
[h264 @ 0x1c989010] decode_slice_header error
[h264 @ 0x1c989010] no frame!                                                                                                                                                                                       
[h264 @ 0x1c989010] non-existing PPS 0 referenced                                                                                                                                                                   
    Last message repeated 1 times
[h264 @ 0x1c989010] decode_slice_header error
[h264 @ 0x1c989010] no frame!                                                                                                                                                                                       
[h264 @ 0x1c989010] non-existing PPS 0 referenced  
```

I know that this is not upstream, but you can probably help me and point me to a patch that I could apply locally? 
It is really weird as with the 4.x series the camera is working without any problems, but something must have changed on the switch to 5.x. Would be nice if somebody could help me :)

vianpl, 2020-02-14:
Hi, which device are you using? Support for RPi4 isn't upstream yet, I tested the camera on RPi3 last week with upstream and worked fine for me.

bkchr, 2020-02-14:
RPI3b. Hmm, I don't know what I'm missing :( GStreamer and Fffmpeg fail to record a video with any kernel starting with 5.0