raspberrypi/linux #3467: bcm2835-v4l2: Failed to enable capture port (ffmpeg + 1080p)

cynecx, 2020-02-15:
**Describe the bug**
I am getting `bcm2835-v4l2: Failed to enable capture port - error -1. Disabling camera port again` when trying to capture camera data with ffmpeg.

**To reproduce**
Run `ffmpeg -f video4linux2 -input_format h264 -video_size 1920x1080 -framerate 30 -i /dev/video0 -vcodec copy -an test3.h264`. It should fail with:

```
[video4linux2,v4l2 @ 0x993220] ioctl(VIDIOC_STREAMON): Operation not permitted
/dev/video0: Operation not permitted
```

dmesg:

```
[ 3772.804430] bcm2835-v4l2: Failed to enable capture port - error -1. Disabling camera port again
[ 3772.815109] ------------[ cut here ]------------
[ 3772.815162] WARNING: CPU: 3 PID: 3331 at drivers/media/common/videobuf2/videobuf2-core.c:1365 vb2_start_streaming+0xe4/0x160 [videobuf2_common]
[ 3772.815169] Modules linked in: bnep hci_uart btbcm serdev bluetooth ecdh_generic 8021q garp stp llc brcmfmac brcmutil sha256_generic cfg80211 rfkill bcm2835_codec(C) bcm2835_v4l2(C) v4l2_mem2mem snd_bcm2835(C) bcm2835_mmal_vchiq(C) snd_pcm v4l2_common videobuf2_dma_contig videobuf2_vmalloc raspberrypi_hwmon snd_timer hwmon videobuf2_memops snd videobuf2_v4l2 videobuf2_common videodev media vc_sm_cma(C) uio_pdrv_genirq uio fixed ip_tables x_tables ipv6
[ 3772.815293] CPU: 3 PID: 3331 Comm: ffmpeg Tainted: G         C        4.19.97-v7+ #1294
[ 3772.815298] Hardware name: BCM2835
[ 3772.815329] [<801120c0>] (unwind_backtrace) from [<8010d5f4>] (show_stack+0x20/0x24)
[ 3772.815347] [<8010d5f4>] (show_stack) from [<80845f28>] (dump_stack+0xe0/0x124)
[ 3772.815365] [<80845f28>] (dump_stack) from [<80120c9c>] (__warn+0x104/0x11c)
[ 3772.815381] [<80120c9c>] (__warn) from [<80120dec>] (warn_slowpath_null+0x50/0x58)
[ 3772.815421] [<80120dec>] (warn_slowpath_null) from [<7f1e66f8>] (vb2_start_streaming+0xe4/0x160 [videobuf2_common])
[ 3772.815491] [<7f1e66f8>] (vb2_start_streaming [videobuf2_common]) from [<7f1e8130>] (vb2_core_streamon+0x130/0x16c [videobuf2_common])
[ 3772.815547] [<7f1e8130>] (vb2_core_streamon [videobuf2_common]) from [<7f1fe8a8>] (vb2_streamon+0x40/0x60 [videobuf2_v4l2])
[ 3772.815591] [<7f1fe8a8>] (vb2_streamon [videobuf2_v4l2]) from [<7f1fe918>] (vb2_ioctl_streamon+0x50/0x54 [videobuf2_v4l2])
[ 3772.815720] [<7f1fe918>] (vb2_ioctl_streamon [videobuf2_v4l2]) from [<7f150e3c>] (v4l_streamon+0x28/0x2c [videodev])
[ 3772.815919] [<7f150e3c>] (v4l_streamon [videodev]) from [<7f153a38>] (__video_do_ioctl+0x268/0x528 [videodev])
[ 3772.816114] [<7f153a38>] (__video_do_ioctl [videodev]) from [<7f157470>] (video_usercopy+0x218/0x600 [videodev])
[ 3772.816311] [<7f157470>] (video_usercopy [videodev]) from [<7f157878>] (video_ioctl2+0x20/0x24 [videodev])
[ 3772.816506] [<7f157878>] (video_ioctl2 [videodev]) from [<7f14f144>] (v4l2_ioctl+0x4c/0x60 [videodev])
[ 3772.816613] [<7f14f144>] (v4l2_ioctl [videodev]) from [<802e4d68>] (do_vfs_ioctl+0xbc/0x804)
[ 3772.816630] [<802e4d68>] (do_vfs_ioctl) from [<802e54f4>] (ksys_ioctl+0x44/0x6c)
[ 3772.816643] [<802e54f4>] (ksys_ioctl) from [<802e5534>] (sys_ioctl+0x18/0x1c)
[ 3772.816657] [<802e5534>] (sys_ioctl) from [<80101000>] (ret_fast_syscall+0x0/0x28)
[ 3772.816664] Exception stack(0x8fd25fa8 to 0x8fd25ff0)
[ 3772.816676] 5fa0:                   00993220 00000020 00000003 40045612 7edc3ac8 753f0510
[ 3772.816689] 5fc0: 00993220 00000020 7edc3b68 00000036 c044560f 7edc3b70 009939b0 34363248
[ 3772.816698] 5fe0: 76efca84 7edc3a94 76ee1720 753f051c
[ 3772.816706] ---[ end trace 51f8dd700a39135b ]---
```

**Expected behaviour**
The ffmpeg command should succeed and start capturing the raw h264 stream.

This however, `ffmpeg -f video4linux2 -input_format h264 -video_size 1280x720 -framerate 30 -i /dev/video0 -vcodec copy -an test3.h264` (notice the difference resolution) works just fine.

Also raspivid also works correctly with 1080p resolution.

**Actual behaviour**
See above.

**System**
 Copy and paste the results of the raspinfo command in to this section. Alternatively, copy and paste a pastebin link, or add answers to the following questions:

* Which model of Raspberry Pi? Pi3A+, PiZeroW
* Which OS and version?
```
Raspberry Pi reference 2020-02-13
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 5f884374b6ac6e155330c58caa1fb7249b8badf1, stage2
```
* Which firmware version (`vcgencmd version`)?
```
Feb 12 2020 12:39:27 
Copyright (c) 2012 Broadcom
version 53a54c770c493957d99bf49762dfabc4eee00e45 (clean) (release) (start_x)
```
* Which kernel version (`uname -a`)?
`Linux raspberrypi 4.19.97-v7+ #1294 SMP Thu Jan 30 13:15:58 GMT 2020 armv7l GNU/Linux`

bwsw, 2020-04-24:
I get the same bug with the same config incl. kernel version, etc.

bwsw, 2020-04-24:
Just for a note - it's not a bug:
set 
```
gpu_mem=256
```
in `/boot/config.txt` and it's fine.

6by9, 2020-04-24:
```gpu_mem=256``` sounds a touch excessive, but I haven't checked it out.
You can use ```sudo vcdbg reloc``` to show how much gpu_mem is in use whilst you are streaming, and then amend the value accordingly.

bwsw, 2020-04-27:
@6by9 Thanks, but because I don't have any other tasks rather then copy and upload h264, it's fine for me to use 256.

deadmoto, 2022-11-04:
Still facing the same problem with RPI 4 Model B now in 2022.
Tweak `gpu_mem=256` only helped to go up to 1280x720@60 but any resolution higher than that still resulted in `ioctl(VIDIOC_STREAMON): Operation not permitted`.
After coming across [a similar problem and a recommendation](https://raspberrypi.stackexchange.com/questions/106389/unable-to-stream-live-on-youtube-using-raspberry-pi-zero-w-using-ffmpeg#comment178852_106403) to tweak `over_voltage=1` and `force_turbo` options, I ignored the tweaks and simply tried different 3A power adapters.
With a different 3A power supply I finally got it working at 1920x1080@30.

6by9, 2022-11-04:
> helped to go up to 1280x720@60 but any resolution higher than that still resulted in ioctl(VIDIOC_STREAMON): Operation not permitted.

Be aware of H264 levels - https://en.wikipedia.org/wiki/H.264/MPEG-4_AVC#Levels

The hardware spec is for level 4.1, or 245,760 16x16 macroblocks/sec, and that is the default for the driver. That translates to 1080p30.1, or 720p68.3. The encoder validates the configuration of level, framerate, and resolution and will refuse to start if invalid. Increasing the resolution above 720p at 60fps is likely to be invalid.

The V4L2 drivers do have an H264 level control and will allow selection of level 4.2, but it is not 100% guaranteed to achieve the frame rate.