raspberrypi/linux #3468: Fe-Pi Audio inputs no longer working in v4.19

Fe-Pi, 2020-02-15:
The sgtl5000 CHIP_ANA_CTRL register, MUTE_ADC bit is no longer being cleared since this commit, https://github.com/raspberrypi/linux/commit/84bb4024b358a94bac23fa0ad434ec1fbe8deb54

v5.4 does not have this issue, because of this commit, https://github.com/raspberrypi/linux/commit/694b14554d75f2a1ae111202e71860d58b434a21 which is not included in v4.19

These are part of patchset https://lore.kernel.org/alsa-devel/20190719100524.23300-1-oleksandr.suvorov@toradex.com/

popcornmix, 2020-02-17:
Can you confirm that rpi-4.19.y kernel with 694b145 cherry-picked resolves your issue?
If so, I'm okay with cherry-picking it (assuming it is confirmed to fix the issue).

Fe-Pi, 2020-02-17:
Yes it did resolve the issue. Please cherry-pick and Thank you.

popcornmix, 2020-02-21:
The commit should be in the kernel tree and latest rpi-update kernel.

closed 2020-02-22 by Fe-Pi

Dloranger, 2021-02-24:
I think there may still be an issue with this, see my comments on #3502 

#uname -a
Linux openrepeater 5.4.83-v7l+ #1379 SMP Mon Dec 14 13:11:54 GMT 2020 armv7l GNU/Linux