raspberrypi/linux #3469: WARN::dwc_otg_hcd_handle_hc_fsm:2462: Unexpected IRQ state on FSM transaction:dev_addr=4 ep=0 fsm=6, hcint=0x00000002

ramana07, 2020-02-15:
**Describe the bug**
I am getting `WARN::dwc_otg_hcd_handle_hc_fsm:2462: Unexpected IRQ state on FSM transaction:dev_addr=4 ep=0 fsm=6, hcint=0x00000002`  constantly after running some time

**To reproduce**

**Expected behaviour**
To work normal

**Actual behaviour**
Add a clear and concise description of what actually happened.

**System**
 Copy and paste the results of the raspinfo command in to this section. Alternatively, copy and paste a pastebin link, or add answers to the following questions:

* Which model of Raspberry Pi? 
` Pi3B`
* Which OS and version (`cat /etc/rpi-issue`)
```
  Raspberry Pi reference 2019-04-08
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, e910bf55ac92e27c6e77043e8995bcb5e0a2d742, stage5
```
* Which firmware version (`vcgencmd version`)
```
Aug 15 2019 12:06:42 
Copyright (c) 2012 Broadcom
version 0e6daa5106dd4164474616408e0dc24f997ffcf3 (clean) (release) (start)

```
* Which kernel version (`uname -a`)?
`Linux raspberrypi 4.19.66-v7+ #1253 SMP Thu Aug 15 11:49:46 BST 2019 armv7l GNU/Linux
`
**Logs**
If applicable, add the relevant output from `dmesg` or similar.
`Feb 15 06:51:12 raspberrypi kernel: [  257.693243] WARN::dwc_otg_hcd_handle_hc_fsm:2462: Unexpected IRQ state on FSM transaction:dev_addr=4 ep=0 fsm=6, hcint=0x00000002
Feb 15 06:51:12 raspberrypi kernel: [  257.693243] 
Feb 15 06:51:12 raspberrypi kernel: [  257.693277] WARN::dwc_otg_hcd_handle_hc_fsm:2462: Unexpected IRQ state on FSM transaction:dev_addr=4 ep=0 fsm=6, hcint=0x00000002
Feb 15 06:51:12 raspberrypi kernel: [  257.693277] 
Feb 15 06:51:12 raspberrypi kernel: [  257.693310] WARN::dwc_otg_hcd_handle_hc_fsm:2462: Unexpected IRQ state on FSM transaction:dev_addr=4 ep=0 fsm=6, hcint=0x00000002
Feb 15 06:51:12 raspberrypi kernel: [  257.693310] 
Feb 15 06:51:12 raspberrypi kernel: [  257.693341] WARN::dwc_otg_hcd_handle_hc_fsm:2462: Unexpected IRQ state on FSM transaction:dev_addr=4 ep=0 fsm=6, hcint=0x00000002
Feb 15 06:51:12 raspberrypi kernel: [  257.693341] 
Feb 15 06:51:12 raspberrypi kernel: [  257.693372] WARN::dwc_otg_hcd_handle_hc_fsm:2462: Unexpected IRQ state on FSM transaction:dev_addr=4 ep=0 fsm=6, hcint=0x00000002
Feb 15 06:51:12 raspberrypi kernel: [  257.693372] 
Feb 15 06:51:12 raspberrypi kernel: [  257.693404] WARN::dwc_otg_hcd_handle_hc_fsm:2462: Unexpected IRQ state on FSM transaction:dev_addr=4 ep=0 fsm=6, hcint=0x00000002
Feb 15 06:51:12 raspberrypi kernel: [  257.693404] 
Feb 15 06:51:12 raspberrypi kernel: [  257.693435] WARN::dwc_otg_hcd_handle_hc_fsm:2462: Unexpected IRQ state on FSM transaction:dev_addr=4 ep=0 fsm=6, hcint=0x00000002
Feb 15 06:51:12 raspberrypi kernel: [  257.693435] 
Feb 15 06:51:12 raspberrypi kernel: [  257.693467] WARN::dwc_otg_hcd_handle_hc_fsm:2462: Unexpected IRQ state on FSM transaction:dev_addr=4 ep=0 fsm=6, hcint=0x00000002
Feb 15 06:51:12 raspberrypi kernel: [  257.693467] `
**Additional context**
i am using sixfab 3g/4g base sheild  and Xbee and in other system with same setup we are getting any problem like this

P33M, 2020-02-15:
What is the full output of `sudo lsusb -v`
What is the output of `dmesg` prior to the error occurring?

ramana07, 2020-02-15:
**output of sudo lsusb -v**

```
Bus 001 Device 006: ID 2c7c:0125 Quectel Wireless Solutions Co., Ltd. EC25 LTE modem
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass          239 Miscellaneous Device
  bDeviceSubClass         2 ?
  bDeviceProtocol         1 Interface Association
  bMaxPacketSize0        64
  idVendor           0x2c7c Quectel Wireless Solutions Co., Ltd.
  idProduct          0x0125 EC25 LTE modem
  bcdDevice            3.18
  iManufacturer           1 Android
  iProduct                2 Android
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength          209
    bNumInterfaces          5
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass    255 Vendor Specific Subclass
      bInterfaceProtocol    255 Vendor Specific Protocol
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x01  EP 1 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           3
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
[... 549 lines cut ...]
[ 2585.233948] usb 1-1.2: New USB device found, idVendor=0403, idProduct=6001, bcdDevice= 6.00
[ 2585.233956] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 2585.233961] usb 1-1.2: Product: FT232R USB UART
[ 2585.233966] usb 1-1.2: Manufacturer: FTDI
[ 2585.233971] usb 1-1.2: SerialNumber: 00000000
[ 2585.281850] ftdi_sio 1-1.2:1.0: FTDI USB Serial Device converter detected
[ 2585.281950] usb 1-1.2: Detected FT232RL
[ 2585.285215] usb 1-1.2: FTDI USB Serial Device converter now attached to ttyUSB1
[ 2585.382731] usb 1-1.5: new full-speed USB device number 10 using dwc_otg
[ 2585.592770] usb 1-1.5: new high-speed USB device number 11 using dwc_otg
[ 2585.732763] usb 1-1.5: New USB device found, idVendor=2c7c, idProduct=0125, bcdDevice= 3.18
[ 2585.732778] usb 1-1.5: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[ 2585.732787] usb 1-1.5: Product: Android
[ 2585.732797] usb 1-1.5: Manufacturer: Android
[ 2585.734486] option 1-1.5:1.0: GSM modem (1-port) converter detected
[ 2585.734817] usb 1-1.5: GSM modem (1-port) converter now attached to ttyUSB2
[ 2585.735456] option 1-1.5:1.1: GSM modem (1-port) converter detected
[ 2585.735925] usb 1-1.5: GSM modem (1-port) converter now attached to ttyUSB3
[ 2585.736613] option 1-1.5:1.2: GSM modem (1-port) converter detected
[ 2585.736971] usb 1-1.5: GSM modem (1-port) converter now attached to ttyUSB4
[ 2585.737665] option 1-1.5:1.3: GSM modem (1-port) converter detected
[ 2585.737973] usb 1-1.5: GSM modem (1-port) converter now attached to ttyUSB5
[ 2585.743032] qmi_wwan 1-1.5:1.4: cdc-wdm0: USB WDM device
[ 2585.744430] qmi_wwan 1-1.5:1.4 wwan0: register 'qmi_wwan' at usb-3f980000.usb-1.5, WWAN/QMI device, 1e:0c:d2:d1:bb:c8
```

ksorathiya, 2020-05-07:
Facing a similar issue like this. A similar error pops up, after which, all the attached devices get disconnected and gets back connected after a couple of seconds.

Please help me get an idea about what this error means and what might be the cause of it.

rarealphacat, 2021-05-29:
same problem and I think my pi is dying. Cause it also reboots itself regularly.

popcornmix, 2021-06-01:
Reboots are very often caused by inadequate power supply or power supply cable.
I'd be very surprised if it's due to the pi dying.

P33M, 2021-06-01:
For the source of the original error message, there is a not-well-understood behaviour of the dwc2 hardware where it encounters an error condition and the root port enters the disconnected state, but doesn't signal a disconnect interrupt for some time. During this window, all active host channels report halt conditions without any other interrupt bits set. The host channel that's going into a halt state should specify the reason for halting, which is why the FSM complains.

The prime candidate has always been unstable power, but the exact conditions to cause *just* this issue (without e.g. rebooting the Pi or causing SD card errors) have never been found out.

gorky, 2022-09-16:
External USB 3.0 hubs, and I generally see this after 2-4 days.  This is the last message before the PI in question (with an NFS Boot) halts.