raspberrypi/linux #3473: "ifup wlanX" on RPI 4B reports DHCPDISCOVER send but tcpdump reports nothing

freekdk, 2020-02-19:
I am using the Feb 9 2020 version of full Buster.
After configuring the wifi interfaces, the build-in and a USB dongle,
after giving the command "ifup wlan1" I got the following:
```
Internet Systems Consortium DHCP Client 4.4.1
Copyright 2004-2018 Internet Systems Consortium.
All rights reserved.
For info, please visit https://www.isc.org/software/dhcp/

Listening on LPF/wlan1/00:0b:81:93:de:5c
Sending on   LPF/wlan1/00:0b:81:93:de:5c
Sending on   Socket/fallback
DHCPDISCOVER on wlan1 to 255.255.255.255 port 67 interval 7
DHCPDISCOVER on wlan1 to 255.255.255.255 port 67 interval 19
DHCPDISCOVER on wlan1 to 255.255.255.255 port 67 interval 9
DHCPDISCOVER on wlan1 to 255.255.255.255 port 67 interval 9
DHCPDISCOVER on wlan1 to 255.255.255.255 port 67 interval 12
DHCPDISCOVER on wlan1 to 255.255.255.255 port 67 interval 5
No DHCPOFFERS received.
No working leases in persistent database - sleeping.
```
During the above output in another console I gave the command "tcpdump -v -i wlan1" and got not a single packet. 0 packets send/received.

**Expected behaviour**
I would expect at least the send packets (DHCPDISCOVER) to be catched.

**Actual behaviour**
See above.

I assume that DHCPDISCOVER will only be send when the verification process between the system and the AP has been finished. So traffic must have been exchanged before.

SOI2016, 2020-02-21:
can you comment wpa config from /etc/network/interfaces file as shown in the below and check one more time.

wpa-ssid
wpa-psk

freekdk, 2020-02-21:
I am sure the values of these parameters are OK. On very rare occasions the DHCPDISCOVER gets answered and tcpdump saw packets. See https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=264485

freekdk, 2020-02-21:
Another observation is that "iwlist wlanX scan" lists the AP I want to connect to with a Signal strength that is quite acceptable.