raspberrypi/linux #3477: rpi-5.6.y cross compile on x86 arch linux errors out

danieljordan10, 2020-02-21:
**Describe the bug**
A cross-compile of rpi-5.6.y with an unmodified bcmrpi3_defconfig hits errors in video and sound drivers.

I'm building on Arch Linux (please see below).  Is that a supported environment?  If not, where should I build?

Modified the below output to play well with the text formatting.

**To reproduce**
$ uname -a
Linux parnassus 5.5.3-arch1-1 #1 SMP PREEMPT Tue, 11 Feb 2020 15:35:41 +0000 x86_64 GNU/Linux
$ git remote -v |grep rpi
rpi	https://github.com/raspberrypi/linux.git (fetch)
rpi	https://github.com/raspberrypi/linux.git (push)
$ git fetch rpi rpi-5.6.y
From https://github.com/raspberrypi/linux
   branch                      rpi-5.6.y  -> FETCH_HEAD
$ git branch rpi-5.6.y FETCH_HEAD
$ git checkout rpi-5.6.y
Updating files: 100% (18546/18546), done.
Switched to branch 'rpi-5.6.y'
$ git log --oneline -1
eb83c8958784 (HEAD -> rpi-5.6.y, rpi/rpi-5.6.y) Revert "Revert "pinctrl: bcm2835: Pass irqchip when adding gpiochip""
$ make ARCH=arm64 CROSS_COMPILE=aarch64-linux-gnu- bcmrpi3_defconfig
  HOSTCC  scripts/basic/fixdep
  HOSTCC  scripts/kconfig/conf.o
  HOSTCC  scripts/kconfig/confdata.o
  HOSTCC  scripts/kconfig/expr.o
  LEX     scripts/kconfig/lexer.lex.c
  YACC    scripts/kconfig/parser.tab.[ch]
  HOSTCC  scripts/kconfig/lexer.lex.o
  HOSTCC  scripts/kconfig/parser.tab.o
  HOSTCC  scripts/kconfig/preprocess.o
  HOSTCC  scripts/kconfig/symbol.o
  HOSTCC  scripts/kconfig/util.o
  HOSTLD  scripts/kconfig/conf
 .
 . configuration written to .config
 .
$ make ARCH=arm64 CROSS_COMPILE=aarch64-linux-gnu- -j8 Image modules dtbs
  WRAP    arch/arm64/include/generated/uapi/asm/kvm_para.h
  WRAP    arch/arm64/include/generated/uapi/asm/errno.h
  WRAP    arch/arm64/include/generated/uapi/asm/ioctl.h
<snip!>
drivers/video/fbdev/bcm2708_fb.c: In function 'bcm2708_compat_ioctl':
drivers/video/fbdev/bcm2708_fb.c:790:15: error: implicit declaration of function 'compat_ptr'; did you mean 'compat_lr'? [-Werror=implicit-function-declaration]
  790 |   param.dst = compat_ptr(param32.dst);
      |               ^~~~~~~~~~
      |               compat_lr
drivers/video/fbdev/bcm2708_fb.c:790:13: warning: assignment to 'void *' from 'int' makes pointer from integer without a cast [-Wint-conversion]
  790 |   param.dst = compat_ptr(param32.dst);
      |             ^
<snip!>
sound/soc/bcm/iqaudio-codec.c: In function 'snd_rpi_iqaudio_pll_control':
sound/soc/bcm/iqaudio-codec.c:40:50: error: passing argument 2 of 'snd_soc_get_pcm_runtime' from incompatible pointer type [-Werror=incompatible-pointer-types]
   40 |   snd_soc_get_pcm_runtime(card, card->dai_link[0].name);
      |                                 ~~~~~~~~~~~~~~~~~^~~~~
      |                                                  |
      |                                                  const char *
In file included from sound/soc/bcm/iqaudio-codec.c:24:
./include/sound/soc.h:467:29: note: expected 'struct snd_soc_dai_link *' but argument is of type 'const char *'
  467 | struct snd_soc_pcm_runtime *snd_soc_get_pcm_runtime(struct snd_soc_card *card,
      |                             ^~~~~~~~~~~~~~~~~~~~~~~

**Expected behaviour**
A clean compile.

**Actual behaviour**
A compile which errors out.

**Additional context**
My final goal is to reproduce a kernel warning reported on an upstream linux mailing list.

6by9, 2020-02-21:
Which compiler are you using? I have an open PR to update our docs on the toolchain to use
https://github.com/raspberrypi/documentation/pull/1399

6by9, 2020-02-21:
Not that I think that is your particular issue here.

danieljordan10, 2020-02-21:
Hi, thanks for looking at this, @6by9.  This one:

$ pacman -Qi aarch64-linux-gnu-gcc
Name            : aarch64-linux-gnu-gcc
Version         : 9.2.0-1
Description     : The GNU Compiler Collection - cross compiler for ARM64 target
Architecture    : x86_64
URL             : https://gcc.gnu.org/
Licenses        : GPL  LGPL  FDL
Groups          : None
Provides        : None
Depends On      : aarch64-linux-gnu-binutils  aarch64-linux-gnu-glibc  libmpc  zlib
Optional Deps   : None
Required By     : aarch64-linux-gnu-glibc
...

I've now stumbled on the documentation (funny how it happens after you post the issue) and tried with arm-bcm2708/gcc-linaro-arm-linux-gnueabihf-raspbian-x64/bin from raspberrypi/tools, also to no avail:

$ make distclean
...
$ KERNEL=kernel7
$ make ARCH=arm CROSS_COMPILE=arm-linux-gnueabihf- bcm2709_defconfig
...
$ make -j8 ARCH=arm CROSS_COMPILE=arm-linux-gnueabihf- zImage modules dtbs
  SYSHDR  arch/arm/include/generated/uapi/asm/unistd-oabi.h
  SYSHDR  arch/arm/include/generated/uapi/asm/unistd-common.h
  SYSHDR  arch/arm/include/generated/uapi/asm/unistd-eabi.h
...
In file included from scripts/gcc-plugins/gcc-common.h:22,
                 from scripts/gcc-plugins/arm_ssp_per_task_plugin.c:3:
scripts/gcc-plugins/arm_ssp_per_task_plugin.c: In function ‘unsigned int arm_pertask_ssp_rtl_execute()’:
scripts/gcc-plugins/arm_ssp_per_task_plugin.c:36:18: error: ‘sext_hwi’ was not declared in this scope
   36 |   mask = GEN_INT(sext_hwi(sp_mask, GET_MODE_PRECISION(Pmode)));
      |                  ^~~~~~~~
/home/dmjordan/src/rpi-tools/arm-bcm2708/gcc-linaro-arm-linux-gnueabihf-raspbian-x64/bin/../lib/gcc/arm-linux-gnueabihf/4.8.3/plugin/include/rtl.h:2363:51: note: in definition of macro ‘GEN_INT’
 2363 | #define GEN_INT(N)  gen_rtx_CONST_INT (VOIDmode, (N))
      |                                                   ^
  DTCO    arch/arm/boot/dts/overlays/disable-wifi.dtbo
make[2]: *** [scripts/Makefile.host:158: scripts/gcc-plugins/arm_ssp_per_task_plugin.o] Error 1
  DTCO    arch/arm/boot/dts/overlays/dpi18.dtbo
make[1]: *** [scripts/Makefile.build:505: scripts/gcc-plugins] Error 2

By the way, I'm targeting a raspberry pi 3b+.

danieljordan10, 2020-02-21:
Oh, I'll go look at that PR now.

danieljordan10, 2020-02-21:
Yep, changing my PATH according to that PR fixed the build.  Thanks, @6by9!

closed 2020-02-21 by danieljordan10