raspberrypi/linux #3478: bcm2385 broken on pi3 with 5.4.y

anarchpenguin, 2020-02-21:
[  372.937800] bcm2835_audio bcm2835_audio: Failed to get DT property 'brcm,pwm-channels'
[  372.937844] bcm2835_audio: probe of bcm2835_audio failed with error -22

This was working fine with 5.4.3 after upgrading to 5.4.18 audio is no longer functional.

pelwell, 2020-02-22:
Have you also updated the DTBs using the 5.4.18 source? The brcm,pwm-channels property has moved.

anarchpenguin, 2020-02-22:
> Have you also updated the DTBs using the 5.4.18 source? The brcm,pwm-channels property has moved.

Yes the DTB is from 5.4.18 as well.

anarchpenguin, 2020-02-25:
Just went ahead and retested every with the 5.4.21 it shows same results. I make sure the DTB was updated as well.

pelwell, 2020-02-25:
It turns out that I bodged moving the DT audio properties on devices other than the Pi 4 - the node name has to match the module name, i.e. `bcm2835_audio` rather than `audio`. With that fixed (in rpi-5.4.y - rpi-5.5.y is in progress) it is working again.

anarchpenguin, 2020-02-26:
> bcm2835_audio

I appreciate you looking into this and finding the issue. Sorry I am not up to speed on device tree layout and all the workings. I will have to make some time to understand them for the future.

anarchpenguin, 2020-02-27:
Other then having to load the module manually now it does work as expected with 5.4.22 so I am gonna close. Thanks for your work is much appreciated.

closed 2020-02-27 by anarchpenguin