raspberrypi/linux #3480: memcpy performance issue

zoff99, 2020-02-23:
**Describe the bug**
memcpy take a long time (see example program)

**To reproduce**
make a && ./a

**Expected behaviour**
hopfully be faster

**Actual behaviour**
takes up to 20ms on a rpi4

**System**
 Copy and paste the results of the raspinfo command in to this section. Alternatively, copy and paste a pastebin link, or add answers to the following questions:

* Which model of Raspberry Pi?
rpi4
* Which OS and version (`cat /etc/rpi-issue`)?
Raspberry Pi reference 2020-02-14
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, e577677b623b577f2a0ec7cfaffc3c27da005da3, stage2
* Which firmware version (`vcgencmd version`)?
* Which kernel version (`uname -a`)?

**Additional context**
```
#include <time.h>
#include <stdio.h>
#include <sys/time.h>
#include <stdlib.h>
#include <string.h>

static void __utimer_start(struct timeval *tm1)
{
    gettimeofday(tm1, NULL);
}

static unsigned long long __utimer_stop(struct timeval *tm1)
{
    struct timeval tm2;
    gettimeofday(&tm2, NULL);
    unsigned long long t = 1000 * (tm2.tv_sec - tm1->tv_sec) + (tm2.tv_usec - tm1->tv_usec) / 1000;

    printf("%llu ms\n", t);
    return t;
}

int main()
{
    int size1 = 30000000;
    int size2 = 8*1024*1024;

    char *buf = calloc(1, size1);
    char *buf2 = calloc(1, size1);

    struct timeval tm_01_007;
    __utimer_start(&tm_01_007);

    memcpy(buf, buf2, size2);
    memcpy(buf, buf2, size2);
    memcpy(buf, buf2, size2);

    long long timspan_in_ms;
    timspan_in_ms = __utimer_stop(&tm_01_007);

    free(buf);

    return 0;
}
```

save as a.c

JamesH65, 2020-02-23:
I believe that these functions are already highly optimized. Why do you think they can be improved?

pelwell, 2020-02-23:
The default malloc alignment ought to be sensible, but feel free to align it further (256 bytes should be more then enough).

1. Have you set arm_freq in config.txt?
2. Have you enabled the Performance governor? If not, performance will be reduced. Alternatively, add `force_turbo=1` to config.txt.
3. What displays are attached (and with what resolution and frame rate)?
4. Is there any other activity occurring at the time?

zoff99, 2020-02-23:
@JamesH65 because it seems to slow. rpi4 should be able to memcpy some GB/s

JamesH65, 2020-02-23:
Are you sure that your timer functions are accurate - I'm not familiar with __utimer_start and google seems to think it an android function. Might be worth trying stuff from this page.https://stackoverflow.com/questions/6749621/how-to-create-a-high-resolution-timer-in-linux-to-measure-program-performance

pelwell, 2020-02-23:
__utimer_start  is simply a wrapper around gettimeofday, which is a POSIX function.

I think it's important to establish the clock speeds, and any memory bandwidth taken up by the display, etc.

zoff99, 2020-02-24:
the function is included in the source attached here, its just a wrapper.
with a timespan of 30ms gettimeofday is accurate enough.

can somebody try the attached source on their rpi and just post the result?

JamesH65, 2020-02-24:
Tried it, increase the number of copies to 10, and it completed in 34ms, so 3.4ms per copy.

pelwell, 2020-02-24:
With performance governor enabled (i.e. with the ARMs at 1.5GHz) I get a fairly consistent 3.3GB/s bandwidth. The powersave governor (600MHz) drops performance to 1.46GB/s. On-demand will yield a result somewhere between the two.

JamesH65, 2020-02-24:
Which is about what I was seeing (approx 2.5GBits/s)

pelwell, 2020-02-24:
GB/s is Giga Bytes per second.

N.B. My results were with the screen blanked - an active display will eat into that bandwidth (1080p 60-70 drops the bandwidth to about 3.1GB/s.

JamesH65, 2020-02-24:
Apologies, I've use the wrong units - that should read 2.5GBytes/s.

pelwell, 2020-02-24:
N.B.2. My figures are for reading _and_ writing, so you could naively double the results. In practise, read and write speeds are different to two separate figures is more useful.

zoff99, 2020-02-24:
thanks guys for your results. i will try to change to the performance governor

pelwell, 2020-02-24:
```
$ sudo sh -c "echo performance > /sys/devices/system/cpu/cpufreq/policy0/scaling_governor"
```
should do it.

AllNamesRTaken, 2021-08-09:
A year later since it has not been closed I found this when researching why the memory bandwidth is resulting in bad webGL/OpenGl performance for large windows. I.E. it scales down dramatically as the window grows even though a 1080p swap should only result in about 0.5 GB/s and the memory bandwidth should be 4-5GB/s

Maybe you could boot your PI in console only and rerun the test?