raspberrypi/linux #3481: snd_soc_hifiberry_dacplus broken in 5.4.21

fichtennadel, 2020-02-23:
**Describe the bug**
Using kernel 5.4.21 module snd_soc_hifiberry_dacplus fails to initialize the hifiberry hardware.

**To reproduce**
Raspi 3 with hifiberry DAC+ , boot kernel 5.4.21 (not ok) or 5.4.11 (ok)

**Expected behaviour**
module loads, initializes hifiberry DAC+,  `aplay -l` lists `sndrpihifiberry [snd_rpi_hifiberry_dacplus]`
`dmesg | grep hifi` shows
```
[    6.829712] snd-rpi-hifiberry-dacplus soc: sound: ASoC: failed to init link HiFiBerry DAC+: -517
[    7.112002] snd-rpi-hifiberry-dacplus soc: sound: ASoC: failed to init link HiFiBerry DAC+: -517
[    7.429348] snd-rpi-hifiberry-dacplus soc: sound: ASoC: failed to init link HiFiBerry DAC+: -517
[    7.632537] snd-rpi-hifiberry-dacplus soc: sound: pcm512x-hifi <-> 3f203000.i2s mapping ok
```
**Actual behaviour**
module loads but fails to initialize hifiberry DAC+,  therefore `aplay -l` lists `aplay: device_list:272: no soundcards found...`
`dmesg | grep hifi` shows
```
[    6.829712] snd-rpi-hifiberry-dacplus soc: sound: ASoC: failed to init link HiFiBerry DAC+: -517
[    7.112002] snd-rpi-hifiberry-dacplus soc: sound: ASoC: failed to init link HiFiBerry DAC+: -517
[    7.429348] snd-rpi-hifiberry-dacplus soc: sound: ASoC: failed to init link HiFiBerry DAC+: -517
```
Please note the missing last line, this means the driver failed to load or initialize the hifiberry

Last working kernel version was 5.4.11 , can be reproduced by down/upgrading (see links to forum discusssion below)

**Logs**
dmesg see https://paste.kodi.tv/sasuxiqure 

**Additional info**
see also discussions on
https://forum.kodi.tv/showthread.php?tid=343068&pid=2926110#pid2926110
and
https://support.hifiberry.com/hc/en-us/community/posts/360009449558-module-snd-soc-hifiberry-dacplus-broken-in-kernel-5-4-18

popcornmix, 2020-02-24:
I think we want this: https://patchwork.kernel.org/patch/11394967/ from @HiassofT

HiassofT, 2020-02-24:
It shouldn't hurt to include this patch until it dribbles down into stable trees, it'll at least avoid the warning and splat in regulator core.

The real issue with the hifiberry dacplus card seems to be though that it's (downstream) clock driver isn't loaded/loading (the warning/splat is just a consequence of that) - see eg lsmod output in the dmesg info linked above.

Most likely the clock driver and/or DT overlay need some changes after the clocks DT changes in https://github.com/raspberrypi/linux/pull/3385/commits/3292779df6217ae59445aed447bdfd7d5719f275

pelwell, 2020-02-27:
By removing "simple-bus", the clock DT change broke the instantiation of its children. I've fixed it with a partial reversion.

pelwell, 2020-02-28:
@lategoodbye I now understand what is happening, and I owe you an apology. of_clk_init (called early in the boot sequence) searches the tree for all compatible strings for drivers declared with CLK_OF_DECLARE, including clk_fixed_rate. This is why your change is fine for upstream DTBs.

AFAIK the HifiBerry clock driver can't make use of CLK_OF_DECLARE because it's a module, so it does have to rely on something else causing its driver to be instantiated. If we say that `/clocks` is only for static system clocks then the overlays that declare custom clocks have to be changed to put the clock node somewhere that will be probed - either in the root or under one of the other  `simple-bus` nodes; as an overlay it shouldn't bother any of the kernel checkers.

fichtennadel, 2020-03-01:
Builds over at https://forum.kodi.tv/showthread.php?tid=343068&pid=2928218#pid2928218 confirm the fix, hifiberry dacplus card initializes successfully now.

popcornmix, 2020-05-19:
Closing as issue is fixed.

closed 2020-05-19 by popcornmix