raspberrypi/linux #3482: USB won't seen as ttyAMC1

poudenes, 2020-02-24:
Hi All,

I have a Z-Wave USB Stick working for months perfect but suddenly its not shown anymore as /dev/ttyAMC1

I see this error:

```
[ 1113.122934] usb 1-1.4.4: new full-speed USB device number 15 using xhci_hcd
[ 1128.363469] usb 1-1.4.4: device descriptor read/64, error -110
[ 1143.723781] usb 1-1.4.4: device descriptor read/64, error -110
[ 1143.943602] usb 1-1.4.4: new full-speed USB device number 16 using xhci_hcd
[ 1159.164078] usb 1-1.4.4: device descriptor read/64, error -110
[ 1174.524394] usb 1-1.4.4: device descriptor read/64, error -110
[ 1174.644354] usb 1-1.4-port4: attempt power cycle
[ 1175.584216] usb 1-1.4.4: new full-speed USB device number 17 using xhci_hcd
[ 1176.210467] xhci_hcd 0000:01:00.0: Setup ERROR: setup address command for slot 5.
[ 1176.424337] xhci_hcd 0000:01:00.0: Setup ERROR: setup address command for slot 5.
[ 1176.644245] usb 1-1.4.4: device not accepting address 17, error -22
[ 1176.744259] usb 1-1.4.4: new full-speed USB device number 18 using xhci_hcd
[ 1176.744369] xhci_hcd 0000:01:00.0: Setup ERROR: setup address command for slot 5.
[ 1176.964321] xhci_hcd 0000:01:00.0: Setup ERROR: setup address command for slot 5.
[ 1177.184257] usb 1-1.4.4: device not accepting address 18, error -22
[ 1177.184460] usb 1-1.4-port4: unable to enumerate USB device
```

What can I do?

pelwell, 2020-02-24:
Are you attaching the Z-Wave through a hub? There is a known problem with the device that makes it appear like a USB3 device when in fact it is a USB2 device.

How is the Pi (I assume that's what you are using) powered? What peripherals are attached, and are they powered?

poudenes, 2020-02-24:
> Are you attaching the Z-Wave through a hub? There is a known problem with the device that makes it appear like a USB3 device when in fact it is a USB2 device.
> 
> How is the Pi (I assume that's what you are using) powered? What peripherals are attached, and are they powered?

Hi @pelwell , I attached a usb hub simple small thing. Everything worked ok. It’s not a powered one. Pi powered to a original adapter.. what I will try is reinstall Debian buster clear and see what happen. Will that work?

closed 2022-10-24 by poudenes