raspberrypi/linux #3484: bcm2835-v4l2: codec has invalid maximum frame size and no 4k support

moorthy-bs, 2020-02-26:
Invalid maximum frame height has been given in the [bcm2835-v4l2-codec](https://github.com/raspberrypi/linux/blob/rpi-4.19.y/drivers/staging/vc04_services/bcm2835-codec/bcm2835-v4l2-codec.c#L94-L95). Also whether this blocks for 4k support?

**To reproduce**
build with meta-raspberrypi master branch
MACHINE=raspberrypi4
BSP = 4.19.102, branch=rpi-4.19.y

**Expected behaviour**
4k video has to play

**Actual behaviour**
only video till have full hd (1920x1080) played. If any video higher to this resolution or 4k video played, the maximum frame size reports as 1920x1920.

**System**
 Copy and paste the results of the raspinfo command in to this section. Alternatively, copy and paste a pastebin link, or add answers to the following questions:

* Which model of Raspberry Pi? e.g. Pi3B+, PiZeroW
Pi4

* Which OS and version (`cat /etc/rpi-issue`)?
No file available

* Which firmware version (`vcgencmd version`)?
Jan 30 2020 13:50:45
Copyright (c) 2012 Broadcom

version 8f792e011c8a0ea3c5d47e7cc10172cc10b93c09 (clean) (release) (start)

* Which kernel version (`uname -a`)?
Linux raspberrypi4 4.19.102-v7l #1 SMP Wed Feb 19 09:41:09 UTC 2020 armv7l GNU/Linux

**Logs**
westerossink-ioctl: ioct( 8, c02c564a ( VIDIOC_ENUM_FRAMESIZES ) )
westerossink-ioctl: ioct( 8, c02c564a ) rc 0
westerossink-ioctl: fmt 34363248 idx 0 type 3
westerossink-ioctl: stepwise minw 32 maxw 1920 stepw 1 minh 32 maxh 1920 steph 1                                                                                       
westeros-sink: max frame (1920x1920)

6by9, 2020-02-26:
4k video decode is only supported through the HEVC decoder.
The V4L2 stateless API driver for that is being finalised at present, but there is modified version of FFmpeg that drives the hardware from userspace in Raspbian.

The H264 block is the same as earlier Pi versions, and was originally specified as level 4.1 (8192 macroblocks/frame, and 245760 macroblocks/second for 1080p30), although level 4.2 (522240 macroblocks/second for 1080p60) can generally be achieved except on very complex streams.

The block doesn't care whether the frame is 1920x1080, or 1080x1920.
I suspect that it will even accept 1920x1920 if you define the level appropriately, but haven't tried it. We don't enforce the level limit as we can generally decode things above that bitrate and/or framerate, but not in realtime. It means you can offline transcode clips.

V4L2 allows you to advertise either discrete, stepwise, or continuous resolutions via enum_framesizes. To encompass both 1920x1080 and 1080x1920 we declare stepwise with a max of 1920x1920. I don't believe there is an alternate way to describe the provided functionality.

If you're really getting issues, then please provide a sample video stream so we can investigate.

moorthy-bs, 2020-02-26:
The test case is playing a 4k video on v4l2. When we try to play 4k stream, we got blocking that the log shows maximum of 1920x1920 size. So really v4l2 support 4k either (4kp30 / 4kp60)?

JamesH65, 2020-02-26:
No. Maximum is 1920x1920 (that is the maximum the HW H264 decoder can handle).

The HEVC (H265) decoder for V4L2 is not yet ready.

moorthy-bs, 2020-02-26:
oh okay. Is there any tentative date of arrival of HEVC (H265) decoder for v4l2?

6by9, 2020-02-26:
What software are you using to decode, and does it support the V4L2 stateless decoder API natively?
The API for HEVC hasn't been finalised in the kernel as yet, so this is all fairly cutting edge for Linux as a whole. We're trying to fix the issues with it via LibreElec and Kodi first, rather than Raspbian. I'd hope within a month.

The main application that supports that the stateless API is FFmpeg, and the [Raspbian packaged version](https://github.com/RPi-Distro/ffmpeg) includes being able to talk direct to the HEVC hardware via the registers being mmaped.
Use that for now to talk to the HEVC block via the memory mapped registers, and I believe you should be able to switch over to the V4L2 API when available.

(The H264 decoder uses the V4L2 stateful decoder API, which is far simpler, and more widely supported. It does not match up with the way the HEVC hardware is designed though, whilst the stateless API does).

balav08, 2020-03-03:
Understood that rpi-4.19.35 doesn't have HEVC(H265) support, is there any possibilities whether rpi-5.4.y or rpi-5.6.y has this support? if so, when can we expect an official release of the same?

moorthy-bs, 2020-03-03:
> What software are you using to decode, and does it support the V4L2 stateless decoder API natively?
> The API for HEVC hasn't been finalised in the kernel as yet, so this is all fairly cutting edge [...]

We 've wayland based Linux environment (weston / westeros compositor) and mainly a [v4l2 test application](https://github.com/rdkcmf/v4l2test) is used to decode which supports v4l2 driver APIs and not using raspbian

6by9, 2020-03-03:
That test app appears to support the [V4L2 stateful decoder API](https://linuxtv.org/downloads/v4l-dvb-apis-new/uapi/v4l/dev-decoder.html) only, not the [stateless one](https://linuxtv.org/downloads/v4l-dvb-apis-new/uapi/v4l/dev-stateless-decoder.html). It will therefore not work with the HEVC decoder, but should work with the existing H264 decoder (dev/video10), and up to level 4.2 (1080p60).

6by9, 2020-03-03:
I'm working on cleaning up the patches and backporting them to 5.4 (the next LTS release). Hopefully within the week.
As above, note that it is implementing the V4L2 stateless decoder API, and will be cherry-picking back some bits that haven't been accepted for 5.7 yet, let alone actually hit a mainline release.

6by9, 2020-03-03:
That test app also appears to assume the world is NV12 only, when we support YU12, YV12, NV12, and NV21.

We also support the multi-planar API but using the single planar formats. That is legal within the V4L2 API and was the recommended solution on the linux-media list. However I don't think it's going to play nicely with the output (V4L2 capture) queue setup in your app.

balav08, 2020-06-11:
Hi, @6by9  we have moved to linux kernel-5.4 version from 4.19 and tried to play 4K video with rpi4 platform. we are facing the same issue as we mentioned earlier that maximum resolution got restricted to 1920.
Is that the 4k support is still not available in linux kernel-5.4 ?
If the support is not available when can we expect the supported kernel version?

6by9, 2020-06-11:
4k H264? That's never going to be supported as the hardware can't do it. (Already stated that earlier in this thread). bcm2835-codec states a max of 1920x1920 and that is never going to change.

4k HEVC is supported, and the V4L2 stateless codec driver is in the tree - https://github.com/raspberrypi/linux/tree/rpi-5.4.y/drivers/staging/media/rpivid.
The version of FFmpeg shipped with RPiOS supports the HEVC stateless codec API, which is how come VLC can play HEVC material.
The kernel APIs for HEVC stateless codec drivers haven't been finalised, therefore the implementation may change in future to match upstream.