raspberrypi/linux #3485: Pi4 crash when using 2nd USB3 device when booting from USB3 SSD

poliboy, 2020-02-26:
I use the Pi4 booting with a USB3 connected SSD.
This works nicely, but when I connect in addition an external USB3 HDD to the USB3 port, then the system crashes with "IO errors". Any commands like "dmesg" are not available any longer. 
Connection of the HDD to USB2 port works. Booting from SD card and connecting the USB3 SSD and the USB3 HDD at the same time works.

**Actual behaviour**
The GUI disappears (only background is visible). Some "IO errors" are indicated.

**System**
System Information
------------------

Raspberry Pi 4 Model B Rev 1.1
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"

Raspberry Pi reference 2019-09-26
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 80d486687ea77d31fc
3fc13cf3a2f8b464e129be, stage5

Linux toni4 4.19.97-v7l+ #1294 SMP Thu Jan 30 13:21:14 GMT 2020 armv7l GNU/Linux
Revision	: c03111
Serial		: 100000000824fe4c
Model		: Raspberry Pi 4 Model B Rev 1.1
Throttled flag  : throttled=0x0
Camera          : supported=0 detected=0

* Which model of Raspberry Pi? Pi4

* Which OS and version (`cat /etc/rpi-issue`)?
Raspberry Pi reference 2019-09-26
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 80d486687ea77d31fc3fc13cf3a2f8b464e129be, stage5

* Which firmware version (`vcgencmd version`)?
Feb 12 2020 12:36:21 
Copyright (c) 2012 Broadcom
version c3c8dbdf147686fb0c3f32aece709d0653368810 (clean) (release) (start)

* Which kernel version (`uname -a`)?
Linux toni4 4.19.97-v7l+ #1294 SMP Thu Jan 30 13:21:14 GMT 2020 armv7l GNU/Linux

**Logs**
unfortunately demsg is not available any longer

**Additional context**
Add any other relevant context for the problem.

poliboy, 2020-02-27:
Good point. The SSD and its USB3 interface are bus powered and working standalone when booting from it. The additional USB3 HDD has a power supply attached. My expectation is, that there is not more current drawn from USB3 port when plugging in a powered device.
I tried with a powered USB3 hub and both disks worked also in USB3 boot mode. I assume that the driver software used is a different one, because only one USB3 port from the Pi4 is used. Unfortunately I can't check with a non-powered USB3 hub. 
Therefore: I'm pretty sure that there is no issue with the power consumption, but full evidence is missing.

popcornmix, 2020-02-28:
> The additional USB3 HDD has a power supply attached. My expectation is, that there is not more current drawn from USB3 port when plugging in a powered device.

I don't think that is a valid expectation. Often the power supply to a drive just supplies, say 12V for the motor. There may be significant 5V still drawn from the Pi over USB.

poliboy, 2020-02-29:
Thanks for your replies so far.
In order to remove all doubts, I purchased a passive USB3 hub (therefore my answer took some time).
The result: If I attach the passive hub to one USB3 port of the Pi4, and the SSD to the passive hub, then it works (as expected). If I then attach the HDD to the passive hub, it works as well! 
If I instead attach the HDD to the 2nd USB3 port of the Pi4, then the crash happens. This is reproducible also with another HDD (I used Seagate and WD).
I consider this as evidence that the problem is not the too high current drawn from the Pi4.
Some other experiment: I booted with SSD directly in USB3 port of Pi4, and after plugging in the passive USB hub, then the crash happened. Nothing was attached to the passive hub.
Let me know if I could support the bug tracing by some SW analytics...

tiagoboldt, 2020-02-29:
I don't think there's any bug here to be solved. It's an electronic limitation. When you draw more current then the pi can provide, it restarts. No way around that. Maybe this should be better documented -- this issue might do just that for those who have the same problem in the future. 

I guess this can be closed, but I'll leave that feedback to @popcornmix who's the expert here.

poliboy, 2020-02-29:
Maybe my formulation was too complicated: I don't draw more current than the Pi can provide... Otherwise I would see the same issue with the passive hub, which is not the case.

P33M, 2020-03-16:
Plug an optical mouse into one of the spare USB ports. When the crash occurs, does the LED on the bottom of the mouse turn off?

poliboy, 2020-03-22:
Thanks for the hint.
I attached this: 
Bottom USB3: SSD
Bottom USB2: optical mouse
After booting, everything was fine.
Then I plugged the SSD (actively supplied) into the Top USB3.
The crash occurred, but the optical mouse still was working perfectly and the LED was still on.

poliboy, 2020-03-22:
I did another experiment, maybe this refers better to your question:
Starting with
Bottom USB3: SSD
I booted without mouse and keyboard.
Then I plugged the optical mouse into the Top USB3. The crash happened, the mouse LED was on, but the mouse didn't work.

P33M, 2020-03-25:
Do you have a powered USB3.0 hub, and if so, what is the behaviour with one or both SSDs plugged into the hub?

poliboy, 2020-03-28:
I have a powered USB3 hub (combined with a HDD). 
If I plug this into the Pi4 instead of the passive hub, and attach the drives to the powered hub, then everything works fine.

I did some more investigations on the issue and found out, that a message appears at the boot screen, which was not visible in dmesg. It took several trials to take a photo at the right point in time.
The message said:
"Failed to start persistent change of wireless mouse interrupt to avoid conflict with USB3 SSD. See 'systemctl status mouse_irq.service' for details."
The message didn't disappear when I used a wired mouse and keyboard.

The output of the mentioned "systemctl" command was:
```
● mouse_irq.service - persistent change of wireless mouse interrupt to avoid conflict with USB3 SSD
   Loaded: loaded (/etc/systemd/system/mouse_irq.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Sat 2020-03-28 03:32:50 CET; 8h ago
  Process: 519 ExecStart=/bin/bash -c echo 2 | sudo tee /proc/irq/55/smp_affinity (code=exited, status=1/FAILURE)
 Main PID: 519 (code=exited, status=1/FAILURE)

Mär 28 03:32:49 toni4 systemd[1]: Starting persistent change of wireless mouse interrupt to avoid conflict with USB3 SSD...
Mär 28 03:32:50 toni4 sudo[523]:     root : TTY=unknown ; PWD=/ ; USER=root ; COMMAND=/usr/bin/tee /proc/irq/55/smp_affinity
Mär 28 03:32:50 toni4 sudo[523]: pam_unix(sudo:session): session opened for user root by (uid=0)
Mär 28 03:32:50 toni4 bash[519]: tee: /proc/irq/55/smp_affinity: Datei oder Verzeichnis nicht gefunden
Mär 28 03:32:50 toni4 bash[519]: 2
Mär 28 03:32:50 toni4 sudo[523]: pam_unix(sudo:session): session closed for user root
Mär 28 03:32:50 toni4 systemd[1]: mouse_irq.service: Main process exited, code=exited, status=1/FAILURE
Mär 28 03:32:50 toni4 systemd[1]: mouse_irq.service: Failed with result 'exit-code'.
Mär 28 03:32:50 toni4 systemd[1]: Failed to start persistent change of wireless mouse interrupt to avoid conflict with USB3 SSD.
```

"Datei oder Verzeichnis nicht gefunden" is translated: "File or directory not found".

Maybe this is related to the issue I described at the beginning and you know what's going on there...

akishima2, 2021-02-18:
I'm seeing exactly the same issue. The issue didn't exist for many weeks when I was running an older version of Buster imaged from the NOOBS install. I re-imaged my drive with the latest version of Buster using the Raspberry Pi Imager a few days ago and since then this issue has been persistent.

m-t-smith, 2021-03-08:
Hey Poliboy,

Did you ever determine the exact issue? I experienced a similar problem; I booted and ran an HP laptop from a USB3 attached external SSD Ubuntu 20.04 installation. This setup worked fine until I plugged a second SSD into the other USB3 port, at which point Ubuntu crashed and I got a black screen full of a rapidly repeating I/O error. If this is the same problem, I'm glad to hear that using a USB2 port for one of the drives or booting from an SD card works (though I'm unsure is this means you're booting the OS on your SSD with a bootloader on your SD card, or booting fully from the SD card and leaving the SSD-based OS to languish.).

I plan to use the SSD-based OS on a pi4, but have you tried recreating the issue on any other type of computer? I share your suspicion that something besides power is causing the problem, though that's probably usually a fair bet for a pi, and hence why everyone seems so hung up on it even after you ran both through a passive hub. My gut feeling is that the problem lies with the usb3 driver (probably UAS) integrated into the Linux kernel. USB3s cause interference; I'm guessing part of the solution to that was a bit of a hack job that involved blocking all/the other USB3 ports temporarily whenever something new was plugged into one. That would obviously be a problem for this kind of setup, because it cuts of the OS, so the interrupt handler is never reached and the request just keeps running because it's supposed to be non-blocking.

That's just my guess, would like to hear if you or anyone else figured it out.

clssn, 2021-04-06:
Hi everyone,
looks like I have a very similar issue, though I don't boot from the SSD (only have `/var` mounted from one of them, which makes a crash equally destructive). I initially thought it was power related but after some weeks of trying around similar things as @poliboy power doesn't appear to be the problem. In my setup the crash happens after a couple of hours of flawless operation which makes it harder to inspect, together with journal disappearing with `/var`. 
For now, I just wanted to confirm the problem. I'll check next, whether using only the bottom USB3 port works for me as well, and whether by plugging in arbitrary devices in the top port I can provoke the crash.
@m-t-smith, according to what you tell, it's about the point in time of connection, while I don't connect anything new when the crash happens. 

Good to know I'm not alone, though it's a bit quiet in here while I can't believe we're the only ones with that particular use-case and problem.

sourabhjaiswal, 2021-04-21:
Hi,

I am also facing the same problem. My setup is :-
Raspberry pi 4 4gb
USB SSD: https://www.amazon.in/gp/product/B08N187N6N/
USB HDD: https://www.amazon.in/gp/product/B01N07NBLA

I am booting system with USB SSD and connecting HDD later on. but as soon as I connect the HDD it crashes.
I tried with one Seagate HDD too but same result.

Any feedback on resolving it would be greatly appreciated.

andrum99, 2021-04-21:
> Hi,
>
> I am also facing the same problem. My setup is :-
> Raspberry pi 4 4gb
> USB SSD: https://www.amazon.in/gp/product/B08N187N6N/
> USB HDD: https://www.amazon.in/gp/product/B01N07NBLA
>
> I am booting [...]

Use a decent quality powered hub. That applies to anyone with USB problems with drives attached, even just one drive.

An example of a decent quality powered USB 3.0 hub is the D-Link DUB-1340, which is what I use and it works great. See https://eu.dlink.com/uk/en/products/dub-1340-4-port-superspeed-usb-3-hub.

xMadNet, 2021-05-05:
Hi,
Same issue here
A Pi 4 8Gb, update, upgrade done, Rpi-update done
Linux XPi4 5.10.33-v7l+ #1415 SMP Fri Apr 30 15:50:57 BST 2021 armv7l GNU/Linux
I have a mSata linked with a x857 card. I Boot on this ssd 
Everything fine
I bought an ORICO 9548RU3 HDD Enclosure (so externally powed) plug on the remaining USB3
It usually works....
But sometimes (once a day) I have those IO errors (ls, df ... returns errors)
After a hard reboot all is fine
If I plug in the USB2 it works fine (but half hard disk throughput :-( )
Any ideas ?
BR

xMadNet, 2021-05-06:
Hi,
To solve my problem
I've connected booting SSD tu UBS2 and HHD enclosure to USB3 (only this device on USB3)
No issues with this configuration !
So I don't understantd the link made with Poor Quality HDD Enclosure.
BR

daedalia, 2021-08-05:
I've had some similar issues. This article provides some insight and good suggestions. https://www.pragmaticlinux.com/2021/03/fix-for-getting-your-ssd-working-via-usb-3-on-your-raspberry-pi/

schmidchr, 2021-09-15:
> I've had some similar issues. This article provides some insight and good suggestions. https://www.pragmaticlinux.com/2021/03/fix-for-getting-your-ssd-working-via-usb-3-on-your-raspberry-pi/

thanks to @daedalia , disabling UASP ('Fix 2' in above article) enabled me to have two 'active devices' (SSD) in raid1. Before the second SSD was reported 'removed' by `mdam -D /dev/md1` <br> As indicated in the article, hopefully a software or firmware fix becomes available in the not too distant future, to get that extra 10-30% that the UASP protocol could squeeze out.

KeenMaron, 2022-01-13:
Hi! I have the same problem, as soon as I connect a second drive (with external power) the GUI of raspberian goes black and nothing is clickable anymore on the desktop. I will try it with an active hub, but it's weird, that the external power of the hard drive is not enough??

JohannesArtmeier, 2022-06-25:
I have some similar issues with Ubuntu Server 22.04, but its happing also on USB 2.0 ports.
Here you can find my [StackOverflow post](https://stackoverflow.com/questions/72752807/rp4-ubuntu-server-22-04-filesystem-crash-on-usb-plug-unplug).

peterstamps, 2023-08-25:
A lot of issues with powered USB 3.0 hubs, Raspberry Pi 4 and a HDD/SSD with a SATA-USB 3.0 cable or in a Case may be solved by adding a command into your /boot/cmdline.txt file of your Raspberry Pi 4.

This solution solved everthing for me!

Read here how you can fix that: https://www.pragmaticlinux.com/2021/03/fix-for-getting-your-ssd-working-via-usb-3-on-your-raspberry-pi/#:~:text=For%20some%20reason%20the%20Raspberry%20PI%204%20does,connecting%20your%20SSD%20to%20your%20Raspberry%20PI%204.

LinuxfarmerHH, 2025-04-21:
Raspi 4B with Bookworm
OMV7 installed
Geekworm NASPI Gemini 2.5 with JMicron 551 RAID-1

Everything worked fine for weeks, until i plugged an external Intenso SSD on the second USB-3 port. Copying small files are okay, but had a 9GB video file in my data. That was the last response, the usb-adapter went lost until reboot.

The Gemini 2.5 is powered by a 12V adapter that feed the raspi over the ports. So only the Intenso SSD could be the problem.

Fix above is in use now.