raspberrypi/linux #3487: How to compile 32 bit kernel on Raspbian when 64 bit kernel is enabled

guysoft, 2020-03-01:
**Describe the bug**
I have a build server with the official 64 bit kernel (32 bit userspace Raspbian).
The build server is 64 bit because I have 64 bit things to compile now. However this breaks the build of the 32bit realtime kernel. Seem like the build decides what to build according to the host kernel.

**To reproduce**
Add ``arm_64bit=1`` in ``/boot/config.txt`` and try to build the kernel

**Expected behaviour**
Kernel should build according to the toolchain installed

**Actual behaviour**
Kernel tries to build as ``arm64`` and fails

**System**

* Which model of Raspberry Pi? e.g. Pi3B+, PiZeroW
Rpi 4

* Which OS and version (`cat /etc/rpi-issue`)?
Raspberry Pi reference 2019-07-10
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 175dfb027ffabd4b8d5080097af0e51ed9a4a56c, stage2

* Which firmware version (`vcgencmd version`)?
```
Oct 21 2019 16:47:24 
Copyright (c) 2012 Broadcom
version 8c7c951122d2cae572e891f04ddf287078ff2f11 (tainted) (release) (start)
```
* Which kernel version (`uname -a`)?

**Logs**
```
Allow slab caches to be merged (SLAB_MERGE_DEFAULT) [Y/n/?] y
SLAB freelist randomization (SLAB_FREELIST_RANDOM) [N/y/?] n
Harden slab freelist metadata (SLAB_FREELIST_HARDENED) [N/y/?] n
Profiling support (PROFILING) [Y/n/?] y
*
* Linux/arm64 4.19.71 Kernel Configuration
*
*
* Compiler: gcc (Raspbian 8.3.0-6+rpi1) 8.3.0

[...]

make -j4 zImage modules dtbs
arch/arm64/Makefile:27: ld does not support --fix-cortex-a53-843419; kernel may be susceptible to erratum
arch/arm64/Makefile:40: LSE atomics not supported by binutils
make: *** No rule to make target 'zImage'.  Stop.

```

**Additional context**
Attaching full log
[consoleText.txt](https://github.com/raspberrypi/linux/files/4271691/consoleText.txt)

pelwell, 2020-03-01:
Treat it as a cross compile (which it is) and add ARCH=arm to the make command?

guysoft, 2020-03-01:
Ok, seems like I indeed missed an ``ARCH=arm`` in this line: https://github.com/guysoft/CustomPiOS/blob/076e5dcbee4276499e25421c54a031926cb890c9/src/modules/kernel/end_chroot_script#L45

Letting it build and making sure its finishing, then will confirm this indeed fixes the issue.

guysoft, 2020-03-02:
Ok works, closing

closed 2020-03-02 by guysoft

guysoft, 2020-03-03:
Works, closing