raspberrypi/linux #3488: smsc75xx broken on 5.4.21

Pigi-102, 2020-03-02:
**Describe the bug**
Just few minutes fter the boot, and as soon as strting some heavy load on Pi I get:
[Thu Feb 27 15:19:31 2020] ------------[ cut here ]------------
[Thu Feb 27 15:19:31 2020] WARNING: CPU: 3 PID: 0 at net/sched/sch_generic.c:447 dev_watchdog+0x310/0x314
[Thu Feb 27 15:19:31 2020] NETDEV WATCHDOG: eth1 (smsc75xx): transmit queue 0 timed out
[Thu Feb 27 15:19:31 2020] Modules linked in: ipt_REJECT nf_reject_ipv4 xt_REDIRECT xt_nat xt_MASQUERADE xt_multiport xt_recent ipt_SYNPROXY nf_synproxy_core xt_CT xt_tcpudp xt_state xt_conntrack iptable_raw iptable_nat iptable_filter ip_tables x_tables nf_nat_ftp nf_nat nf_conntrack_ftp nf_conntrack nf_defrag_ipv4 bridge stp llc tun ipv6 nf_defrag_ipv6 dm_mod evdev smsc75xx brcmfmac brcmutil rtc_ds1307 sha256_generic libsha256 cfg80211 bcm2835_codec(C) videobuf2_dma_contig bcm2835_v4l2(C) rfkill v4l2_mem2mem bcm2835_mmal_vchiq(C) videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 videobuf2_common raspberrypi_hwmon hwmon i2c_bcm2835 videodev vc_sm_cma(C) mc uio_pdrv_genirq uio fixed
[Thu Feb 27 15:19:31 2020] CPU: 3 PID: 0 Comm: swapper/3 Tainted: G         C        5.4.21-v7 #3
[Thu Feb 27 15:19:31 2020] Hardware name: BCM2835
[Thu Feb 27 15:19:31 2020] Backtrace: 
[Thu Feb 27 15:19:31 2020] [<8010de94>] (dump_backtrace) from [<8010e1e4>] (show_stack+0x20/0x24)
[Thu Feb 27 15:19:31 2020]  r7:ba93a000 r6:00000000 r5:600e0113 r4:80d920dc
[Thu Feb 27 15:19:31 2020] [<8010e1c4>] (show_stack) from [<8088b4c8>] (dump_stack+0xd4/0x118)
[Thu Feb 27 15:19:31 2020] [<8088b3f4>] (dump_stack) from [<801202bc>] (__warn+0xe0/0x108)
[Thu Feb 27 15:19:31 2020]  r10:bb053440 r9:00000009 r8:8079a520 r7:000001bf r6:8079a520 r5:00000009
[Thu Feb 27 15:19:31 2020]  r4:80b0832c r3:00000000
[Thu Feb 27 15:19:31 2020] [<801201dc>] (__warn) from [<801206bc>] (warn_slowpath_fmt+0xa8/0xc4)
[Thu Feb 27 15:19:31 2020]  r7:000001bf r6:80b0832c r5:80b082f0 r4:ba93a000
[Thu Feb 27 15:19:31 2020] [<80120618>] (warn_slowpath_fmt) from [<8079a520>] (dev_watchdog+0x310/0x314)
[Thu Feb 27 15:19:31 2020]  r9:ba93a000 r8:ba33f000 r7:00000003 r6:80d03d00 r5:ba33f2a8 r4:00000000
[Thu Feb 27 15:19:31 2020] [<8079a210>] (dev_watchdog) from [<80198544>] (call_timer_fn+0x40/0x1a0)
[Thu Feb 27 15:19:31 2020]  r8:0025fe28 r7:00000100 r6:8079a210 r5:ba33f2a8 r4:ba33f2a8
[Thu Feb 27 15:19:31 2020] [<80198504>] (call_timer_fn) from [<80198cd0>] (run_timer_softirq+0x248/0x63c)
[Thu Feb 27 15:19:31 2020]  r8:0025fe28 r7:00000000 r6:80d97ee0 r5:ba33f2a8 r4:ba93be28
[Thu Feb 27 15:19:31 2020] [<80198a88>] (run_timer_softirq) from [<8010247c>] (__do_softirq+0x18c/0x410)
[Thu Feb 27 15:19:31 2020]  r10:ba93a000 r9:00000282 r8:ba89c000 r7:00000100 r6:00000001 r5:00000002
[Thu Feb 27 15:19:31 2020]  r4:80d03084
[Thu Feb 27 15:19:31 2020] [<801022f0>] (__do_softirq) from [<8012625c>] (irq_exit+0xdc/0x100)
[Thu Feb 27 15:19:31 2020]  r10:00000000 r9:ba93a000 r8:ba89c000 r7:00000001 r6:00000000 r5:00000000
[Thu Feb 27 15:19:31 2020]  r4:ffffe000
[Thu Feb 27 15:19:31 2020] [<80126180>] (irq_exit) from [<8017d7e0>] (__handle_domain_irq+0x70/0xc0)
[Thu Feb 27 15:19:31 2020]  r5:00000000 r4:80c99264
[Thu Feb 27 15:19:31 2020] [<8017d770>] (__handle_domain_irq) from [<80102224>] (bcm2836_arm_irqchip_handle_irq+0x64/0xa4)
[Thu Feb 27 15:19:31 2020]  r9:ba93a000 r8:80d974f4 r7:ba93bf6c r6:ffffffff r5:600e0013 r4:80109ad0
[Thu Feb 27 15:19:31 2020] [<801021c0>] (bcm2836_arm_irqchip_handle_irq) from [<80101a3c>] (__irq_svc+0x5c/0x7c)
[Thu Feb 27 15:19:31 2020] Exception stack(0xba93bf38 to 0xba93bf80)
[Thu Feb 27 15:19:31 2020] bf20:                                                       00000000 3a3c1000
[Thu Feb 27 15:19:31 2020] bf40: 600e0093 600e0093 ba93a000 00000003 80d04d64 80d04dac 80d974f4 80a9a8a0
[Thu Feb 27 15:19:31 2020] bf60: 00000000 ba93bf94 ffffe000 ba93bf88 ba93a000 80109ad0 600e0013 ffffffff
[Thu Feb 27 15:19:31 2020] [<80109a9c>] (arch_cpu_idle) from [<808a97d4>] (default_idle_call+0x34/0x48)
[Thu Feb 27 15:19:31 2020] [<808a97a0>] (default_idle_call) from [<801539a0>] (do_idle+0xec/0x16c)
[Thu Feb 27 15:19:31 2020] [<801538b4>] (do_idle) from [<80153cf8>] (cpu_startup_entry+0x28/0x2c)
[Thu Feb 27 15:19:31 2020]  r9:410fd034 r8:0000406a r7:80da8710 r6:10c0387d r5:00000003 r4:00000089
[Thu Feb 27 15:19:31 2020] [<80153cd0>] (cpu_startup_entry) from [<8010fc14>] (secondary_start_kernel+0x160/0x16c)
[Thu Feb 27 15:19:31 2020] [<8010fab4>] (secondary_start_kernel) from [<0010278c>] (0x10278c)
[Thu Feb 27 15:19:31 2020]  r5:00000055 r4:3a92406a
[Thu Feb 27 15:19:31 2020] ---[ end trace bf88d81f97966385 ]---

**To reproduce**
Boot with 5.4.21, config the ethernet on smsc75xx ( eth1 on my installation ).
After that ( it has happened every time ) I do 
dump 0uf - / | ssh OTHERHOST ( cat > /tmp/whatever ) ( I'm transmitting on eth0 thought ) and eth1 stop working.
I do ifconfig eth1 down; ifconfig eth1 up and transmit starts again, but as soon as you do other load ( also a simple cpio on the filesystem, or a kernel compile ) the ethernet stops transmitting again.
In any case, during the day, sometimes the eth1 stops transmitting and I have to ifconfig down and up to get her back

**Expected behaviour**
No problems on smsc75xx. With 4.19.x ( 80 and 106 for sure ) I have no stops in what every thing I do with PI ( dump, kernel compile and so on )

**System**

* Which model of Raspberry Pi? e.g. Pi3B+ 
Hardware        : BCM2835
Revision        : a02082
Serial          : 00000000a804788e
Model           : Raspberry Pi 3 Model B Rev 1.2

* Which OS and version (`cat /etc/rpi-issue`)?
 Slackware current 

* Which firmware version (`vcgencmd version`)?
vcgencmd version
Feb 20 2020 16:43:56 
Copyright (c) 2012 Broadcom
version 1614a1ded604d11f395044be137350df84e3d8ee (clean) (release) (start)

* Which kernel version (`uname -a`)?
Linux 5.4.21-v7 hardfloat 

**Additional context**
 smsc75xx 1-1.2:1.0 eth1: register 'smsc75xx' at usb-3f980000.usb-1.2, smsc75xx USB 2.0 Gigabit Ethernet

6by9, 2020-03-03:
When you say heavy load, do you mean CPU or network?
What device do you have that includes an smsc75xx chip? They aren't terribly common, and not one I believe we have available to test.
If you try an alternate USB gigabit adapter do you also see issues? It could be either the USB subsystem or the network driver that is having issues, so it'd be nice to identify which.

Pigi-102, 2020-03-03:
When I say heavy load I mean whatever heavy load: cpu/disk ( sd card )/network ( but not directly on eth1 ). 
For what I've seen it seems cpu/sd related as a kernel compile, or a container ( docker ) execution have this kind of pattern load.
The only thing that stop working is the "eth1" ( smsc75xx on usb ).
Everything else works well.

I will try to find another dongle, I don't have one at the moment.

Moreover, the kernel oops talks about smsc75xx....

6by9, 2020-03-03:
The kernel oops says that the smsc75xx has transmit queue has timed out. Seeing as the device connects over USB, the failure could be in either part of the system.

If it's not directly heavy network traffic on eth1 that triggers it then I would suspect that it is smsc75xx specific, but there haven't been many changes to it between 4.19 and 5.4, and it seems a slightly odd failure mode.
As I say, it's a rare chip and I don't believe we have one to test. If you can tell us what the adapter is that you are using then we can try to obtain one.

Pigi-102, 2020-03-03:
I can give you the info ( hopefully ) in the evening as the Pi is at home.
In any case I've done some reseach and you can find some info's searching for:
StarTech USB 2.0 to Gigabit Ethernet NIC Network Adapter (USB21000S2)

It seem to be quite high level as chipset ( it's quite expensive compared to RealTek chipset ).

6by9, 2020-03-03:
It looks like that adapter is pretty much obsolete. It's not unsurprising as USB2 can only deliver a theoretical max of 480Mbit/s, so can't cope with a full speed gigabit ethernet link, so flow control is a must (the same issue as we have with the 3B+ and LAN7800). USB3 solves that problem.

There was one available on Ebay, so I've ordered it. I'll see what happens once it has turned up.

Pigi-102, 2020-03-03:
Yes, USB2 can go max 480Mbit/sec, but it' way more than a 100Mbit you an reach on mi Pi3 ( old version, so 100Mbit on frontend :) )
Cool for you buy, I'm trying to buy a RealTek to check if the problem is somewhere.
In effect smsc75xx driver hasn't change for years in kernel :(

Pigi-102, 2020-03-03:
Sorry, on my dongle there is no label nor nothing to identify it in any way.
But I an confirm that with 4.19 I had no issue in the last 5 days ( while on 5.4.21 quite a bit of stops per day ).
I ordered another dongle on Amazon and plan to make some tests on saturday/sunday with exactly the same kernel ( I kept kernel and moduls from the previous install)

Pigi-102, 2020-03-04:
I managed to find information about by smsc75xx dongle:
https://www.dell.com/support/home/us/en/04/drivers/driversdetails?driverid=t83fm
It's a rather old, but a quality one ( so it seems ) being sell by Dell.
Part Number OTFXX8 if needed.
The link says it's from 2013 but with drivers updated to 2019.

In any case I've received the RTL one:
Bus 001 Device 007: ID 0bda:8153 Realtek Semiconductor Corp. RTL8153 Gigabit Ethernet Adapter

I will make the tests during the week end as told, as I need internet for work in these days.

Pigi-102, 2020-03-05:
I do suspect problem shoudl be in other places.
With the new adapter I haven't had the same problem ( ethernet stops trasmitting ) but I've had other kind of oops:

[  827.757446] swapper/0: page allocation failure: order:3, mode:0x40a20(GFP_ATOMIC|__GFP_COMP), nodemask=(null),cpuset=/,mems_allowed=0
[  827.757481] CPU: 0 PID: 0 Comm: swapper/0 Tainted: G         C        5.4.21-v7 #3
[  827.757484] Hardware name: BCM2835
[  827.757486] Backtrace: 
[  827.757506] [<8010de94>] (dump_backtrace) from [<8010e1e4>] (show_stack+0x20/0x24)
[  827.757518]  r7:80d00000 r6:00000000 r5:60070113 r4:80d920dc
[  827.757531] [<8010e1c4>] (show_stack) from [<8088b4c8>] (dump_stack+0xd4/0x118)
[  827.757543] [<8088b3f4>] (dump_stack) from [<802bdcfc>] (warn_alloc+0xd8/0x16c)
[  827.757549]  r10:00000003 r9:00000003 r8:ffffe000 r7:80ab1c68 r6:00000000 r5:00000000
[  827.757553]  r4:80d00000 r3:799c52f2
[  827.757560] [<802bdc24>] (warn_alloc) from [<802beeb8>] (__alloc_pages_nodemask+0x1128/0x11f0)
[  827.757564]  r3:00000003 r2:80ab1c68
[  827.757570]  r8:00040a20 r7:00000a20 r6:00000201 r5:00040a20 r4:00000200
[  827.757591] [<802bdd90>] (__alloc_pages_nodemask) from [<7f2c7a4c>] (alloc_rx_agg+0x60/0x138 [r8152])
[  827.757597]  r10:00000000 r9:bab33548 r8:ba89c000 r7:00000a20 r6:00000003 r5:bab33540
[  827.757600]  r4:9f99dd00
[  827.757620] [<7f2c79ec>] (alloc_rx_agg [r8152]) from [<7f2c8534>] (r8152_poll+0x6c8/0x7a4 [r8152])
[  827.757626]  r7:ac4431c0 r6:bab33548 r5:00000001 r4:20070113
[  827.757641] [<7f2c7e6c>] (r8152_poll [r8152]) from [<80758fac>] (net_rx_action+0x1ac/0x4a8)
[  827.757648]  r10:0000012b r9:00000001 r8:ba89c000 r7:00000040 r6:00000000 r5:00000001
[  827.757651]  r4:bab33548
[  827.757659] [<80758e00>] (net_rx_action) from [<8010247c>] (__do_softirq+0x18c/0x410)
[  827.757666]  r10:80d00000 r9:00000001 r8:ba89c000 r7:00000100 r6:00000003 r5:00000001
[  827.757669]  r4:80d0308c
[  827.757678] [<801022f0>] (__do_softirq) from [<8012625c>] (irq_exit+0xdc/0x100)
[  827.757684]  r10:80c64a40 r9:80d00000 r8:ba89c000 r7:00000001 r6:00000000 r5:00000000
[  827.757687]  r4:ffffe000
[  827.757697] [<80126180>] (irq_exit) from [<8017d7e0>] (__handle_domain_irq+0x70/0xc0)
[  827.757700]  r5:00000000 r4:80c99264
[  827.757707] [<8017d770>] (__handle_domain_irq) from [<80102224>] (bcm2836_arm_irqchip_handle_irq+0x64/0xa4)
[  827.757714]  r9:80d00000 r8:80d974f4 r7:80d01f0c r6:ffffffff r5:60070013 r4:80109ad0
[  827.757721] [<801021c0>] (bcm2836_arm_irqchip_handle_irq) from [<80101a3c>] (__irq_svc+0x5c/0x7c)
[  827.757725] Exception stack(0x80d01ed8 to 0x80d01f20)
[  827.757730] 1ec0:                                                       00000000 3a385000
[  827.757736] 1ee0: 60070093 60070093 80d00000 00000000 80d04d64 80d04dac 80d974f4 80a9a8a0
[  827.757743] 1f00: 80c64a40 80d01f34 ffffe000 80d01f28 80d00000 80109ad0 60070013 ffffffff
[  827.757755] [<80109a9c>] (arch_cpu_idle) from [<808a97d4>] (default_idle_call+0x34/0x48)
[  827.757763] [<808a97a0>] (default_idle_call) from [<801539a0>] (do_idle+0xec/0x16c)
[  827.757769] [<801538b4>] (do_idle) from [<80153cf8>] (cpu_startup_entry+0x28/0x2c)
[  827.757776]  r9:bb7ff9c0 r8:00000000 r7:80d00000 r6:80da83c0 r5:80d04d40 r4:000000cd
[  827.757783] [<80153cd0>] (cpu_startup_entry) from [<808a3324>] (rest_init+0xb4/0xbc)
[  827.757794] [<808a3270>] (rest_init) from [<80c00b70>] (arch_call_rest_init+0x18/0x1c)
[  827.757798]  r5:80d04d40 r4:80da8414
[  827.757807] [<80c00b58>] (arch_call_rest_init) from [<80c01098>] (start_kernel+0x4ac/0x4dc)
[  827.757813] [<80c00bec>] (start_kernel) from [<00000000>] (0x0)
[  827.757844] Mem-Info:
[  827.757859] active_anon:15979 inactive_anon:18587 isolated_anon:0
                active_file:85100 inactive_file:89996 isolated_file:32
                unevictable:0 dirty:17 writeback:0 unstable:0
                slab_reclaimable:4467 slab_unreclaimable:3604
                mapped:6912 shmem:1022 pagetables:1004 bounce:0
                free:16358 free_pcp:823 free_cma:1080
[  827.757870] Node 0 active_anon:63916kB inactive_anon:74348kB active_file:340400kB inactive_file:359984kB unevictable:0kB isolated(anon):0kB isolated(file):128kB mapped:27648kB dirty:68kB writeback:0kB shmem:4088kB writeback_tmp:0kB unstable:0kB all_unreclaimable? no
[  827.757884] Normal free:65432kB min:12664kB low:13636kB high:14608kB active_anon:63916kB inactive_anon:74348kB active_file:340528kB inactive_file:360012kB unevictable:0kB writepending:68kB present:983040kB managed:961352kB mlocked:0kB kernel_stack:2264kB pagetables:4016kB bounce:0kB free_pcp:3292kB local_pcp:240kB free_cma:4320kB
[  827.757888] lowmem_reserve[]: 0 0
[  827.757917] Normal: 5971*4kB (UMEHC) 4204*8kB (UMEHC) 377*16kB (UMEHC) 35*32kB (C) 13*64kB (C) 1*128kB (C) 0*256kB 0*512kB 0*1024kB 0*2048kB 0*4096kB = 65628kB
[  827.757975] 176152 total pagecache pages
[  827.757983] 8 pages in swap cache
[  827.757989] Swap cache stats: add 347, delete 339, find 0/1
[  827.757994] Free swap  = 1046780kB
[  827.757998] Total swap = 1048572kB
[  827.758003] 245760 pages RAM
[  827.758008] 0 pages HighMem/MovableOnly
[  827.758012] 5422 pages reserved
[  827.758017] 2048 pages cma reserved

The command that triggers this kind of problem is:
/usr/local/sbin/dump -0u -f - / | ssh root@topo "cat > /extbig/CasaDump.dmp"

The command complete correctly, but I haven't tested the output if is completely working. It seems fine thought.

This does not happens with previuos kernel. 
I will do more tests to see if it will happens again after the first ooops.

6by9, 2020-03-05:
Does it happen against the standard SMSC9514 on the Pi3? I know that's only 100Mbit/s, but it's still valid.
Is flow control enabled on your switch? ```sudo ethtool -a eth1``` IIRC.

Pigi-102, 2020-03-06:
The oops is not related to new usb dongle this time.
And the new dongle hasn't stopped working since yestrday morning.
I do use the "internal" SMSC9514 as interface during the copy:

Internet------------eth1-Pi-eth0------- internalnet

The dump copy is ( always ) from pi to internal net so trough the eth0 ( SMSC9514 ).
That was initally the strange thing: I've never user the SMSC75xx for transfer but she was the one doing the ops but was she that stop working.

6by9, 2020-03-07:
My dongle has turned up, and does identify as "idVendor=0424, idProduct=7500" and "smsc75xx USB 2.0 Gigabit Ethernet", so I'll see what I can find next week.