raspberrypi/linux #3489: AES Encryption issue when using CBC mode

amrithathorath, 2020-03-04:
I'm trying to perform AES encryption using kernel crypto. I've pasted the code below. The code generates right output for ECB but not CBC.

```
        switch (params->mode) {
                case MODE_ECB:
                        tfm = crypto_alloc_blkcipher("ecb(aes)", 0, 0); break;
                case MODE_CBC:
                        tfm = crypto_alloc_blkcipher("cbc(aes)", 0, 0); break;
                default:
                        LOG_INFO("ERROR: could not allocate cipher.  %s, line %d\n", __FILE__, __LINE__);
                        return;
        }

        LOG_DEBUG("Cleared switch statement");
        if (!tfm) {
                printk("Unable to alocate cipher\n");
                return;
        }
        desc.tfm = tfm;
        desc.flags = 0;

        LOG_DEBUG("Cleared switch statement 2");

        if (crypto_blkcipher_setkey(tfm, params->key, params->keyLength)) {
                printk("Unable to set Key\n");
                goto free_return;
        }
        LOG_DEBUG("Cleared crypto_blkcipher_setkey");
```

Here's what the logs look like for ECB and CBC.

```
ECB
===
Mar  4 16:03:24 alg-kern kernel: [192691.244307] DEBUG: Cleared switch statement
Mar  4 16:03:24 alg-kern kernel: [192691.244309] DEBUG: Cleared switch statement 2
Mar  4 16:03:24 alg-kern kernel: [192691.245967] DEBUG: Cleared crypto_blkcipher_setkey
Mar  4 16:03:24 alg-kern kernel: [192691.246796] DEBUG: Cleared crypto_blkcipher_set_iv

CBC
===
Mar  4 16:05:57 alg-kern kernel: [192843.442628] DEBUG: Cleared switch statement
Mar  4 16:05:57 alg-kern kernel: [192843.442631] DEBUG: Cleared switch statement 2
Mar  4 16:05:57 alg-kern kernel: [192843.443557] BUG: unable to handle kernel NULL pointer dereference at 000000000000000e
```
The code for ECB and CBC is exactly the same except for the allocation of "tfm". And the error seems to be at crypto_blkcipher_setkey(). Has someone come across this issue? Currently I'm trying to build this on 4.15.18-041518-generic and once I can build it I want to be able to port this code on ARM.

pelwell, 2020-03-04:
Read the comment above the source code for crypto_alloc_blkcipher, then work out why your error checking is inadequate.

amrithathorath, 2020-03-05:
I'm not sure that there's something wrong with the code. Like I said it works fine when I'm running ECB mode but fails only when I'm running CBC mode.

pelwell, 2020-03-05:
I'm telling you that there is something wrong with the code - I was trying to give you a hint, and failing.

crypto_alloc_blkcipher is defined to return a pointer or a negative error code - it should never return NULL. The error code will pass your `(!tfm)` test, causing the later code to use a garbage pointer.

Change your code to use `IS_ERR(tfm)` instead.

amrithathorath, 2020-03-05:
@pelwell : My bad, I misunderstood. The IS_ERR() definitely did help me to understand that the error was caused while allocating the cipher. That's the first helpful comment I've got in weeks.
What stumps about this issue is that when I use ECB mode the allocation works perfectly fine but dies with CBC. I've checked the /proc/crypto on the system that I'm running on and I do see "cbc(aes)".

```
name         : cbc(aes)
driver       : cbc-aes-aesni
module       : aesni_intel
priority     : 400
refcnt       : 1
selftest     : passed
internal     : no
type         : skcipher
async        : yes
blocksize    : 16
min keysize  : 16
max keysize  : 32
ivsize       : 16
chunksize    : 16
walksize     : 16

name         : ecb(aes)
driver       : ecb-aes-aesni
module       : aesni_intel
priority     : 400
refcnt       : 1
selftest     : passed
internal     : no
type         : skcipher
async        : yes
blocksize    : 16
min keysize  : 16
max keysize  : 32
ivsize       : 0
chunksize    : 16
walksize     : 16
```

6by9, 2020-03-05:
```
module       : aesni_intel
```
You are aware that this is a repository for the Raspberry Pi Linux kernel - an ARM SBC. 
aesni_intel is an Intel x86 specific crypto module.

amrithathorath, 2020-03-05:
Yes, my intention is to move this code to ARM system once I can fix the errors.
I was just seeing if anyone saw anything that was very obviously wrong with my code because then I would see the same error when I move to ARM system as well.

6by9, 2020-03-05:
```
pi@raspberrypi:~ $ cat /proc/crypto 
name         : ecdh
driver       : ecdh-generic
module       : ecdh_generic
priority     : 100
refcnt       : 1
selftest     : passed
internal     : no
type         : kpp

name         : pkcs1pad(rsa,sha256)
driver       : pkcs1pad(rsa-generic,sha256)
module       : kernel
priority     : 100
refcnt       : 1
selftest     : passed
internal     : no
type         : akcipher

name         : sha224
driver       : sha224-generic
module       : sha256_generic
priority     : 100
refcnt       : 1
selftest     : passed
internal     : no
type         : shash
blocksize    : 64
digestsize   : 28

name         : sha256
driver       : sha256-generic
module       : sha256_generic
priority     : 100
refcnt       : 1
selftest     : passed
internal     : no
type         : shash
blocksize    : 64
digestsize   : 32

name         : crc32
driver       : crc32-generic
module       : kernel
priority     : 100
refcnt       : 1
selftest     : passed
internal     : no
type         : shash
blocksize    : 1
digestsize   : 4

name         : crc32c
driver       : crc32c-generic
module       : kernel
priority     : 100
refcnt       : 3
selftest     : passed
internal     : no
type         : shash
blocksize    : 1
digestsize   : 4

name         : des3_ede
driver       : des3_ede-generic
module       : kernel
priority     : 100
refcnt       : 1
selftest     : passed
internal     : no
type         : cipher
blocksize    : 8
min keysize  : 24
max keysize  : 24

name         : des
driver       : des-generic
module       : kernel
priority     : 100
refcnt       : 1
selftest     : passed
internal     : no
type         : cipher
blocksize    : 8
min keysize  : 8
max keysize  : 8

name         : ecb(cipher_null)
driver       : ecb-cipher_null
module       : kernel
priority     : 100
refcnt       : 1
selftest     : passed
internal     : no
type         : skcipher
async        : no
blocksize    : 1
min keysize  : 0
max keysize  : 0
ivsize       : 0
chunksize    : 1
walksize     : 1

name         : digest_null
driver       : digest_null-generic
module       : kernel
priority     : 0
refcnt       : 1
selftest     : passed
internal     : no
type         : shash
blocksize    : 1
digestsize   : 0

name         : compress_null
driver       : compress_null-generic
module       : kernel
priority     : 0
refcnt       : 1
selftest     : passed
internal     : no
type         : compression

name         : cipher_null
driver       : cipher_null-generic
module       : kernel
priority     : 0
refcnt       : 1
selftest     : passed
internal     : no
type         : cipher
blocksize    : 1
min keysize  : 0
max keysize  : 0

name         : rsa
driver       : rsa-generic
module       : kernel
priority     : 100
refcnt       : 1
selftest     : passed
internal     : no
type         : akcipher
```
No CBC option, although I don't know if that is down to just a missing module.

amrithathorath, 2020-03-05:
Ah I see. Thanks!

closed 2020-03-05 by amrithathorath

vk5tu, 2020-03-06:
> No CBC option, although I don't know if that is down to just a missing module.

Maybe kernel module aes_arm_bs is missing? Here's Raspbian on Rpi4 4GB

```
$ grep 'cbc.*aes' < /proc/crypto
name         : cbc(aes)
driver       : cbc-aes-neonbs
name         : __cbc(aes)
driver       : __cbc-aes-neonbs
$ uname -a
Linux redacted 4.19.97-v7l+ #1294 SMP Thu Jan 30 13:21:14 GMT 2020 armv7l GNU/Linux
$ grep Model < /proc/cpuinfo
Model		: Raspberry Pi 4 Model B Rev 1.1
$ lsmod
Module                  Size  Used by
tun                    45056  0
aes_arm_bs             20480  0
crypto_simd            16384  1 aes_arm_bs
cryptd                 24576  1 crypto_simd
algif_skcipher         16384  0
af_alg                 28672  1 algif_skcipher
snd_seq_dummy          16384  0
snd_seq                69632  1 snd_seq_dummy
uinput                 20480  0
joydev                 20480  0
evdev                  24576  6
snd_usb_audio         192512  1
snd_hwdep              16384  1 snd_usb_audio
snd_usbmidi_lib        32768  1 snd_usb_audio
snd_rawmidi            32768  1 snd_usbmidi_lib
snd_seq_device         16384  2 snd_seq,snd_rawmidi
uvcvideo               98304  0
hid_apple              16384  0
fuse                  110592  5
rfcomm                 49152  4
bnep                   20480  2
hci_uart               40960  1
btbcm                  16384  1 hci_uart
serdev                 20480  1 hci_uart
bluetooth             389120  29 hci_uart,bnep,btbcm,rfcomm
ecdh_generic           28672  1 bluetooth
binfmt_misc            20480  1
brcmfmac              311296  0
brcmutil               16384  1 brcmfmac
vc4                   176128  9
sha256_generic         20480  0
cfg80211              647168  1 brcmfmac
drm_kms_helper        184320  2 vc4
v3d                    73728  6
snd_soc_core          192512  1 vc4
gpu_sched              28672  1 v3d
rfkill                 28672  4 bluetooth,cfg80211
drm                   442368  16 v3d,vc4,gpu_sched,drm_kms_helper
snd_compress           20480  1 snd_soc_core
raspberrypi_hwmon      16384  0
snd_bcm2835            24576  3
hwmon                  16384  1 raspberrypi_hwmon
snd_pcm_dmaengine      16384  1 snd_soc_core
drm_panel_orientation_quirks    16384  1 drm
snd_pcm               102400  5 snd_usb_audio,vc4,snd_pcm_dmaengine,snd_bcm2835,snd_soc_core
bcm2835_codec          36864  0
snd_timer              32768  2 snd_seq,snd_pcm
snd                    73728  19 snd_compress,snd_hwdep,snd_seq,snd_usb_audio,snd_timer,snd_rawmidi,snd_usbmidi_lib,snd_seq_device,snd_bcm2835,snd_soc_core,snd_pcm
bcm2835_v4l2           45056  0
v4l2_mem2mem           24576  1 bcm2835_codec
syscopyarea            16384  1 drm_kms_helper
sysfillrect            16384  1 drm_kms_helper
sysimgblt              16384  1 drm_kms_helper
fb_sys_fops            16384  1 drm_kms_helper
bcm2835_mmal_vchiq     32768  2 bcm2835_codec,bcm2835_v4l2
videobuf2_dma_contig    20480  1 bcm2835_codec
v4l2_common            16384  1 bcm2835_v4l2
videobuf2_vmalloc      16384  2 uvcvideo,bcm2835_v4l2
videobuf2_memops       16384  2 videobuf2_dma_contig,videobuf2_vmalloc
videobuf2_v4l2         24576  4 uvcvideo,bcm2835_codec,bcm2835_v4l2,v4l2_mem2mem
videobuf2_common       45056  5 uvcvideo,bcm2835_codec,bcm2835_v4l2,v4l2_mem2mem,videobuf2_v4l2
videodev              200704  7 uvcvideo,bcm2835_codec,v4l2_common,videobuf2_common,bcm2835_v4l2,v4l2_mem2mem,videobuf2_v4l2
media                  36864  4 uvcvideo,bcm2835_codec,videodev,v4l2_mem2mem
vc_sm_cma              36864  5 bcm2835_mmal_vchiq
rpivid_mem             16384  0
uio_pdrv_genirq        16384  0
uio                    20480  1 uio_pdrv_genirq
i2c_dev                20480  0
ip_tables              24576  0
x_tables               32768  1 ip_tables
ipv6                  454656  76
```

6by9, 2020-03-06:
I did say that I didn't know if it was just down to a missing module.