raspberrypi/linux #3492: VLC & Kodi Crashing with some H265 videos.

Bleep42, 2020-03-09:
Hi,
I have a few H265 Videos that are crashing VLC. This is using a Pi4, Latest updated Raspbian.
[Link to a 2 minute sample](https://www.dropbox.com/s/42tqvsmfy5hxrbt/Apollo%2013%20%281995%29%20%281%29-001.mkv?dl=0)
They are low resolution and used to play fine. They will just about play using Kodie under Raspbian, but are very blocky and distorted, and eventually Kodi will lock up.
All other higher resolution H265 play well.
[There is a fuller investigation on the Pi Forum] (https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=266762) Where the current kernal was rolled back, which fixed the problem.
Would be greatful if someone could take a look.

JamesH65, 2020-03-09:
@popcornmix FYI

popcornmix, 2020-03-09:
Was Jan 10 firmware the last version that worked (so Jan 14 has the issue?)
Can you confirm if it is just the firmware update that causes the issue (i.e. start*.elf/fixup*.dat).

I can check this at some point but not in the office today.

Bleep42, 2020-03-09:
Sorry I've never gone back to a specific version before, I've looked at documentation [here](https://www.raspberrypi.org/documentation/linux/kernel/) but that doesn't help.
Regards, Kevin.

pelwell, 2020-03-09:
This is a VFAQ, so avoid retyping it all you can read a good explanation from MrEngman: https://www.raspberrypi.org/forums/viewtopic.php?t=79031#p561307

Bleep42, 2020-03-09:
Hi @popcornmix 
I can confirm that the 22nd Jan kernel 993F475 Is the last working version.
The 30th Jan update 1526192 is the first kernel with the problem.
Hope this is of help? Let me know if I can check anything else.
Regards, Kevin.

popcornmix, 2020-03-09:
Thanks @Bleep42 . The changes that seem most likely are:
firmware: MMAL / video_render: Allow column stride to be set on column formats
firmware: vc_image/video_decode: Move +16 lines for di_adv from vc_image to decoder

I'll try to reproduce tomorrow and confirm which one it is.

JamesH65, 2020-03-09:
Maybe related? https://github.com/raspberrypi/firmware/issues/1334

6by9, 2020-03-10:
```Assertion geo.height_y >= video_height && geo.height_c >= video_height / 2 failed at src/libavcodec/rpi_zc.c:509```
reported in the forum thread is @jc-kynesim 's code complaining about a format.
Indeed those patches have removed a load of padding from YUVUV formats.

(I did try playing that clip yesterday and it was fine. Then I noticed my kernel didn't have the relevant /dev nodes for the hevc decoder!)

popcornmix, 2020-03-10:
We've pushed a potential fix to rpi-update firmware. Please test and confirm is issue is resolved.

Bleep42, 2020-03-10:
Hi @popcornmix 
Ok, that has fixed the VLC, but not Kodi under Raspbian, I am using 18.5 Git 20191208, though I am unsure you will be concerned about this.
Regards, Kevin.

popcornmix, 2020-03-10:
Hardware decode isn't supported with 64-bit kernel yet, so that is expected.

closed 2020-03-11 by Bleep42