raspberrypi/linux #3494: Bluetooth occurs: Internal error: Oops: 207 [#1] SMP ARM

zorazor, 2020-03-09:
**Is this the right place for my bug report?**
Yes, the error starts with:
`raspberrypi kernel: [ 4188.416710] Unable to handle kernel NULL pointer dereference at virtual address 00000100`

**Describe the bug**
This error occurs when connecting with my bluetooth keyboard incl. trackpad and the Raspberry Pi 4. The error doesn't occur with every connection establishment (approximately after 50 connects and disconnects).
It affects bluetooth (hidp, rfcomm, hci ... ).

**To reproduce**
- Power on the RPi bluetooth and pair/connect a bluetooth device (I'm using a keyboard with trackpad).
- Turn the bluetooth device off, wait a few seconds and turn it on again.
- Repeat it until the error occurs (usually within 50 attempts).

**System**
 Copy and paste the results of the raspinfo command in to this section. Alternatively, copy and paste a pastebin link, or add answers to the following questions:

* Which model of Raspberry Pi? e.g. Pi3B+, PiZeroW
Pi4B 4GB

* Which OS and version (`cat /etc/rpi-issue`)?
Raspberry Pi reference 2019-09-26
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen,
80d486687ea77d31fc3fc13cf3a2f8b464e129be, stage4

* Which firmware version (`vcgencmd version`)?
Feb 12 2020 12:36:21
Copyright (c) 2012 Broadcom
version c3c8dbdf147686fb0c3f32aece709d0653368810 (clean) (release) (start)

* Which kernel version (`uname -a`)?
Linux raspberrypi 4.19.97-v7l+ #1294 SMP Thu Jan 30 13:21:14 GMT 2020
armv7l GNU/Linux

**Logs**
If applicable, add the relevant output from `dmesg` or similar.

`Mar  8 21:48:51 raspberrypi kernel: [ 4188.416710] Unable to handle kernel NULL pointer dereference at virtual address 00000100
Mar  8 21:48:51 raspberrypi kernel: [ 4188.416732] pgd = 024205b4
Mar  8 21:48:51 raspberrypi kernel: [ 4188.416743] [00000100] *pgd=80000000004003, *pmd=00000000
Mar  8 21:48:51 raspberrypi kernel: [ 4188.416761] Internal error: Oops: 207 [#1] SMP ARM
Mar  8 21:48:51 raspberrypi kernel: [ 4188.416772] Modules linked in: hidp rfcomm bnep hci_uart btbcm serdev bluetooth ecdh_generic fuse 8021q garp stp llc evdev sg snd_usb_audio snd_hwdep uvcvideo snd_usbmidi_lib snd_rawmidi snd_seq_device vc4 brcmfmac brcmutil sha256_generic drm_kms_helper v3d snd_soc_core snd_compress gpu_sched snd_pcm_dmaengine bcm2835_v4l2(C) bcm2835_codec(C) v4l2_mem2mem snd_bcm2835(C) bcm2835_mmal_vchiq(C) snd_pcm cfg80211 v4l2_common drm snd_timer raspberrypi_hwmon rfkill hwmon syscopyarea drm_panel_orientation_quirks videobuf2_dma_contig sysfillrect sysimgblt fb_sys_fops snd videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 videobuf2_common videodev media vc_sm_cma(C) rpivid_mem uio_pdrv_genirq uio i2c_dev ip_tables x_tables ipv6
Mar  8 21:48:51 raspberrypi kernel: [ 4188.416923] CPU: 3 PID: 10046 Comm: kworker/u9:2 Tainted: G         C        4.19.97-v7l+ #1294
Mar  8 21:48:51 raspberrypi kernel: [ 4188.416934] Hardware name: BCM2835
Mar  8 21:48:51 raspberrypi kernel: [ 4188.417035] Workqueue: hci0 hci_rx_work [bluetooth]
Mar  8 21:48:51 raspberrypi kernel: [ 4188.417060] PC is at __wake_up_common+0x6c/0x144
Mar  8 21:48:51 raspberrypi kernel: [ 4188.417077] LR is at _raw_spin_unlock_irqrestore+0x6c/0x70`

pelwell, 2020-03-09:
Please include the whole stack trace - everything between the start and end lines.

lukaszbel, 2020-04-30:
I have similar problem with TP-Link UB400 BT dongle. After some times the dongle stops working and in logs is:

Message from syslogd@rpi at Apr 30 14:59:39 ...
 kernel:[ 5513.495930] Internal error: Oops: 207 [#1] SMP ARM

Message from syslogd@rpi at Apr 30 14:59:39 ...
 kernel:[ 5513.496317] Process kworker/u9:1 (pid: 325, stack limit = 0x38da11fe)

Message from syslogd@rpi at Apr 30 14:59:39 ...
 kernel:[ 5513.496326] Stack: (0xd406fde8 to 0xd4070000)

Message from syslogd@rpi at Apr 30 14:59:39 ...
 kernel:[ 5513.496336] fde0:                   d55098a4 d5509000 00000000 d5509008 d406fe24 d406fe08

Message from syslogd@rpi at Apr 30 14:59:39 ...
 kernel:[ 5513.496348] fe00: bf91ed48 bf91c730 d5509000 d2252e40 c1004d88 cb24e000 d406fec4 d406fe28

Message from syslogd@rpi at Apr 30 14:59:39 ...
 kernel:[ 5513.496360] fe20: bf92603c bf91ec7c d406fe5c d406fe60 eff32440 c09d9ee8 d40a4b00 00000000

Message from syslogd@rpi at Apr 30 14:59:39 ...
 kernel:[ 5513.496372] fe40: d406fe6c c0e9cad8 c09dfd9c 00000000 d5509000 005097d8 d550200d bf91ec70

Message from syslogd@rpi at Apr 30 14:59:39 ...
 kernel:[ 5513.496384] fe60: 00000000 d406fe70 c0305db8 c0306fbc 40000013 d55097d8 d55097e4 bf937504

Message from syslogd@rpi at Apr 30 14:59:39 ...
 kernel:[ 5513.496395] fe80: 00000000 d448c800 d2252e40 d5509000 d406fec4 cf92532f bf937504 d5484200

Message from syslogd@rpi at Apr 30 14:59:39 ...
 kernel:[ 5513.496407] fea0: d5509700 d2252e40 d5509000 d55097d8 d5509704 00000001 d406fefc d406fec8

Message from syslogd@rpi at Apr 30 14:59:39 ...
 kernel:[ 5513.496419] fec0: bf917e88 bf925cf0 d5509704 00000001 d406fefc d5484200 d5509700 efb1fe00

Message from syslogd@rpi at Apr 30 14:59:39 ...
 kernel:[ 5513.496430] fee0: dc494700 00000000 d5509704 00000001 d406ff34 d406ff00 c023e080 bf917d28

Message from syslogd@rpi at Apr 30 14:59:39 ...
 kernel:[ 5513.496442] ff00: 40000093 efb1fe00 efb1fe00 efb1fe00 d5484214 efb1fe00 efb1fe18 c1003d00

Message from syslogd@rpi at Apr 30 14:59:39 ...
 kernel:[ 5513.496454] ff20: 00000088 d5484200 d406ff7c d406ff38 c023e3c4 c023df1c d406ff5c 00000000

Message from syslogd@rpi at Apr 30 14:59:39 ...
 kernel:[ 5513.496465] ff40: c1003d00 c1003d00 c1097356 d406e038 d406ff7c d4004c40 d4004100 00000000

Message from syslogd@rpi at Apr 30 14:59:39 ...
 kernel:[ 5513.496477] ff60: d5484200 c023e368 d4004c5c efb61e74 d406ffac d406ff80 c02446f8 c023e374

Message from syslogd@rpi at Apr 30 14:59:39 ...
 kernel:[ 5513.496489] ff80: c02043d8 d4004100 c02445c0 00000000 00000000 00000000 00000000 00000000

Message from syslogd@rpi at Apr 30 14:59:39 ...
 kernel:[ 5513.496500] ffa0: 00000000 d406ffb0 c02010ac c02445cc 00000000 00000000 00000000 00000000

Message from syslogd@rpi at Apr 30 14:59:39 ...
 kernel:[ 5513.496512] ffc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000

Message from syslogd@rpi at Apr 30 14:59:39 ...
 kernel:[ 5513.496523] ffe0: 00000000 00000000 00000000 00000000 00000013 00000000 00000000 00000000

Message from syslogd@rpi at Apr 30 14:59:39 ...
 kernel:[ 5513.496937] Code: e92dd8f0 e24cb004 e52de004 e8bd4000 (e5d05012)

My configuration:
Which model of Raspberry Pi? 

Pi4B 4GB

 Which OS and version (cat /etc/rpi-issue)?
Raspberry Pi reference 2019-09-26
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 80d486687ea77d31fc3fc13cf3a2f8b464e129be, stage2

Which firmware version (vcgencmd version)?
Apr 27 2020 14:33:37
Copyright (c) 2012 Broadcom
version 3a8f1793b758d6fb7f375edaa260e069ecd34c88 (clean) (release) (start)

Which kernel version (uname -a)?
Linux rpi 4.19.118-v7l+ #1311 SMP Mon Apr 27 14:26:42 BST 2020 armv7l GNU/Linux

lukaszbel, 2020-04-30:
And also:

Apr 30 09:52:51 rpi kernel: [56353.481787] Unable to handle kernel NULL pointer dereference at virtual address 00000012
Apr 30 09:52:51 rpi kernel: [56353.481803] pgd = 6bcf787a
Apr 30 09:52:51 rpi kernel: [56353.481811] [00000012] *pgd=80000000004003, *pmd=00000000
Apr 30 09:52:51 rpi kernel: [56353.481824] Internal error: Oops: 207 [#1] SMP ARM
Apr 30 09:52:51 rpi kernel: [56353.481832] Modules linked in: aes_arm_bs crypto_simd cryptd algif_skcipher af_alg xt_nat veth ipt_MASQUERADE nf_conntrack_netlink nft_chain_nat_ipv4 nf_nat_ipv4 br_netfilter bridge overlay rfcomm cmac bnep 8021q garp stp llc binfmt_misc btusb btrtl btintel sg btbcm cdc_acm bluetooth ecdh_generic brcmfmac brcmutil vc4 bcm2835_codec(C) drm_kms_helper v3d bcm2835_v4l2(C) cfg80211 v4l2_mem2mem bcm2835_mmal_vchiq(C) gpu_sched rfkill snd_soc_core v4l2_common videobuf2_dma_contig videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 snd_compress snd_bcm2835(C) videobuf2_common snd_pcm_dmaengine syscopyarea sysfillrect snd_pcm sysimgblt videodev drm fb_sys_fops media snd_timer raspberrypi_hwmon vc_sm_cma(C) drm_panel_orientation_quirks hwmon snd rpivid_mem uio_pdrv_genirq uio nf_log_ipv6 xt_hl ip6_tables ip6t_rt
Apr 30 09:52:51 rpi kernel: [56353.481949]  nf_log_ipv4 nf_log_common xt_LOG xt_multiport xt_comment nft_limit xt_limit xt_addrtype xt_tcpudp xt_conntrack nft_compat nft_counter nf_conntrack_netbios_ns nf_conntrack_broadcast nf_nat_ftp nf_nat nf_conntrack_ftp nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 nf_tables nfnetlink ip_tables x_tables ipv6
Apr 30 09:52:51 rpi kernel: [56353.481995] CPU: 0 PID: 43 Comm: kworker/u9:0 Tainted: G         C        4.19.118-v7l+ #1311
Apr 30 09:52:51 rpi kernel: [56353.482004] Hardware name: BCM2835
Apr 30 09:52:51 rpi kernel: [56353.482057] Workqueue: hci0 hci_rx_work [bluetooth]
Apr 30 09:52:51 rpi kernel: [56353.482102] PC is at hci_connect_le_scan_cleanup+0x14/0x124 [bluetooth]
Apr 30 09:52:51 rpi kernel: [56353.482145] LR is at create_le_conn_complete+0xd8/0xdc [bluetooth]
Apr 30 09:52:51 rpi kernel: [56353.482154] pc : [<bfa1f738>]    lr : [<bfa21d48>]    psr: 60000013
Apr 30 09:52:51 rpi kernel: [56353.482162] sp : efb61de8  ip : efb61e08  fp : efb61e04
Apr 30 09:52:51 rpi kernel: [56353.482169] r10: 00000001  r9 : ef9b3704  r8 : 0000000e
Apr 30 09:52:51 rpi kernel: [56353.482177] r7 : ef9b3008  r6 : 00000000  r5 : ef9b3000  r4 : ef9b38a4
Apr 30 09:52:51 rpi kernel: [56353.482187] r3 : bfa21c70  r2 : 00000000  r1 : 00000000  r0 : 00000000
Apr 30 09:52:51 rpi kernel: [56353.482198] Flags: nZCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment user
Apr 30 09:52:51 rpi kernel: [56353.482207] Control: 30c5383d  Table: 1aba2540  DAC: fffffffd
Apr 30 09:52:51 rpi kernel: [56353.482216] Process kworker/u9:0 (pid: 43, stack limit = 0x6aabbe7b)
Apr 30 09:52:51 rpi kernel: [56353.482226] Stack: (0xefb61de8 to 0xefb62000)
Apr 30 09:52:51 rpi kernel: [56353.482235] 1de0:                   ef9b38a4 ef9b3000 00000000 ef9b3008 efb61e24 efb61e08
Apr 30 09:52:51 rpi kernel: [56353.482247] 1e00: bfa21d48 bfa1f730 ef9b3000 d2e61300 c1004d88 d2e613c0 efb61ec4 efb61e28
Apr 30 09:52:51 rpi kernel: [56353.482259] 1e20: bfa2903c bfa21c7c efb61e5c efb61e60 eff21440 c09d9ee8 efaa1e00 00000000
Apr 30 09:52:51 rpi kernel: [56353.482271] 1e40: efb61e6c c0e9cad8 c09dfd9c 00000000 ef9b3000 009b37d8 ef9b200d bfa21c70
Apr 30 09:52:51 rpi kernel: [56353.482283] 1e60: 00000000 efb61e70 c0305db8 c0306fbc 40000013 ef9b37d8 ef9b37e4 ef9b3000
Apr 30 09:52:51 rpi kernel: [56353.482295] 1e80: efb61ea4 efb61e90 c09dfd9c c0305d74 ef9b37d8 cf92532f efb61ec4 efb09300
Apr 30 09:52:51 rpi kernel: [56353.482307] 1ea0: ef9b3700 d2e61300 ef9b3000 ef9b37d8 ef9b3704 00000001 efb61efc efb61ec8
Apr 30 09:52:51 rpi kernel: [56353.482318] 1ec0: bfa1ae88 bfa28cf0 ef9b3704 00000001 efb61efc efb09300 ef9b3700 efb1fe00
Apr 30 09:52:51 rpi kernel: [56353.482330] 1ee0: da982000 00000000 ef9b3704 00000001 efb61f34 efb61f00 c023e080 bfa1ad28
Apr 30 09:52:51 rpi kernel: [56353.482342] 1f00: 40000093 efb1fe00 efb1fe00 efb1fe00 efb09314 efb1fe00 efb1fe18 c1003d00
Apr 30 09:52:51 rpi kernel: [56353.482354] 1f20: 00000088 efb09300 efb61f7c efb61f38 c023e3c4 c023df1c efb61f5c 00000000
Apr 30 09:52:51 rpi kernel: [56353.482365] 1f40: c1003d00 c1003d00 c1097356 efb60038 efb61f7c efa42200 efb08740 00000000
Apr 30 09:52:51 rpi kernel: [56353.482377] 1f60: efb09300 c023e368 efa4221c ef8edce4 efb61fac efb61f80 c02446f8 c023e374
Apr 30 09:52:51 rpi kernel: [56353.482388] 1f80: c02043d8 efb08740 c02445c0 00000000 00000000 00000000 00000000 00000000
Apr 30 09:52:51 rpi kernel: [56353.482400] 1fa0: 00000000 efb61fb0 c02010ac c02445cc 00000000 00000000 00000000 00000000
Apr 30 09:52:51 rpi kernel: [56353.482412] 1fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
Apr 30 09:52:51 rpi kernel: [56353.482423] 1fe0: 00000000 00000000 00000000 00000000 00000013 00000000 00000000 00000000
Apr 30 09:52:51 rpi kernel: [56353.482527] [<bfa1f738>] (hci_connect_le_scan_cleanup [bluetooth]) from [<bfa21d48>] (create_le_conn_complete+0xd8/0xdc [bluetooth])
Apr 30 09:52:51 rpi kernel: [56353.482609] [<bfa21d48>] (create_le_conn_complete [bluetooth]) from [<bfa2903c>] (hci_event_packet+0x358/0x2eb4 [bluetooth])
Apr 30 09:52:51 rpi kernel: [56353.482688] [<bfa2903c>] (hci_event_packet [bluetooth]) from [<bfa1ae88>] (hci_rx_work+0x16c/0x234 [bluetooth])
[... cut ...]
Apr 30 09:52:51 rpi kernel: [56353.482806] 1fa0:                                     00000000 00000000 00000000 00000000
Apr 30 09:52:51 rpi kernel: [56353.482818] 1fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
Apr 30 09:52:51 rpi kernel: [56353.482829] 1fe0: 00000000 00000000 00000000 00000000 00000013 00000000
Apr 30 09:52:51 rpi kernel: [56353.482840] Code: e92dd8f0 e24cb004 e52de004 e8bd4000 (e5d05012) 
Apr 30 09:52:51 rpi kernel: [56353.482885] ---[ end trace d8abc6ba1e1db467 ]---

sezeryalcin, 2020-07-03:
I saw this message as well. Wifi didn't work even if I set country in wpa_supplicant.conf file.
I had to connect via LAN and run "raspi-config" to set wifi country from menu and it worked.
Now even if I remove country= line, it still works. There must be something secret with raspi-config and rfkill.
I don't know which files it touches and I don't have time to do a full FS compare from vanilla RPI image.

mreymann, 2020-09-09:
Same problem. Stack trace:

> Sep  8 10:50:24 zero kernel: [67917.777678] 8<--- cut here ---
Sep  8 10:50:24 zero kernel: [67917.779966] Unable to handle kernel NULL pointer dereference at virtual address 00000012
Sep  8 10:50:24 zero kernel: [67917.784621] pgd = 6a69b438
Sep  8 10:50:24 zero kernel: [67917.786909] [00000012] *pgd=00000000
Sep  8 10:50:24 zero kernel: [67917.789169] Internal error: Oops: 17 [#1] ARM
Sep  8 10:50:24 zero kernel: [67917.791421] Modules linked in: algif_hash ecb algif_skcipher af_alg md4 md5 sha512_generic hmac nls_utf8 cifs gcm libarc4 aes_arm aes_generic cmac bnep hci_uart btbcm bluetooth ecdh_generic ecc libaes 8021q garp stp llc brcmfmac brcmutil sha256_generic libsha256 cfg80211 rfkill raspberrypi_hwmon bcm2835_codec(C) v4l2_mem2mem bcm2835_v4l2(C) bcm2835_isp(C) bcm2835_mmal_vchiq(C) snd_bcm2835(C) videobuf2_vmalloc videobuf2_dma_contig videobuf2_memops videobuf2_v4l2 videobuf2_common snd_pcm snd_timer snd videodev vc_sm_cma(C) mc uio_pdrv_genirq uio fixed ip_tables x_tables ipv6 nf_defrag_ipv6
Sep  8 10:50:24 zero kernel: [67917.812666] CPU: 0 PID: 32 Comm: kworker/u3:0 Tainted: G         C        5.4.51+ #1333
Sep  8 10:50:24 zero kernel: [67917.818516] Hardware name: BCM2835
Sep  8 10:50:24 zero kernel: [67917.822156] Workqueue: hci0 hci_rx_work [bluetooth]
Sep  8 10:50:24 zero kernel: [67917.825676] PC is at hci_connect_le_scan_cleanup+0x14/0x12c [bluetooth]
Sep  8 10:50:24 zero kernel: [67917.829280] LR is at create_le_conn_complete+0xd0/0xd8 [bluetooth]
Sep  8 10:50:24 zero kernel: [67917.832446] pc : [<bf3e1e10>]    lr : [<bf3e467c>]    psr: 60000013
Sep  8 10:50:24 zero kernel: [67917.835687] sp : d7637de0  ip : d7637e00  fp : d7637dfc
Sep  8 10:50:24 zero kernel: [67917.838867] r10: d3bfa700  r9 : d4882180  r8 : d3bfa7d4
Sep  8 10:50:24 zero kernel: [67917.841944] r7 : 00000000  r6 : d3bfa008  r5 : d3bfa000  r4 : d3bfa888
Sep  8 10:50:24 zero kernel: [67917.845110] r3 : bf3e45ac  r2 : 00000000  r1 : 00000000  r0 : 00000000
Sep  8 10:50:24 zero kernel: [67917.848248] Flags: nZCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment user
Sep  8 10:50:24 zero kernel: [67917.851492] Control: 00c5387d  Table: 0fea4008  DAC: 00000055
Sep  8 10:50:24 zero kernel: [67917.854726] Process kworker/u3:0 (pid: 32, stack limit = 0x18c88817)
Sep  8 10:50:24 zero kernel: [67917.857936] Stack: (0xd7637de0 to 0xd7638000)
Sep  8 10:50:24 zero kernel: [67917.861059] 7de0: d3bfa888 d3bfa000 d3bfa008 00000000 d7637e1c d7637e00 bf3e467c bf3e1e08
Sep  8 10:50:24 zero kernel: [67917.867121] 7e00: d3bfa000 d4882180 d4882cc0 0000000e d7637ecc d7637e20 bf3ebd98 bf3e45b8
Sep  8 10:50:24 zero kernel: [67917.873244] 7e20: d7637e68 d7637e6c c0013580 00000000 c0a47028 00000003 d7637e84 d7637e48
Sep  8 10:50:24 zero kernel: [67917.879395] 7e40: c00135d0 c00148a8 c00135d0 00000000 bf3db9f0 d7637f0c d7637ed0 00000000
Sep  8 10:50:24 zero kernel: [67917.885673] 7e60: 00882180 c060200d bf3e45ac 00000000 d3bfa7d4 00000008 d7637eac d7637e88
Sep  8 10:50:24 zero kernel: [67917.892304] 7e80: c00d9884 c00da9cc d3bfa7d4 d4882180 60000013 d3bfa7d4 d3bfa000 668220f8
Sep  8 10:50:24 zero kernel: [67917.899068] 7ea0: d7637ecc d3bfa6fc d3bfa6fc d3bfa888 d3bfa000 d3bfa7d4 d4882180 d3bfa700
Sep  8 10:50:24 zero kernel: [67917.906135] 7ec0: d7637f0c d7637ed0 bf3dbb4c bf3eba44 d40d7600 00000008 d7637f0c d3bfa00c
Sep  8 10:50:24 zero kernel: [67917.913524] 7ee0: c00d9884 d3bfa6fc d757fc00 00000060 00000000 d40d7600 00000008 d3bfa700
Sep  8 10:50:24 zero kernel: [67917.921027] 7f00: d7637f44 d7637f10 c003e2f8 bf3db9bc c003e998 d75fe600 d7637f44 d757fc00
Sep  8 10:50:24 zero kernel: [67917.928745] 7f20: d75fe600 d757fc14 d75fe614 c0a578c0 00000088 d75fe600 d7637f7c d7637f48
Sep  8 10:50:24 zero kernel: [67917.936481] 7f40: c003e90c c003e1d0 d7637f7c 00000000 c00433d0 d75fc300 d7608240 00000000
Sep  8 10:50:24 zero kernel: [67917.944292] 7f60: d757fc00 c003e8d8 d74e1d00 d75fc318 d7637fac d7637f80 c0044134 c003e8e4
Sep  8 10:50:24 zero kernel: [67917.952030] 7f80: ffffffff d7608240 c0044018 00000000 00000000 00000000 00000000 00000000
Sep  8 10:50:24 zero kernel: [67917.959827] 7fa0: 00000000 d7637fb0 c00090ac c0044024 00000000 00000000 00000000 00000000
Sep  8 10:50:24 zero kernel: [67917.967565] 7fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
Sep  8 10:50:24 zero kernel: [67917.975375] 7fe0: 00000000 00000000 00000000 00000000 00000013 00000000 00000000 00000000
Sep  8 10:50:24 zero kernel: [67917.983092] Backtrace:
Sep  8 10:50:24 zero kernel: [67917.987741] [<bf3e1dfc>] (hci_connect_le_scan_cleanup [bluetooth]) from [<bf3e467c>] (create_le_conn_complete+0xd0/0xd8 [bluetooth])
Sep  8 10:50:24 zero kernel: [67917.995433]  r7:00000000 r6:d3bfa008 r5:d3bfa000 r4:d3bfa888
Sep  8 10:50:24 zero kernel: [67917.999999] [<bf3e45ac>] (create_le_conn_complete [bluetooth]) from [<bf3ebd98>] (hci_event_packet+0x360/0x2ed4 [bluetooth])
Sep  8 10:50:24 zero kernel: [67918.007549]  r7:0000000e r6:d4882cc0 r5:d4882180 r4:d3bfa000
Sep  8 10:50:24 zero kernel: [67918.012106] [<bf3eba38>] (hci_event_packet [bluetooth]) from [<bf3dbb4c>] (hci_rx_work+0x19c/0x278 [bluetooth])
Sep  8 10:50:24 zero kernel: [67918.019435]  r10:d3bfa700 r9:d4882180 r8:d3bfa7d4 r7:d3bfa000 r6:d3bfa888 r5:d3bfa6fc
Sep  8 10:50:24 zero kernel: [67918.026716]  r4:d3bfa6fc
Sep  8 10:50:24 zero kernel: [67918.030650] [<bf3db9b0>] (hci_rx_work [bluetooth]) from [<c003e2f8>] (process_one_work+0x134/0x3f4)
[... cut ...]
Sep  8 10:50:24 zero kernel: [67918.095142] 7fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
Sep  8 10:50:24 zero kernel: [67918.101761] 7fe0: 00000000 00000000 00000000 00000000 00000013 00000000
Sep  8 10:50:24 zero kernel: [67918.105189]  r10:00000000 r9:00000000 r8:00000000 r7:00000000 r6:00000000 r5:c0044018
Sep  8 10:50:24 zero kernel: [67918.111689]  r4:d7608240 r3:ffffffff
Sep  8 10:50:24 zero kernel: [67918.114936] Code: e92dd8f0 e24cb004 e52de004 e8bd4000 (e5d05012)
Sep  8 10:50:24 zero kernel: [67918.118500] ---[ end trace 5d209432273c77ec ]---

rossifr, 2021-01-15:
Same problem, reconnecting a bluetooth keyboard:

Jan 15 18:51:07 4GB kernel: [435222.428754] Unable to handle kernel NULL pointer dereference at virtual address 00000100
Jan 15 18:51:07 4GB kernel: [435222.428776] pgd = cccd97ad
Jan 15 18:51:07 4GB kernel: [435222.428790] [00000100] *pgd=80000000004003, *pmd=00000000
Jan 15 18:51:07 4GB kernel: [435222.428822] Internal error: Oops: 207 [#1] SMP ARM
Jan 15 18:51:07 4GB kernel: [435222.428835] Modules linked in: snd_seq_dummy snd_seq snd_seq_device joydev hidp rfcomm bnep hci_uart btbcm serd
ev bluetooth ecdh_generic fuse 8021q garp stp llc binfmt_misc evdev sg vc4 drm_kms_helper v3d gpu_sched brcmfmac brcmutil drm drm_panel_orienta
tion_quirks sha256_generic snd_soc_core snd_bcm2835(C) cfg80211 snd_compress snd_pcm_dmaengine snd_pcm snd_timer raspberrypi_hwmon rfkill hwmon
 syscopyarea sysfillrect sysimgblt fb_sys_fops snd bcm2835_codec(C) bcm2835_v4l2(C) v4l2_mem2mem bcm2835_mmal_vchiq(C) v4l2_common videobuf2_dm
a_contig videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 videobuf2_common videodev vc_sm_cma(C) media rpivid_mem uio_pdrv_genirq uio fixed i2
c_dev ip_tables x_tables ipv6
Jan 15 18:51:07 4GB kernel: [435222.429349] CPU: 1 PID: 15257 Comm: kworker/u9:2 Tainted: G         C        4.19.75-v7l+ #1270
Jan 15 18:51:07 4GB kernel: [435222.429364] Hardware name: BCM2835
Jan 15 18:51:07 4GB kernel: [435222.429488] Workqueue: hci0 hci_rx_work [bluetooth]
Jan 15 18:51:07 4GB kernel: [435222.429523] PC is at __wake_up_common+0x6c/0x144
Jan 15 18:51:07 4GB kernel: [435222.429540] LR is at _raw_spin_unlock_irqrestore+0x6c/0x70
Jan 15 18:51:07 4GB kernel: [435222.429553] pc : [<c0269b80>]    lr : [<c099d30c>]    psr: 80000193
Jan 15 18:51:07 4GB kernel: [435222.429566] sp : d3663c70  ip : d3663bd8  fp : d3663ca4
Jan 15 18:51:07 4GB kernel: [435222.429578] r10: 00000000  r9 : 00000001  r8 : d3663cb8
Jan 15 18:51:07 4GB kernel: [435222.429591] r7 : 00000001  r6 : d72a6dc4  r5 : 00000000  r4 : 000000f4
Jan 15 18:51:07 4GB kernel: [435222.429604] r3 : 3f5f7b7d  r2 : 00000100  r1 : 00000001  r0 : 00000000
Jan 15 18:51:07 4GB kernel: [435222.429619] Flags: Nzcv  IRQs off  FIQs on  Mode SVC_32  ISA ARM  Segment user
Jan 15 18:51:07 4GB kernel: [435222.429633] Control: 30c5383d  Table: 1c635280  DAC: 55555555
Jan 15 18:51:07 4GB kernel: [435222.429648] Process kworker/u9:2 (pid: 15257, stack limit = 0xbce70ce9)
Jan 15 18:51:07 4GB kernel: [435222.429663] Stack: (0xd3663c70 to 0xd3664000)
Jan 15 18:51:07 4GB kernel: [435222.429678] 3c60:                                     00000000 00000000 20000113 d72a6dc0
Jan 15 18:51:07 4GB kernel: [435222.429694] 3c80: 00000000 d3663cb8 00000000 00000000 00000001 d8fe73ec d3663cfc d3663ca8
Jan 15 18:51:07 4GB kernel: [435222.429710] 3ca0: c0269eb8 c0269b20 00000000 d3663cb8 20000113 c1004d88 00000000 00000000
Jan 15 18:51:07 4GB kernel: [435222.429726] 3cc0: 00000000 d3663cc4 d3663cc4 3f5f7b7d c0304440 00000068 dadd0c00 00000000
Jan 15 18:51:07 4GB kernel: [435222.429742] 3ce0: 00000004 d8fe7200 00000004 d8fe73ec d3663d14 d3663d00 c0269f60 c0269e40
Jan 15 18:51:07 4GB kernel: [435222.429758] 3d00: 00000000 d3663d10 d3663d24 d3663d18 c084cfa0 c0269f48 d3663d4c d3663d28
Jan 15 18:51:07 4GB kernel: [435222.429773] 3d20: bfa0e748 c084cf68 bfa0e628 d8fe7200 db830e40 00000068 00000004 db830e40
Jan 15 18:51:07 4GB kernel: [435222.429789] 3d40: d3663d74 d3663d50 bfa01e9c bfa0e634 bfa0c7ac d8fe7200 00000004 d6b27814
Jan 15 18:51:07 4GB kernel: [435222.429806] 3d60: 00000004 db830e40 d3663e9c d3663d78 bfa097ac bfa01e68 d3663ddc 001c6180
Jan 15 18:51:07 4GB kernel: [435222.429822] 3d80: d8bb3c00 db445008 d3663dcc d3663d98 c039ae28 c039a84c 00000001 c0853dbc
Jan 15 18:51:07 4GB kernel: [435222.429838] 3da0: d3663dcc bfa2b680 c1004dd4 0000004a c1004d88 da50d0c0 db830ed8 db830ee0
Jan 15 18:51:07 4GB kernel: [435222.429854] 3dc0: d3663de4 d3663dd0 07063dbc c0390004 00000001 da50d240 d3663e04 004a0040
Jan 15 18:51:07 4GB kernel: [435222.429869] 3de0: c08550a4 c0853d48 00000000 db445000 da50d240 00000000 d3663ec4 d3663e08
Jan 15 18:51:07 4GB kernel: [435222.429885] 3e00: bf9ee01c c085506c eff32ec0 ef03e980 c1004d88 c100550c eff32e40 ef03e900
Jan 15 18:51:07 4GB kernel: [435222.429901] 3e20: d3663e84 c0e97ad8 c024ca28 ef91bc00 eff32e40 c0997428 ef03e900 00000000
Jan 15 18:51:07 4GB kernel: [435222.429916] 3e40: d3663e6c c0e97ad8 c099d2dc db4457e4 db445000 db4457d8 db445704 00000000
Jan 15 18:51:07 4GB kernel: [435222.429932] 3e60: d3663e8c d3663e70 c0304440 3f5f7b7d 60000113 db830e40 da50d0c0 0000000c
Jan 15 18:51:07 4GB kernel: [435222.429948] 3e80: db445000 db4457d8 00000002 0000000b d3663ec4 d3663ea0 bfa0c61c bfa0918c
Jan 15 18:51:07 4GB kernel: [435222.429963] 3ea0: d3663ec4 3f5f7b7d d8bb3c00 d8bb3c00 da50d0c0 db445700 d3663efc d3663ec8
Jan 15 18:51:07 4GB kernel: [435222.429979] 3ec0: bf9dfe6c bfa0c410 db445704 db445008 d3663efc d698f300 db445700 efb27a00
Jan 15 18:51:07 4GB kernel: [435222.429995] 3ee0: db6f5400 00000000 db445704 00000000 d3663f34 d3663f00 c023db40 bf9dfd28
Jan 15 18:51:07 4GB kernel: [435222.430011] 3f00: 40000093 efb27a00 efb27a00 efb27a00 d698f314 efb27a00 efb27a18 c1003d00
Jan 15 18:51:07 4GB kernel: [435222.430026] 3f20: 00000088 d698f300 d3663f7c d3663f38 c023de84 c023d9dc d3663f5c 00000000
Jan 15 18:51:07 4GB kernel: [435222.430044] 3f40: c1003d00 c1003d00 c10963d6 d3662038 d3663f7c d36c7ac0 d73a5700 00000000
[... cut ...]
Jan 15 18:51:07 4GB kernel: [435222.431295] 3fa0:                                     00000000 00000000 00000000 00000000
Jan 15 18:51:07 4GB kernel: [435222.431310] 3fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
Jan 15 18:51:07 4GB kernel: [435222.431324] 3fe0: 00000000 00000000 00000000 00000000 00000013 00000000
Jan 15 18:51:07 4GB kernel: [435222.431343] Code: e3a07000 e244400c ea000005 e284200c (e594300c) 
Jan 15 18:51:07 4GB kernel: [435222.431364] ---[ end trace f477681596618d15 ]---

No0ne, 2022-05-26:
Same happpend here, command was:
```
noone@silverpi:~ $ sudo rfcomm connect 0 98:D3:B1:xx:xx:xx
Message from syslogd@silverpi at May 26 10:42:18 ...
 kernel:[ 2159.075334] Internal error: Oops: 96000005 [#1] PREEMPT SMP

Message from syslogd@silverpi at May 26 10:42:18 ...
 kernel:[ 2159.101327] Code: 910003fd f9000bf3 aa0003f3 f940dc00 (f9402001) 

Segmentation fault
noone@silverpi:~ $
```
dmesg says:
```
noone@silverpi:~ $ sudo dmesg -c
[  206.009408] Bluetooth: RFCOMM TTY layer initialized
[  206.009455] Bluetooth: RFCOMM socket layer initialized
[  206.009483] Bluetooth: RFCOMM ver 1.11
[ 2159.067241] Unable to handle kernel NULL pointer dereference at virtual address 0000000000000040
[ 2159.069007] Mem abort info:
[ 2159.069554]   ESR = 0x96000005
[ 2159.070074]   EC = 0x25: DABT (current EL), IL = 32 bits
[ 2159.070595]   SET = 0, FnV = 0
[ 2159.071096]   EA = 0, S1PTW = 0
[ 2159.071596]   FSC = 0x05: level 1 translation fault
[ 2159.072156] Data abort info:
[ 2159.072678]   ISV = 0, ISS = 0x00000005
[ 2159.073197]   CM = 0, WnR = 0
[ 2159.073708] user pgtable: 4k pages, 39-bit VAs, pgdp=0000000011860000
[ 2159.074246] [0000000000000040] pgd=0000000000000000, p4d=0000000000000000, pud=0000000000000000
[ 2159.075334] Internal error: Oops: 96000005 [#1] PREEMPT SMP
[ 2159.075899] Modules linked in: rfcomm cmac algif_hash aes_arm64 algif_skcipher af_alg bnep hci_uart btbcm bluetooth ecdh_generic ecc g_mass_storage usb_f_mass_storage libcomposite pps_ldisc 8021q garp stp llc exfat btrfs blake2b_generic xor xor_neon zstd_compress raid6_pq dm_mod brcmfmac brcmutil raspberrypi_hwmon cfg80211 rfkill dwc2 bcm2835_v4l2(C) bcm2835_codec(C) bcm2835_isp(C) v4l2_mem2mem bcm2835_mmal_vchiq(C) videobuf2_dma_contig videobuf2_vmalloc videobuf2_memops roles videobuf2_v4l2 videobuf2_common videodev vc_sm_cma(C) mc pps_gpio uio_pdrv_genirq uio nbd drm drm_panel_orientation_quirks backlight fuse ip_tables x_tables ipv6
[ 2159.081232] CPU: 0 PID: 1457 Comm: btcheck.php Tainted: G         C        5.15.32-v8+ #1538
[ 2159.082689] Hardware name: Raspberry Pi 3 Model A Plus Rev 1.0 (DT)
[ 2159.083420] pstate: 60000005 (nZCv daif -PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[ 2159.084172] pc : rfcomm_dev_shutdown+0x20/0x50 [rfcomm]
[ 2159.084950] lr : tty_port_shutdown+0xa8/0xd8
[ 2159.085716] sp : ffffffc00a233c30
[ 2159.086459] x29: ffffffc00a233c30 x28: ffffff8002133c80 x27: 0000000000000000
[ 2159.087242] x26: 0000000000000000 x25: 0000000000000000 x24: 0000000000100000
[ 2159.088027] x23: 0000000000000000 x22: ffffff801a3eb200 x21: ffffff8019443800
[ 2159.088810] x20: ffffff800442a700 x19: ffffff800442a600 x18: 0000000000000000
[ 2159.089601] x17: 0000000000000000 x16: 0000000000000000 x15: 0000000000000000
[ 2159.090394] x14: 0000000000000000 x13: 0000000000000000 x12: 0000000000000000
[ 2159.091166] x11: 0000000000000000 x10: 0000000000000000 x9 : ffffffc00870a9d0
[ 2159.091919] x8 : ffffffc0096b8260 x7 : 0000000000000000 x6 : 000000000000003f
[ 2159.092653] x5 : 0000000000000040 x4 : 0000000000000000 x3 : ffffff800442a6f0
[ 2159.093378] x2 : 0000000000000000 x1 : ffffffc0011f48f0 x0 : 0000000000000000
[ 2159.094087] Call trace:
[ 2159.094758]  rfcomm_dev_shutdown+0x20/0x50 [rfcomm]
[ 2159.095432]  tty_port_shutdown+0xa8/0xd8
[ 2159.096074]  tty_port_close+0x58/0xb0
[ 2159.096689]  rfcomm_tty_close+0x24/0x30 [rfcomm]
[ 2159.097295]  tty_release+0x118/0x4a0
[ 2159.097870]  __fput+0x78/0x240
[ 2159.098416]  ____fput+0x18/0x28
[ 2159.098937]  task_work_run+0x90/0x168
[ 2159.099443]  do_notify_resume+0x1f4/0x470
[ 2159.099936]  el0_svc+0x58/0x60
[ 2159.100409]  el0t_64_sync_handler+0x90/0xb8
[ 2159.100872]  el0t_64_sync+0x1a0/0x1a4
[ 2159.101327] Code: 910003fd f9000bf3 aa0003f3 f940dc00 (f9402001) 
[ 2159.101799] ---[ end trace c81b55681132b2bf ]---
noone@silverpi:~ $
```
Only rebooting fixes bluetooth connections again.