raspberrypi/linux #3495: Pi4 hostapd wireless and hardware pwm conflict?

Swap-File, 2020-03-10:
**Describe the bug**
If I enable hostapd via systemctl ("sudo systemctl enable hostapd.service") reboot, and then try to use hardware PWM on pin 12 or 13 (like this: http://www.airspayce.com/mikem/bcm2835/pwm_8c-example.html), no PWM output occurs.

If I disable hostapd  ("sudo systemctl disable hostapd.service") and reboot, Hardware PWM works again. 

My temporary fix has been to keep hostapd disabled in systemctl and manually bring up hostapd ("sudo hostapd /etc/hostapd/hostapd.conf").  Then HW PWM seems to work fine.  I did notice that when starting hostapd in this way I get the message "wlan0: Could not connect to kernel driver" but it doesn't seem to affect anything negatively.

**System**
Pi4
Raspberry Pi reference 2020-02-13
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 5f884374b6ac6e155330c58caa1fb7249b8badf1, stage2
Feb 28 2020 18:45:36
Copyright (c) 2012 Broadcom
version 9520663ed4fa4ca8c2a6923a522bff85fcdedef5 (clean) (release) (start_x)
Linux raspberrypi 4.19.106-v7l+ #1297 SMP Tue Feb 25 13:19:54 GMT 2020 armv7l GNU/Linux

Ideas?

Swap-File, 2020-03-11:
I found that by spamming the following functions PWM eventually starts working with hostapd running in the background.

```
			bcm2835_pwm_set_mode(12, 1, 1); //PWM_CHANNEL 0
			bcm2835_pwm_set_range(0, 1024);//PWM_CHANNEL 0 Set Range to 1024
			
			bcm2835_pwm_set_mode(13, 1, 1);  //PWM_CHANNEL 1
			bcm2835_pwm_set_range(1, 1024);   //PWM_CHANNEL 1 Set Range to 1024
```

I'm assuming there must be some interaction between how hostapd and bcm2835 write to the peripheral registers?  The issue might be with bcm2835 and not the wireless driver/host apd.  I'm not really familiar with it, but will poke around.

pelwell, 2020-03-11:
The PWM library you are using is bare-metal code that bypasses the kernel. Use it at your own risk - we won't investigate any problems you may encounter while doing so.

closed 2020-03-11 by pelwell

Swap-File, 2020-03-11:
Is there a way to access the Pi's two hardware PWM without writing to the registers directly?  I haven't seen any alternative options, but am open to suggestions.

pelwell, 2020-03-11:
There is a Linux pwm driver, which you can enable using the pwm and pwm-2chan overlays (`dtoverlay -h pwm`, etc.). That should then create a sysfs interface giving general PWM control (see https://developer.ridgerun.com/wiki/index.php?title=Linux_PWM_Pulse_Width_Modulator#Control_PWM_through_sysfs for an example), but if you are wanting to "abuse" the PWM to do something else then you're out of luck.

Swap-File, 2020-03-11:
Thanks for the link, I tried out pwm-2chan and it seemed to work well.  

I did notice that "dtoverlay -h pwm-2chan" mentioned "4) Currently the clock must have been enabled and configured by other means."

In practice I didn't need to configure the clock, it just seemed to work out of the box.  Is this clock configuration not needed on a Pi?

Another page I found mentioned configuring the clock with a helper program.... writes directly to the registers.  Which seems like it would be counter productive if the goal is to avoid writing directly to registers. 

https://librpip.frasersdev.net/peripheral-config/pwm0and1/

pelwell, 2020-03-11:
Unlike previous Pis, Pi4 (thanks to its use of BCM2711) has two independent 2-channel PWM blocks. PWM1 is available on GPIOs 40, 41 and 45 (i.e. the audio outputs), leaving PWM0 for GPIOs 12, 13, 18 & 19 (on the 40-pin header).

I'm guessing that you still have "dtparam=audio=on" in config.txt, and that is enabling the clock for you but not interfering.

By the way, can you confirm explicitly that using PWM this way doesn't interfere with hostapd?

Swap-File, 2020-03-11:
I can verify that using PWM through pwm-2chan does not affect hostapd or the wireless driver, they both work well together.  Oddly enough I haven't seen the hostapd stability problems I saw other people mention in other github issues, my problem was related more to hostapd affecting the pwm.

I can also report I have no idea where the pwm clock is coming from.  I commented out the following items in config.txt, and it just keeps working with the overlay.  The only thing I have enabled at this point are my video card settings, and the pwm overlay.
```
#force_turbo=1
#dtoverlay=fe-pi-audio
#start_x=1
#dtparam=i2c_arm=on
#dtparam=i2s=on
#dtparam=spi=on
#dtparam=audio=on
```
I also tried setting them explicitly to off instead of just commenting them out.  Maybe the pwm clock is just on by default now?

I also noticed that sometimes the bare metal library would seem to work after a reboot, and then stop working after a hard power cycle.  Is it possible the hw pwm settings can persist across reboots?  I may need to re-run some of my testing if this is the case.