raspberrypi/linux #3496: DMA and I2S driver/DT changes between Busta and Stretch

flatmax, 2020-03-10:
A user has tested I2S differences between stretch and busta w.r.t. mutlichannel audio :
https://github.com/Audio-Injector/Octo/issues/20

The user tests a modified ALSAFullDuplexTest.C file from here with the variable ch=8 :
https://github.com/flatmax/gtkiostream/blob/master/test/ALSAFullDuplexTest.C
They compile it with :
`g++ -o ALSAFullDuplexTest ALSAFullDuplexTest.C 'pkg-config --cflags --libs gtkIOStream'`

On Busta it works as expected channels are not remapped. On Stretch channels are remapped randomly between executions. Only rebooting changes that behaviour.

Reported for a Pi 4.

Is it possible to link to the different bcm2835-i2s.c files which are used in Busta and Stretch ?
It is also possible that something has changed w.r.t. DMA setup.

flatmax, 2020-03-10:
The history of the bcm2835-i2s.c file shows no change since Jan 2018 :
https://github.com/raspberrypi/linux/commits/rpi-4.19.y/sound/soc/bcm/bcm2835-i2s.c

Commits on Jan 18, 2018
Commits on Nov 11, 2017

pelwell, 2020-03-10:
Locating a last known good kernel version would be a good start. rpi-update allows the kernel and firmware to be reverted to earlier versions. Be careful not to revert to the point before the support for your Pi was added - it may be better to use a Pi 3 for this testing.

flatmax, 2020-03-10:
Thanks for the guidance - future debugs will use that approach.

We are solving the problem of resetting channel synchronisation by reloading the modules :
https://github.com/Audio-Injector/Octo/issues/14#issuecomment-475045553