raspberrypi/linux #3501: USB broken on 5.5 on Pi 4 (regression from a few days ago)

hvenev, 2020-03-12:
The device is a Raspberry Pi 4 with 4 GB of RAM, running a 64-bit kernel produced by merging the rpi-5.5.y branch with upstream v5.5.9.

Both USB 2 (using dwc_otg) and USB 3 (using xhci_pci) are broken, as of 2e995e7bc975b9399bff2dc8ebd32697b6fee9ec. They work fine with 4c4958f903d8644a00b4ea20889f4356e698a917.

Nothing appears in the kernel log when a device is plugged in, but there are some errors:

```
pi4 ~ # dmesg | grep 'Rasp\|usb\|xhci\|dwc'
[    0.000000] Machine model: Raspberry Pi 4 Model B Rev 1.1
[    0.198745] usbcore: registered new interface driver usbfs
[    0.198789] usbcore: registered new interface driver hub
[    0.198875] usbcore: registered new device driver usb
[    0.315595] dwc_otg: version 3.00a 10-AUG-2012 (platform bus)
[    0.315696] dwc_otg: FIQ enabled
[    0.315700] dwc_otg: NAK holdoff enabled
[    0.315703] dwc_otg: FIQ split-transaction FSM enabled
[    0.315709] Module dwc_common_port init
[    0.316084] usbcore: registered new interface driver cdc_acm
[    0.316136] usbcore: registered new interface driver cdc_wdm
[    0.316169] usbcore: registered new interface driver usbtmc
[    0.320207] usbcore: registered new interface driver usbhid
[    0.320246] usbhid: USB HID core driver
[    3.510018] xhci_hcd 0000:01:00.0: xHCI Host Controller
[    3.510934] xhci_hcd 0000:01:00.0: new USB bus registered, assigned bus number 1
[    3.511561] xhci_hcd 0000:01:00.0: hcc params 0x002841eb hci version 0x100 quirks 0x0000001000000890
[    3.514878] usb usb1: New USB device found, idVendor=1d6b, idProduct=0002, bcdDevice= 5.05
[    3.514913] usb usb1: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[    3.514931] usb usb1: Product: xHCI Host Controller
[    3.514945] usb usb1: Manufacturer: Linux 5.5.9-3-hvenev-pi4.aarch64 xhci-hcd
[    3.514959] usb usb1: SerialNumber: 0000:01:00.0
[    3.523494] xhci_hcd 0000:01:00.0: xHCI Host Controller
[    3.523894] xhci_hcd 0000:01:00.0: new USB bus registered, assigned bus number 2
[    3.523940] xhci_hcd 0000:01:00.0: Host supports USB 3.0 SuperSpeed
[    3.524039] xhci_hcd 0000:01:00.0: WARNING: Host System Error
[    3.524265] usb usb2: New USB device found, idVendor=1d6b, idProduct=0003, bcdDevice= 5.05
[    3.524286] usb usb2: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[    3.524310] usb usb2: Product: xHCI Host Controller
[    3.524324] usb usb2: Manufacturer: Linux 5.5.9-3-hvenev-pi4.aarch64 xhci-hcd
[    3.524339] usb usb2: SerialNumber: 0000:01:00.0
[    3.727102] usb usb1-port1: couldn't allocate usb_device
```

pelwell, 2020-03-13:
What you seem to be saying is "I took your tree, and changed it, and now it's broken".

1. Your edited dmesg log misses lots of vital information. In particular I'd like to see the CMA allocation and PCI address mappings, but really - just attach the whole thing, it saves so much time.

2. Test https://github.com/raspberrypi/linux/commit/2e995e7bc975b9399bff2dc8ebd32697b6fee9ec as is, and report back. It works for me, but if it doesn't work for you I'll need the full dmesg output and contents of configt.txt.

3. If you are going to make modifications you're not guaranteed to get help, but the very least you'll need to provide is a pointer to a github fork with your modifications.

hvenev, 2020-03-13:
Never mind, the exact same kernel _binary_ that produced the messages above now seems to be working.

closed 2020-03-13 by hvenev