raspberrypi/linux #3502: fe-pi audio line in/out not working on 4.19.97 (raspberrypi-kernel_1.20200212-1)

preparationh67, 2020-03-13:
**Describe the bug**
After upgrading a Pi 4 B Raspbian system to raspberrypi-kernel_1.20200212-1 from the previous version audio capture from the line input and audio play from the line output stopped working when using the Fe-Pi audio hat. Both devices are still visible to alsa and no errors are reported by applications trying to use them, but there is only silence on the line input and output. Output from the headphone port works as expected. Downgrading the kernel package to the old cached version, raspberrypi-kernel_1.20190925+1-1, restored the functionality of the line input and output. raspberrypi-firmware did **not** have to be downgraded. 

**To reproduce**
Attach and enable a fe-pi audio hat. Update to the system to use raspberrypi-kernel_1.20200212-1 and attempt to capture audio from the line input using arecord, ffmpeg, or audacity or attempt to play audio via the line out instead of the headphone out. 

**Expected behaviour**
Audio is captured or played through the audio cards line in and out when using appropriate software

**Actual behaviour**
No sound output via the line output and no audio is captured from the line input. 

**System**

raspinfo: https://pastebin.com/WrH5Gv1J

* Which model of Raspberry Pi? Pi 4 B
* Which OS and version (`cat /etc/rpi-issue`)? 175dfb027ffabd4b8d5080097af0e51ed9a4a56c, stage4
* Which firmware version (`vcgencmd version`)?
Feb 12 2020 12:37:37 
Copyright (c) 2012 Broadcom
version c3c8dbdf147686fb0c3f32aece709d0653368810 (clean) (release) (start_x)
* Which kernel version (`uname -a`)? 4.19.97-v7l+

**Logs**
Log output from dmesg when broken: https://pastebin.com/trVRjpq2
Log output from dmesg when working: https://pastebin.com/5Bxun60p

**Additional context**
Presumably other audio cards using the SGTL5000 codec have the same issue, but I have no way to confirm that.

pelwell, 2020-03-13:
@Fe-Pi Is this something you can reproduce?

Fe-Pi, 2020-03-13:
This issue has been fixed.
https://github.com/raspberrypi/linux/issues/3468
Let me know if there still is an issue.

jaytee117, 2020-04-02:
I am having the same issue, what was the fix?

joerango, 2020-12-30:
This is happening again on my Raspberry Pi 3 Model B Rev 1.2 with the latest Raspberry Pi buster OS image

5.4.79-v7+ #1373 SMP Mon Nov 23 13:22:33 GMT 2020 armv7l GNU/Linux

Raspberry Pi reference 2020-12-02
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, cce27bd6f44a3b2e83855645986b3e21f771e852, stage2

Nov 25 2020 17:12:03 
Copyright (c) 2012 Broadcom
version 2e6292a5304b95ecfd700fc5ab2b793402056332 (clean) (release) (start)

@Fe-Pi I checked the kernel sources and the commit you mentioned is in the code.

Dloranger, 2021-02-13:
@Fe-Pi, @pelwell,

I think I might be running into a similar issue today (first update to kernel in a LONG time).  I only did the update to get support for the 4B/8G hardware.  This is running a lite distro (completely headless)

#uname -a
Linux openrepeater 5.4.83-v7l+ #1379 SMP Mon Dec 14 13:11:54 GMT 2020 armv7l GNU/Linux

# dmesg | grep sgtl5000
[    7.203881] sgtl5000 1-000a: sgtl5000 revision 0x11

# dmesg | grep fe-pi
#

The hardware is seen an loaded in the device tree as far as I can tell

# i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- UU -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: UU -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --

The hardware hat uses the SGTL5000 codec with line-in/line-out ports and everything has been working for years prior to the update.  Now I get no audio at all and the SVXLINK package that consumes the audio behaves like it is not getting any audio packets at all on the line-in port (long story how I know this, but this is the symptom I have seen before).  This has been reproduced on multiple pieces of hardware both rpi (4b) and hats.

Further, _running speaker-test -c 2 -t sine -f 500_ never toggles away from channel 0, so it appears the output is broken there as well.

Any ideas what might be the issue?

Dloranger, 2021-03-13:
I think there is still something not quite right here, but I was able to get it working finally.  

https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=306337