raspberrypi/linux #3507: "error -22 whilst initialising sdio card" showing while boot

yeochoony, 2020-03-14:
[image]

This happens on every boot on my Raspberry Pi zero W.
But it still boots up to the GUI screen and functions properly except wifi.
By the way, Bluetooth works fine. 
I tried clean install of the latest version of the Raspbian.
Also tried some various brands of SD cards to check if it was a SD card fault, but it still shows the message everytime.

Anyone has the solution?

pelwell, 2020-03-14:
How long have you had the ZeroW? Has the WiFi ever worked? I'm thinking this might be a hardware fault.

yeochoony, 2020-03-14:
> How long have you had the ZeroW? Has the WiFi ever worked? I'm thinking this might be a hardware fault.

Got it from Pimoroni in 2017, but never used it until I discovered this issue yesterday.

pelwell, 2020-03-14:
Even if it had worked before, I'd still think broken hardware. We do sometimes introduce software problems, but for it to be in two versions of Raspbian and for us not to have heard of it makes that unlikely in this case.