raspberrypi/linux #3509: DWC2 host-mode driver corrupts data if there is more than 1GB RAM available

timg236, 2020-03-20:
**Describe the bug**
The DWC2 driver or USB framework seems to fail to configure bounce buffers correctly on systems with more than 1GB of RAM.

**To reproduce**
* Raspberry Pi4 4GB - power via GPIO pins
* Enable dwc2 host mode driver
* Using a USB type C adaptor connect a USB mass storage files.
* Copy a file from the MSD device to the sd card.
* Sync and unmount sd-card
* Compute SHA1 of the file and observe that it's wrong

If total_mem=1024 in config.txt then the problem stops occuring.

The disk cache can make things confusing so it's generally easier to follow if you sync and umount disks between steps.

**Expected behaviour**
Add a clear and concise description of what you expected to happen.

**Actual behaviour**
Add a clear and concise description of what actually happened.

* Which model of Raspberry Pi? e.g. Pi3B+, PiZeroW
Raspberry Pi 4 Model B 2GB or 4GB

**Additional context**
Any 4.19 kernel seems to have this issue.

P33M, 2020-03-20:
The dwc core can't access memory above 1GB as it's a legacy master.

There's already some DMA buffer manipulation in dwc2_map_urb_for_dma() - checking alignment for the start of the buffer. Could be expanded to bolt on something like brcmstb pcie does for the 3GB limit?

timg236, 2020-03-20:
Ah, I was hoping that this would be sorted out by the Linux USB framework before it passes the URB to DWC. Is this a limit of the implementation on Pi or does this not work on any SOC with that USB IP ?

P33M, 2020-03-20:
Hmm - I seem to remember a workaround that we did for day 1 was to set the DMA zone to 1GB - 
https://github.com/raspberrypi/linux/blob/rpi-4.19.y/arch/arm/mach-bcm/board_bcm2835.c#L129

So on bcm2711 allocations for DMAbufs should come from the bottom 1GB. Is your kernel config standard? 64-bit will turn off LPAE.

timg236, 2020-03-20:
It's a regular 32-bit kernel / Raspbian

pelwell, 2020-03-20:
Does it work on the 5.4 kernel? It's not clear to me at the moment how this could ever have worked on 4.19 (which doesn't have SWIOTLB).

timg236, 2020-03-20:
I haven't tried that yet, but will give it a go.

pelwell, 2020-03-27:
DId you manage to try with rpi-5.4.y?

timg236, 2020-03-30:
I haven't done anything for the last week or so.

timg236, 2020-04-01:
I've tested this with a Pi4 ModelB 4GB with rpi-5.4.y latest, top of tree FW and dtoverlay=dwc2,dr_mode=host using a USB C to adaptor and the same USB SD card reader which I had in the office.
As far as I can tell the 5.4 kernel resolves this issue. Tested by copying a directory of JPEGs unmount md5sum repeat etc

pelwell, 2020-04-01:
Given that we're just about to retire 4.19, and backporting SWIOTLB is non-trivial, I'd like to mark this as a won't-fix, fact-of-life etc.

timg236, 2020-04-01:
fine by me

closed 2020-04-01 by pelwell