raspberrypi/linux #3512: sched_netem latency always shows MAX_UINT

kitlaan, 2020-03-20:
**Describe the bug**
When dumping the qdisc tables (using `tc`), the latency and jitter values always seem to show as 274.9s, no matter what value is input.  Digging further using pyroute2, and dumping the raw netlink packet, the value being sent is 0xFFFFFFFF (aka MAX_UINT).

Looking at the kernel, we have this code:
https://github.com/raspberrypi/linux/blob/06606627043f72d22881563d485268fec2acd56d/net/sched/sch_netem.c#L1116-L1119

Realizing that `psched_tdiff_t` is a signed-long, then we're casting `UINT_MAX` to a signed 32-bit value, aka negative. So this code will always return MAX_UINT!

JamesH65, 2020-03-20:
Not entirely sure, but I think that is upstream code so the bug should therefore be reported on the netdev mailing list.

pelwell, 2020-03-20:
I don't follow your logic. min_t is a typed minimum function - it should be returning the minimum of PSCHED_NS2TICKS(q->latency) and UINT_MAX, i.e. the latency value converted to ticks, but capped to not exceed UINT_MAX.

pelwell, 2020-03-22:
Yes, I think you're correct - the disassembly agrees. As @JamesH65 said, this is a bug that appears to affect all 32-bit platforms, so should be reported to the Linux maintainers. I would start by emailing Stephen Hemminger <stephen@networkplumber.org> and David S. Miller <davem@davemloft.net> with a description of the problem.