raspberrypi/linux #3513: [drm:vc4_hdmi_bind] *ERROR* Failed to get pixel clock

jameshilliard, 2020-03-21:
**Describe the bug**
vc4_hdmi driver does not appear to appropriately handle logging for `-EPROBE_DEFER` errors when trying to get the pixel clock.

**To reproduce**
Build and run the kernel with built in(non-module) vc4 drm driver.

**Expected behaviour**
Kernel driver should ignore non-fatal `-EPROBE_DEFER` errors or load the vc4 driver only after the pixel clock is available.

**Actual behaviour**
The vc4 kernel driver spams dmesg with `-EPROBE_DEFER` errors.

**System**
* Which model of Raspberry Pi? 
Raspberry Pi 3B
* Which OS and version (`cat /etc/rpi-issue`)?
Buildroot master
* Which firmware version (`vcgencmd version`)?
commit: [01ecfd2ba2b7cf3a2f4aa75ada895ee4a3e729f5](https://github.com/raspberrypi/firmware/tree/01ecfd2ba2b7cf3a2f4aa75ada895ee4a3e729f5)
* Which kernel version (`uname -a`)?
`Linux buildroot 4.19.97-v8 #1 SMP PREEMPT Fri Mar 20 17:43:53 MDT 2020 aarch64 GNU/Linux`
commit: [64d0a9870ac14d5eb5253f67d984ae348eec1393](https://github.com/raspberrypi/linux/tree/64d0a9870ac14d5eb5253f67d984ae348eec1393)

**Logs**
```
[    0.348120] [drm:vc4_hdmi_bind] *ERROR* Failed to get pixel clock
[    0.348204] vc4-drm soc:gpu: failed to bind 3f902000.hdmi (ops vc4_hdmi_ops): -517
[    0.348392] vc4-drm soc:gpu: master bind failed: -517
[    0.846746] vc4-drm soc:gpu: failed to bind 3f902000.hdmi (ops vc4_hdmi_ops): -517
[    0.846950] vc4-drm soc:gpu: master bind failed: -517
[    0.928787] vc4-drm soc:gpu: failed to bind 3f902000.hdmi (ops vc4_hdmi_ops): -517
[    0.929026] vc4-drm soc:gpu: master bind failed: -517
[    0.932427] vc4-drm soc:gpu: failed to bind 3f902000.hdmi (ops vc4_hdmi_ops): -517
[    0.932588] vc4-drm soc:gpu: master bind failed: -517
[    1.004792] vc4-drm soc:gpu: failed to bind 3f902000.hdmi (ops vc4_hdmi_ops): -517
[    1.005037] vc4-drm soc:gpu: master bind failed: -517
[    1.397344] vc4-drm soc:gpu: failed to bind 3f902000.hdmi (ops vc4_hdmi_ops): -517
[    1.397582] vc4-drm soc:gpu: master bind failed: -517
[    1.775578] vc4-drm soc:gpu: failed to bind 3f902000.hdmi (ops vc4_hdmi_ops): -517
[    1.775834] vc4-drm soc:gpu: master bind failed: -517
[    3.495448] vc4-drm soc:gpu: bound 3f902000.hdmi (ops vc4_hdmi_ops)
[    3.495913] vc4-drm soc:gpu: bound 3f806000.vec (ops vc4_vec_ops)
[    3.496217] vc4-drm soc:gpu: bound 3f004000.txp (ops vc4_txp_ops)
[    3.496358] vc4-drm soc:gpu: bound 3f400000.hvs (ops vc4_hvs_ops)
[    3.497286] vc4-drm soc:gpu: bound 3f206000.pixelvalve (ops vc4_crtc_ops)
[    3.502060] vc4-drm soc:gpu: bound 3f207000.pixelvalve (ops vc4_crtc_ops)
[    3.502962] vc4-drm soc:gpu: bound 3f807000.pixelvalve (ops vc4_crtc_ops)
[    3.548116] vc4-drm soc:gpu: bound 3fc00000.v3d (ops vc4_v3d_ops)
[    3.556721] [drm] Initialized vc4 0.0.0 20140616 for soc:gpu on minor 0
[    3.556805] [drm] Supports vblank timestamp caching Rev 2 (21.10.2013).
[    3.556843] [drm] Driver supports precise vblank timestamp query.
[    3.763082] vc4-drm soc:gpu: fb0: DRM emulated frame buffer device
```

**Additional context**
The driver seems to load fine once the clock is available but during early boot spams the kernel logs with `-EPROBE_DEFER` related error messages when trying to load the vc4 driver before the clock is initialized. I assume we need to either somehow wait to load the vc4 driver until the clock is initialized or otherwise silence the spurious `-EPROBE_DEFER` log messages.

6by9, 2020-04-05:
Both messages come from the core component handling code, not vc4
"master bind failed"  - https://elixir.bootlin.com/linux/latest/source/drivers/base/component.c#L259
"failed to bind ...." - https://elixir.bootlin.com/linux/latest/source/drivers/base/component.c#L613

jameshilliard, 2020-04-11:
@6by9 Yeah, some of those errors were coming from core component code while the first error(`[drm:vc4_hdmi_bind] *ERROR* Failed to get pixel clock`) was coming from vc4.

I wrote a patch fixing the vc4 issue in #3538 along with the component issue in #3539.

6by9, 2020-04-11:
Thanks. I totally agree with the patches.

My only reservation is that particularly drivers/base/component.c is an unmodified file from upstream.
Ideally both patches should be submitted upstream and can then be cherry-picked back (if not marked for stable). Having said that, there are significant changes for vc4 in progress (merged on rpi-5.4.y) that are likely to create conflicts against this. Up to @pelwell whether he accepts the maintenance overhead.

pelwell, 2020-04-11:
I normally avoid patching unmodified upstream files, but I loathe these unavoidable EPROBE_DEFER errors, so I'm willing to risk some future maintenance effort.

jameshilliard, 2020-04-11:
> Ideally both patches should be submitted upstream and can then be cherry-picked back (if not marked for stable).

I already submitted them upstream here:
https://patchwork.kernel.org/patch/11483959/
https://patchwork.kernel.org/patch/11483961/

jameshilliard, 2020-04-11:
Fixed in #3538 and #3539.

closed 2020-04-11 by jameshilliard

jameshilliard, 2020-04-29:
> I normally avoid patching unmodified upstream files, but I loathe these unavoidable EPROBE_DEFER errors, so I'm willing to risk some future maintenance effort.

FYI #3539 has been merged for upstream inclusion [here](https://git.kernel.org/pub/scm/linux/kernel/git/gregkh/driver-core.git/commit/?h=driver-core-linus&id=35a672363ab3e8dfe4ebcadb4dd0b2d06bb85ebe) and should be getting included in a mainline kernel release in the near future.