raspberrypi/linux #3514: mcp251x spi0.0: Probe failed, err=16

godspeed5, 2020-03-22:
I am trying to set up the MCP2515 module with the Raspberry Pi 4B using this guide, the model is also the same(https://www.raspberrypi.org/forums/viewtopic.php?t=141052). The exception is only that I am using level converters for 5 to 3.3v conversion (since the mcp chip needs 5v). This setup has worked for me before but now doesn't.
`dmesg | grep mcp` gives:
```
[    5.158349] mcp251x spi0.0: MCP251x didn't enter in conf mode after reset
[    5.158393] mcp251x spi0.0: Probe failed, err=16
[    5.158453] mcp251x: probe of spi0.0 failed with error -16

```
`lsmod | grep mcp` gives:
```
mcp251x                20480  0
can_dev                28672  1 mcp251x
```
`uname -a` gives:
`Linux raspberrypi 4.19.97-v7l+ #1294 SMP Thu Jan 30 13:21:14 GMT 2020 armv7l GNU/Linux`
the spi portion my /boot/config.txt looks like:

```
dtparam=spi=on
dtoverlay=mcp2515-can0,oscillator=8000000,interrupt=12
```

PS: I had err=19 before and decided to make a GCB to rule out connections as the problem, but now I have err=16 for which I haven't been able to find a solution anywhere.

pelwell, 2020-03-23:
Error (-)19 is ENODEV - it couldn't find the hardware.

Error (-)16 is EBUSY. It is returned explicitly in 4 places in the driver, including immediately after logging the message `MCP251x didn't enter in conf mode after reset`. The trigger for this error is a timeout waiting for the CONF bit in the REQOP register after issuing the RESET instruction - if it hasn't appeared in a second then it errors out.

The driver doesn't appear to have a way to dump the register contents, but you could try enabling the system tracing. Add the following to your cmdline.txt file (anywhere you like, but keeping it all in a single line):
```
trace_event=spi:*
```
Reboot, then grab the tracing output with:
```
$ sudo cat /sys/kernel/debug/tracing/trace > spi_trace.txt
```
You can then upload the trace somewhere and post a link, or try an attachment here.

godspeed5, 2020-03-23:
[spi_trace.txt](https://github.com/raspberrypi/linux/files/4370162/spi_trace.txt)
I've attached the spi_trace.txt file above. There seem to be a lot of lines saying spi_controller_busy which matches the EBUSY deduction, but I have no idea how to fix this.

pelwell, 2020-03-23:
`spi_controller_busy` in that context just means that it has something to do, but a few lines later it returns to `spi_controller_idle` - that isn't the cause of the problem. It's a shame we aren't running the 5.4 kernel yet because it also dumps out the bytes being sent and received.

I think this is a hardware problem - a signal not being pulled correctly, a missing clock, something like this. The ideal debugging tool is a logic analyser of some kind, but they don't come cheap. You can at least check all the idle voltage levels with a multimeter (or perhaps just `raspi-gpio get`).

godspeed5, 2020-03-23:
The following is the output for raspi-gpio get. I used `watch -n 0.1` to see the variation with time, but only GPIO 34 shows any change with time. everything else remains the same
```
BANK0 (GPIO 0 to 27):
GPIO 0: level=1 fsel=0 func=INPUT pull=UP
GPIO 1: level=1 fsel=0 func=INPUT pull=UP
GPIO 2: level=1 fsel=4 alt=0 func=SDA1 pull=UP
GPIO 3: level=1 fsel=4 alt=0 func=SCL1 pull=UP
GPIO 4: level=1 fsel=0 func=INPUT pull=UP
GPIO 5: level=1 fsel=0 func=INPUT pull=UP
GPIO 6: level=1 fsel=0 func=INPUT pull=UP
GPIO 7: level=1 fsel=1 func=OUTPUT pull=UP
GPIO 8: level=1 fsel=1 func=OUTPUT pull=UP
GPIO 9: level=1 fsel=4 alt=0 func=SPI0_MISO pull=DOWN
GPIO 10: level=1 fsel=4 alt=0 func=SPI0_MOSI pull=DOWN
GPIO 11: level=0 fsel=4 alt=0 func=SPI0_SCLK pull=DOWN
GPIO 12: level=0 fsel=0 func=INPUT pull=DOWN
GPIO 13: level=0 fsel=0 func=INPUT pull=DOWN
GPIO 14: level=1 fsel=0 func=INPUT pull=NONE
GPIO 15: level=1 fsel=0 func=INPUT pull=UP
GPIO 16: level=0 fsel=0 func=INPUT pull=DOWN
GPIO 17: level=0 fsel=0 func=INPUT pull=DOWN
GPIO 18: level=0 fsel=0 func=INPUT pull=DOWN
GPIO 19: level=0 fsel=0 func=INPUT pull=DOWN
GPIO 20: level=0 fsel=0 func=INPUT pull=DOWN
GPIO 21: level=0 fsel=0 func=INPUT pull=DOWN
GPIO 22: level=0 fsel=0 func=INPUT pull=DOWN
GPIO 23: level=0 fsel=0 func=INPUT pull=DOWN
GPIO 24: level=0 fsel=0 func=INPUT pull=DOWN
GPIO 25: level=0 fsel=0 func=INPUT pull=DOWN
GPIO 26: level=0 fsel=0 func=INPUT pull=DOWN
GPIO 27: level=0 fsel=0 func=INPUT pull=DOWN
BANK1 (GPIO 28 to 45):
GPIO 28: level=1 fsel=2 alt=5 func=RGMII_MDIO pull=UP
GPIO 29: level=0 fsel=2 alt=5 func=RGMII_MDC pull=DOWN
GPIO 30: level=0 fsel=7 alt=3 func=CTS0 pull=UP
GPIO 31: level=0 fsel=7 alt=3 func=RTS0 pull=NONE
GPIO 32: level=1 fsel=7 alt=3 func=TXD0 pull=NONE
GPIO 33: level=1 fsel=7 alt=3 func=RXD0 pull=UP
GPIO 34: level=0 fsel=7 alt=3 func=SD1_CLK pull=NONE
GPIO 35: level=1 fsel=7 alt=3 func=SD1_CMD pull=UP
GPIO 36: level=1 fsel=7 alt=3 func=SD1_DAT0 pull=UP
GPIO 37: level=1 fsel=7 alt=3 func=SD1_DAT1 pull=UP
GPIO 38: level=1 fsel=7 alt=3 func=SD1_DAT2 pull=UP
GPIO 39: level=1 fsel=7 alt=3 func=SD1_DAT3 pull=UP
GPIO 40: level=0 fsel=4 alt=0 func=PWM1_0 pull=NONE
GPIO 41: level=0 fsel=4 alt=0 func=PWM1_1 pull=NONE
GPIO 42: level=0 fsel=1 func=OUTPUT pull=UP
GPIO 43: level=1 fsel=0 func=INPUT pull=UP
GPIO 44: level=1 fsel=0 func=INPUT pull=UP
GPIO 45: level=1 fsel=0 func=INPUT pull=UP
BANK2 (GPIO 46 to 53):
GPIO 46: level=0 fsel=0 func=INPUT pull=DOWN
GPIO 47: level=0 fsel=0 func=INPUT pull=DOWN
GPIO 48: level=0 fsel=0 func=INPUT pull=DOWN
GPIO 49: level=0 fsel=0 func=INPUT pull=DOWN
GPIO 50: level=0 fsel=0 func=INPUT pull=DOWN
GPIO 51: level=0 fsel=0 func=INPUT pull=DOWN
GPIO 52: level=0 fsel=0 func=INPUT pull=DOWN
GPIO 53: level=0 fsel=0 func=INPUT pull=DOWN

```

pelwell, 2020-03-23:
That looks OK.

We're planning to push a 5.4 kernel to our `next` firmware branch fairly soon - I'll drop a line here when we do so you can get some better tracing.

pelwell, 2020-03-23:
Or you can build your own - it doesn't take so long, and there are tried-and-tested instructions here: https://www.raspberrypi.org/documentation/linux/kernel/building.md

godspeed5, 2020-03-24:
I didnt't get you, does the output mean the hardware side isn't the issue?

pelwell, 2020-03-24:
Tracing from the 4.19 kernel doesn't contain enough information to make any sensible diagnosis. The 5.4 kernel is better.

haribole, 2020-08-25:
Hi godspeed5, any progress on this? Were you able to fix it?

godspeed5, 2020-08-30:
Unfortunately not yet

vbass0903, 2020-09-01:
I found a workaround for this problem if you want to get the MCP2515 module to interface with your Raspberry Pi, but it does not address the root issue of the Pi not recognizing the MCP2515 module while using a level shifter. 

Rather than use a level shifter, I separated the power between the MCP2515 and TJA1050. I did not cut a trace on the bottom of the board, but instead altered the VCC pin of the MCP2515 chip itself. Here is a [video](https://youtu.be/_v3GumyXbhw?t=410) showing the process. In the video he connects the MCP2515 VCC to a pin he soldered on the board, but you could just solder a jumper wire to it as well. I double-checked the voltages for all of the MCP2515 module output pins before connecting them to the Pi to make sure none of them were 5V and the highest I measured was 3.4V. So far, none of my pins have been fried.

From here I just followed this [guide](https://www.beyondlogic.org/adding-can-controller-area-network-to-the-raspberry-pi/) and my setup connected successfully.

JulianAssmann, 2020-10-22:
Are there any updates on this? I'm currently encountering the same problem. 
Linux 5.4.51

mouminali97, 2021-01-25:
> Unfortunately not yet

Any progress? I am having the exact same problem and am using level conversion as well. Could it be the level conversion that is the problem ?

mouminali97, 2021-01-26:
I got it working after eliminating the level conversion circuitry and separating the power between the MCP2515 and TJA1050 by drilling a hole behind the PCB. So it seems the level conversion was the problem.

sethumadhav96, 2021-10-07:
Hello 
I am Trying to Interface MCP 2515 with the  Raspberry Pi Compute Module 4.  Initially I was getting Most Of the Errors (-19, -16, -110) . 
Then I modified the mcp251x.c driver Code with the below code to Allow More time after a reset.(As suggested by pelwell in a git forum)      **Kernel 5.4**
      
**do {
		mdelay(MCP251X_OST_DELAY_MS);
		reg = mcp251x_read_reg(spi, CANSTAT);
	} while (!reg && retries--);**

Now I am getting the ERR -16 . No other errors so far.
I am implementing it on the SPI1 bus, Also Tried on the SPI0, but no luck..

My config.txt configuration

**dtoverlay=spi1-1cs,cs0_spidev=off
dtoverlay=mcp2515-can1-0,oscillator=8000000,interrupt=12,spimaxfrequency=500000**
(   
  1. I tried changing the spimaxfrequency to almost all the combinations
  2. Now Iam using a dtbo file as suggested by pelwell in one of the git forum to use  spi1 bus for mcp2515    
)
I traced the spi message Iam attaching that one also with this:

[spi_trace.txt](https://github.com/raspberrypi/linux/files/7302112/spi_trace.txt)

As Iam a beginner in this field . Iam kind of stuck here. Dont Know how to proceed to the next step
   
Can Any one Tell Where Iam going Wrong.

Thankyou