raspberrypi/linux #3518: Bluetooth stops to connect after random period of time

dmirecki, 2020-03-26:
**Describe the bug**
I have installed home-assistant and [bt-mqtt-gateway](https://github.com/zewelor/bt-mqtt-gateway) on docker. The library periodically updates status of a smart thermostat via bluetooth. Unfortunately, after random period of time the updating process starts to give an error `bluepy.btle.BTLEDisconnectError: Failed to connect to peripheral 00:1A:22:11:72:4B, addr type: public` which is not very meaningful, but I attach logs from journalctl below. The only way to get rid off this error is to restart the bluetooth service: `sudo systemctl restart bluetooth`. After that, the update process works for random period of time - few hours, maybe few minutes, and then the problem returns.

**System**

* Which model of Raspberry Pi? e.g. Pi3B+, PiZeroW
```
Hardware        : BCM2835
Revision        : a02082
Serial          : 00000000ea30f86f
Model           : Raspberry Pi 3 Model B Rev 1.2
```
* Which OS and version (`cat /etc/rpi-issue`)?
```
Raspberry Pi reference 2019-09-26
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 80d486687ea77d31fc3fc13cf3a2f8b464e129be, stage2
```
* Which firmware version (`vcgencmd version`)?
```
Feb 12 2020 12:38:08 
Copyright (c) 2012 Broadcom
version 53a54c770c493957d99bf49762dfabc4eee00e45 (clean) (release) (start)
```
* Which kernel version (`uname -a`)?
```
Linux raspberrypi 4.19.97-v7+ #1294 SMP Thu Jan 30 13:15:58 GMT 2020 armv7l GNU/Linux
```

**Logs**
```
Mar 26 16:17:07 raspberrypi bluetoothd[417]: src/adapter.c:connect_failed_callback() hci0 00:1A:22:11:72:4B status 10
Mar 26 16:17:07 raspberrypi bluetoothd[417]: plugins/policy.c:conn_fail_cb() status 10
Mar 26 16:17:07 raspberrypi bluetoothd[417]: src/adapter.c:bonding_attempt_complete() hci0 bdaddr 00:1A:22:11:72:4B type 1 status 0xa
Mar 26 16:17:07 raspberrypi bluetoothd[417]: src/device.c:device_bonding_complete() bonding (nil) status 0x0a
Mar 26 16:17:07 raspberrypi bluetoothd[417]: src/device.c:device_bonding_failed() status 10
Mar 26 16:17:07 raspberrypi bluetoothd[417]: src/adapter.c:resume_discovery()
Mar 26 16:17:08 raspberrypi bluetoothd[417]: src/adapter.c:connect_failed_callback() hci0 00:1A:22:11:72:4B status 10
Mar 26 16:17:08 raspberrypi bluetoothd[417]: plugins/policy.c:conn_fail_cb() status 10
Mar 26 16:17:08 raspberrypi bluetoothd[417]: src/adapter.c:bonding_attempt_complete() hci0 bdaddr 00:1A:22:11:72:4B type 1 status 0xa
Mar 26 16:17:08 raspberrypi bluetoothd[417]: src/device.c:device_bonding_complete() bonding (nil) status 0x0a
Mar 26 16:17:08 raspberrypi bluetoothd[417]: src/device.c:device_bonding_failed() status 10
Mar 26 16:17:08 raspberrypi bluetoothd[417]: src/adapter.c:resume_discovery()

```

pelwell, 2020-04-07:
That kernel pre-dates fixing a bug in the UART driver which may make a difference - you can try updating to a new build of the kernel, but it might not improve matters. You can run:
```
$ sudo apt-get update
$ sudo apt-get install raspberrypi-kernel
```
to update. If that doesn't get you a kernel from March 2020 then you may be stuck on an older OS - Stretch, say - so you may have to resort to `sudo rpi-update`. That's not something we recommend as a matter of course, but our 4.19 kernel has been pretty stable recently and I'm not aware of any issues.

Unfortunately the 3B is the only Bluetooth-enabled Pi which doesn't use flow control when talking to the BT modem, meaning that if the host is slightly busy at the wrong time it could potentially lose data, causing it to lose sync with the modem. Some people choose to reduce the baudrate on the link to reduce bandwidth but improve latency tolerance. If you wish to try that you need to edit /usr/bin/btuart and change the 921600 in there to 460800 (or 230400, 115200 etc.).

dmirecki, 2020-04-15:
Thank you for your answer! 

I updated the kernel and unfortunately it didn't help. But, after decreasing the baudrate to 115200 the issue almost disappeared - from time to time (few times per day) the error still appears, but the bluetooth does not hang, so I don't have to restart the bluetooth service. I think it's good-enough solution for me. Thank you!

closed 2020-04-15 by dmirecki

Maynafiesta, 2022-12-19:
@dmirecki hi, im facing the same issue on ubuntu 20.04 pc. and still not solution for me. and  restarting the service is not letting me to pair at least short amount of time.