raspberrypi/linux #3522: Kernel panic -- not syncing : attempt to kill init!  exitcode=0×00000100

mohlondon, 2020-03-30:
Can any one help 
The problem occur because of unexpected cut of power here is a picture of the bug
[image]

pelwell, 2020-03-30:
That link is broken - try again. This isn't the best place to report user-generated errors like this - the [Raspberry Pi forums](https://raspberrypi.org/Forums) are intended for that - but if it's obvious I'll tell you.

mohlondon, 2020-03-30:
It's a laptop related problem it's not an SD card sir

pelwell, 2020-03-30:
You appear to be lost - this is a place to report problems in the Raspberry Pi Linux kernel.

closed 2020-03-30 by pelwell

mohlondon, 2020-03-30:
It's a linux problem didn't you see that the os is kali linux

pelwell, 2020-03-30:
"didn't you see that the os is kali linux" - no, because you've only just fixed the broken image (there are no notifications for that).

Anyway "laptop" != "Raspberry Pi".