raspberrypi/linux #3524: [5.6.y/arm64] - rpi4 /sys/class/thermal/thermal_zone0/temp not working.

asavah, 2020-04-01:
Just upgraded with a freshly built kernel from rpi-5.6.y branch, arm64.
No userspace updates were made, just the kernel.
Noticed 2 bugs, will report the second one in a separate issue.

Hardware:  Machine model: Raspberry Pi 4 Model B Rev 1.1 , 4GB
OS: own pet project, arm64

Bug: cat: /sys/class/thermal/thermal_zone0/temp: Invalid argument

It seems to be impossible to read cpu temp on rpi4 with 5.6.y kernel , 5.5 was OK
```
cat /sys/class/thermal/thermal_zone0/temp
cat: /sys/class/thermal/thermal_zone0/temp: Invalid argument
```

dmesg:
```
    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd083]
[    0.000000] Linux version 5.6.0-k2 (asavah@kross-1910) (gcc version 9.3.1 20200323 (GCC)) #1 SMP Wed Apr 1 15:51:45 EEST 2020
[    0.000000] Machine model: Raspberry Pi 4 Model B Rev 1.1
[    0.000000] cma: Reserved 64 MiB at 0x000000002ac00000
[    0.000000] NUMA: No NUMA configuration found
[    0.000000] NUMA: Faking a node at [mem 0x0000000000000000-0x00000000fbffffff]
[    0.000000] NUMA: NODE_DATA [mem 0xfb863100-0xfb864fff]
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x000000003fffffff]
[    0.000000]   DMA32    [mem 0x0000000040000000-0x00000000fbffffff]
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000002fffffff]
[    0.000000]   node   0: [mem 0x0000000040000000-0x00000000fbffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x00000000fbffffff]
[    0.000000] On node 0 totalpages: 966656
[    0.000000]   DMA zone: 3072 pages used for memmap
[    0.000000]   DMA zone: 0 pages reserved
[    0.000000]   DMA zone: 196608 pages, LIFO batch:63
[    0.000000]   DMA32 zone: 12032 pages used for memmap
[    0.000000]   DMA32 zone: 770048 pages, LIFO batch:63
[    0.000000] percpu: Embedded 31 pages/cpu s86552 r8192 d32232 u126976
[    0.000000] pcpu-alloc: s86552 r8192 d32232 u126976 alloc=31*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
[    0.000000] Detected PIPT I-cache on CPU0
[    0.000000] CPU features: kernel page table isolation disabled by kernel configuration
[    0.000000] spectrev2 mitigation disabled by kernel configuration
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 951552
[    0.000000] Policy zone: DMA32
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=1 cma=64M bcm2708_fb.fbwidth=1920 bcm2708_fb.fbheight=1280 bcm2708_fb.fbswap=1 smsc95xx.macaddr=DC:A6:32:0A:92:54 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  dwc_otg.lpm_enable=0 console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 fsck.repair=yes rootwait quiet
[    0.000000] Dentry cache hash table entries: 524288 (order: 10, 4194304 bytes, linear)
[    0.000000] Inode-cache hash table entries: 262144 (order: 9, 2097152 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] software IO TLB: mapped [mem 0x26c00000-0x2ac00000] (64MB)
[    0.000000] Memory: 3638576K/3866624K available (9982K kernel code, 1164K rwdata, 3728K rodata, 960K init, 661K bss, 162512K reserved, 65536K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 31809 entries in 125 pages
[    0.000000] ftrace: allocated 125 pages with 6 groups
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] rcu:     RCU restricting CPUs from NR_CPUS=8 to nr_cpu_ids=4.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 100 jiffies.
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=4
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] GIC: Using split EOI/Deactivate mode
[    0.000000] random: get_random_bytes called from start_kernel+0x308/0x498 with crng_init=0
[    0.000000] arch_timer: cp15 timer(s) running at 54.00MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0xc743ce346, max_idle_ns: 440795203123 ns
[    0.000005] sched_clock: 56 bits at 54MHz, resolution 18ns, wraps every 4398046511102ns
[    0.000217] Console: colour dummy device 80x25
[    0.000242] printk: console [tty1] enabled
[    0.000323] Calibrating delay loop (skipped), value calculated using timer frequency.. 108.00 BogoMIPS (lpj=54000)
[    0.000338] pid_max: default: 32768 minimum: 301
[... 311 lines cut ...]
[    6.370352] br37: port 2(wlan0) entered blocking state
[    6.370361] br37: port 2(wlan0) entered disabled state
[    6.370520] device wlan0 entered promiscuous mode
[    6.828399] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[    6.828500] br37: port 2(wlan0) entered blocking state
[    6.828505] br37: port 2(wlan0) entered forwarding state
[    6.828639] IPv6: ADDRCONF(NETDEV_CHANGE): br37: link becomes ready
[    7.350749] rc rc0: two consecutive events of type space
[   11.105951] bcmgenet fd580000.ethernet eth0: Link is Up - 1Gbps/Full - flow control off
[   11.105991] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[   11.106412] IPv6: ADDRCONF(NETDEV_CHANGE): vlan20: link becomes ready
[   11.106619] IPv6: ADDRCONF(NETDEV_CHANGE): vlan37: link becomes ready
[   11.106715] br37: port 1(vlan37) entered blocking state
[   11.106764] br37: port 1(vlan37) entered forwarding state
[   15.528935] wireguard: WireGuard 1.0.0 loaded. See www.wireguard.com for information.
[   15.528945] wireguard: Copyright (C) 2015-2019 Jason A. Donenfeld <Jason@zx2c4.com>. All Rights Reserved.
```

firmware:
```
vcgencmd version
Mar 10 2020 15:58:31
Copyright (c) 2012 Broadcom
version 74d7f53b091e465285f38fb96896a8fc328e4c47 (clean) (release) (start_x)
```

asavah, 2020-04-01:
Figured this one out.
Upstream added a separate driver bcm2711_thermal.
This driver need to be enabled, however for this driver even to be selectable in menuconfig MFD_SYSCON needs to be enabled first.

Currently arm64 bcm2711_defconfig does not enable said drivers but it does enable unneeded bcm2835_thermal (irrelevant for the issue, but still it's an unneded driver being built).

Enabling these 2 options
```
CONFIG_BCM2711_THERMAL=y
CONFIG_MFD_SYSCON=y
```
make `cat /sys/class/thermal/thermal_zone0/temp` great again.

Feel free to close once you read this )

pelwell, 2020-04-02:
Thanks for the report and fix - see aad3538.

closed 2020-04-02 by pelwell