raspberrypi/linux #3526: Raspbian 10 RPi4 - ext4-fs Journal has aborted - read only fs

slrslr, 2020-04-01:
Pi4, Raspbian 10 latest (i updated yesterday: apt update;apt upgrade)
This error happened 1st time to me. Appeared when i stopped TV playback in Kodi 18.6.

[image]

The sda7 partition is located on the external USB 3.0 drive. It worked many days without problem. Previously i have seen other FS related errors as per my previous submitted issues.

Please let me know if i need to launch any commands. I have not restarted RPi yet so i do not clear any possible logs. Can i restart it? I will do if there is no feedback in some time. Thank You

pelwell, 2020-04-01:
Reboot it - I don't think we will learn anything from it as it is.

How are you powering the external drive?

slrslr, 2020-04-01:
thank you. the drive is not powered except the RPi4 USB 3.0 port. It is connected using [SATA to USB 3.0 adaptor](https://www.axagon.cz/en/produkty/adsa-1s6). The drive is 2.5" SSD drive Samsung 850 EVO 120GB. Sticker on it says: rated: DC + 5.0V 1.4A.
[Here](https://www.raspberrypi.org/documentation/hardware/raspberrypi/usb/README.md#powerlimits) i see: "From the Raspberry Pi 3 onwards, the total power supplied to all USB ports in aggregate is 1200mA."
I guess it can not be increased beyond that using config,txt, "max_usb_current"?
How likely it is caused ba the insufficient power? Enabling any additional logging and if that happen again discovering cause is not possible or likely will not help?

pelwell, 2020-04-02:
We would certainly recommend powering an external drive. We've also seen problems with some USB-to-SATA adaptors - what does `lsusb` report when the drive is attached?

slrslr, 2020-04-02:
> what does `lsusb` report when the drive is attached

[reports this](https://defuse.ca/b/X3zCdv65HEoMZkCKyROCfo)

what is connected:
one 3.0 port is that SATA adaptor
one 2.0 port is USB extension cable with nothing connected to it
the RPi is able to handle also mouse and keyboard connected, but it has to be connected before boot, else it happened not once that when i connected the keyb. and mouse the USB drive got "disconnected" for a moment. But i am not using keyb. and mouse so it it just the drive and the usb cable - i do not think it would help to disconnect that unused USB 2.0 cable right and "max_usb_current" option not helpful either? As mentioned i have 4A power supply adaptor for the RPi4. Enabling more detailed loggin would not be helpful?

pelwell, 2020-04-02:
The SATA adaptor does not appear in the output from lsusb - try reattaching it (with the drive) until it shows up.

slrslr, 2020-04-02:
Is the thing you are requesting so important? Why you need it, can i discover the detail you need different way? Because i rebooted and still seeing the same "lsusb" output. I do not want to re-connect USB drive during the runtime, because the drive contains the root partition.. I am not booting into desktop, but into a command line.

pelwell, 2020-04-02:
If the drive/SATA adaptor is not showing up at all now then I think you have to power it.

You could also try the adaptor without the drive attached...