raspberrypi/linux #3541: Erroneous semicolon after macro definition

krepost, 2020-04-12:
https://github.com/raspberrypi/linux/blob/2e79fd01b4b9a7eea5acb234ad4e4cdca8449d5a/include/linux/iio/iio.h#L602

I believe that the semicolon at the end of the macro definition devm_iio_device_register in linux/iio/iio.h should not be there (since it is a macro definition and not a function declaration).

Thanks,

    /Lars

JamesH65, 2020-04-12:
You are probably right (although there are no side effects so could be either), but this is I think an upstream file, so should be reported there.

krepost, 2020-04-12:
I ran into problems with code that used the syntax

return (devm_iio_device_register(a, b));

In this case, the extra semicolon ended up inside the parenthesis.

Would you have a pointer to where I can report upstream? I'm completely new to developing anything for a linux kernel.

Thanks!

JamesH65, 2020-04-12:
> I ran into problems with code that used the syntax
> 
> return (devm_iio_device_register(a, b));
> 
> In this case, the extra semicolon ended up inside the parenthesis.
> 
> Would you have a pointer to where I can report upstream? I'm completely new to developing anything for a linux kernel.
> 
> Thanks!

Good point, didn't think of that. 

You will need to produce  patch,m and email it to the maintainers of that part of the kernel (Make sure its not already fixed in the very latest version). 

https://www.kernel.org/doc/html/latest/process/maintainers.html

Running 
scripts/get_maintainer.pl -f include/linux/iio/iio.h 

gives 
Jonathan Cameron <jic23@kernel.org> (maintainer:IIO SUBSYSTEM AND DRIVERS)
Hartmut Knaack <knaack.h@gmx.de> (reviewer:IIO SUBSYSTEM AND DRIVERS)
Lars-Peter Clausen <lars@metafoo.de> (reviewer:IIO SUBSYSTEM AND DRIVERS)
Peter Meerwald-Stadler <pmeerw@pmeerw.net> (reviewer:IIO SUBSYSTEM AND DRIVERS)
linux-iio@vger.kernel.org (open list:IIO SUBSYSTEM AND DRIVERS)
linux-kernel@vger.kernel.org (open list)

HTH.

krepost, 2020-04-19:
Thanks for the pointers! I have now reported the fix to the maintainers.

closed 2020-04-19 by krepost