raspberrypi/linux #3542: vc4-drm gpu Error

kicker0429, 2020-04-13:
Error messages: Apr 03 05:28:22 RASPI4 kernel: [drm:vc5_hdmi_init_resources [vc4]] *ERROR* Failed to get HDMI state machine clock
Apr 03 05:28:22 RASPI4 kernel: vc4-drm gpu: failed to bind fef00700.hdmi (ops vc4_hdmi_ops [vc4]): -517
Apr 03 05:28:22 RASPI4 kernel: vc4-drm gpu: master bind failed: -517

Model: Raspberry Pi 4
Kernel version: Linux RASPI4 5.6.2-1-ARCH #1 SMP PREEMPT Sat Apr 11 02:31:09 MDT 2020 aarch64 GNU/Linux
Kernel overlays: gpu_mem=512
enable_gic=1
dtoverlay=vc4-kms-v3d-pi4
initramfs initramfs-linux.img followkernel
kernel=kernel8.img
arm_64bit=1
device_tree=broadcom/bcm2711-rpi-4-b.dtb

6by9, 2020-04-13:
All the main development for vc4-kms-v3d-pi4 is on the rpi-5.4.y branch. Things get pulled up to 5.6 as as and when.

Based on a quick look, I suspect that a number of the patches to the device tree files (arch/arm/boot/dts/bcm*) haven't been pulled across.
df688cb81e3a ARM: dts: bcm283x: Unify CMA configuration
ee25078f9bff dt: Drop I2C for Pi4 HDMI interfaces to 97.5kHz.
46f0e1dc329b drm/vc4: Kick the core clock up during a mode change
d7187eabcd4a dt: Add HDMI audio dma values to bcm2711.dtsi
5d3d56c0754e ARM: dts: bcm2711: Enable the display pipeline
23e8e7367874 ARM: dts: bcm2711: Add HDMI DVP
a8e6d6c06c7d ARM: dts: bcm2711: Add firmware clocks node

kicker0429, 2020-04-13:
Do you recommend using the 5.4 branch or incorporating the patches into the 5.6 branch?

kicker0429, 2020-04-14:
Same errors with 5.4.29.

popcornmix, 2020-04-14:
> Do you recommend using the 5.4 branch or incorporating the patches into the 5.6 branch?

As development and testing happens on 5.4 then that would be recommended.
In time, patches will be ported/picked to 5.6 but no guarantee on timescales.

> Same errors with 5.4.29.

Is that with BRANCH=next rpi-update kernel or your own one?

kicker0429, 2020-04-14:
@popcornmix I pulled the source from the rpi-5.4.y and rpi-5.6.y branches and compiled them without modification on an Arch Linux distribution.

kicker0429, 2020-04-15:
After using a different .config file, the errors no longer appear.

closed 2020-04-15 by kicker0429