raspberrypi/linux #3543: Still seeing device descriptor read/64, error -71 using SUB hub

allanromanreyes, 2020-04-14:
Raspberry Pi 4B 2GB:
Raspbian 4.19.97-v7l
Merkury USB hub (found at Target stores)

dmesg:
[snip]
[   33.529215] print_req_error: I/O error, dev sda, sector 266240
[   33.530052] usb 1-1.3: USB disconnect, device number 18
[   33.568803] print_req_error: I/O error, dev sda, sector 6144
[   33.568820] Buffer I/O error on dev sda1, logical block 512, async page read
[   33.568865] print_req_error: I/O error, dev sda, sector 266240
[   33.568876] Buffer I/O error on dev sda2, logical block 256, async page read
[   33.958517] usb 1-1.3: new high-speed USB device number 19 using xhci_hcd
[   34.058994] usb 1-1.3: Device not responding to setup address.
[   34.278880] usb 1-1.3: Device not responding to setup address.
[   34.498547] usb 1-1.3: device not accepting address 19, error -71
[   34.598557] usb 1-1.3: new high-speed USB device number 20 using xhci_hcd
[   34.698834] usb 1-1.3: device descriptor read/64, error -71
[   34.918842] usb 1-1.3: device descriptor read/64, error -71
[   35.038899] usb 1-1-port3: attempt power cycle
[   35.708552] usb 1-1.3: new high-speed USB device number 21 using xhci_hcd
[   35.708857] usb 1-1.3: Device not responding to setup address.
[   35.928822] usb 1-1.3: Device not responding to setup address.
[   36.148550] usb 1-1.3: device not accepting address 21, error -71
[   36.248552] usb 1-1.3: new high-speed USB device number 22 using xhci_hcd
[   36.248848] usb 1-1.3: Device not responding to setup address.
[   36.468828] usb 1-1.3: Device not responding to setup address.
[   36.688555] usb 1-1.3: device not accepting address 22, error -71
[   36.688842] usb 1-1-port3: unable to enumerate USB device
[end]

Tried /boot/config.txt (max_usb_current=1) and /boot/cmdline.txt (dwc_otg.fiq_enable=0 ) fixes: neither worked. Still need to try dwc_otg masking.

allanromanreyes, 2020-04-14:
Previous error post: https://github.com/raspberrypi/linux/issues/2692

pelwell, 2020-04-14:
Those config/cmdline settings (max_usb_current=1 and dwc_otg.fiq_enable=0) are irrelevant on Pi 4B - it has a fully-featured XHCI controller, and in most circumstances doesn't use the DWC OTG controller.

I think you should pay more attention to the power supply and the USB<->SATA bridge (or whatever it is you are using to attach the storage device).

P33M, 2020-04-14:
Your reported symptom is completely different to the one listed in the other issue.

You are reporting (from an incomplete dmesg log) that a device that was previously enumerated has suffered a disconnect, and then does not get re-initialised. The symptom in the other issue is that the affected device does not get enumerated at all.

As pelwell says, you should investigate whether the arrangements for powering the hub and the storage adapter are adequate.

closed 2020-04-17 by allanromanreyes