raspberrypi/linux #3547: Missing Modules for Internet Wireless Daemon

MagicMuscleMan, 2020-04-15:
The Internet Wireless Daemon (`iwd`, see, e.g. [https://wiki.gentoo.org/wiki/Iwd](https://wiki.gentoo.org/wiki/Iwd)) is a more modern replacement of `wpa_supplicant` and according to my own experience much more reliable in case of troublesome Wifi connections. Due to its low resource usage it's also a great fit for a Raspberry Pi.

However, Raspbian's kernel 4.19.114-v7+ does not have the necessary kernel features activated, which is printed when calling `iwd` with `/usr/libexec/iwd`:

```
No HMAC(SHA1) support found
No HMAC(MD5) support found
No CMAC(AES) support found
No HMAC(SHA256) support not found
No HMAC(SHA512) support found, certain TLS connections might fail
No Diffie-Hellman support found, WPS will not be available
No asymmetric key support found.
TLS based WPA-Enterprise authentication methods will not function.
Kernel 4.20+ is required for this feature.
The following options are missing in the kernel:
        CONFIG_CRYPTO_USER_API_HASH
        CONFIG_ASYMMETRIC_KEY_TYPE
        CONFIG_KEY_DH_OPERATIONS
        CONFIG_CRYPTO_MD5
        CONFIG_CRYPTO_SHA256
        CONFIG_CRYPTO_AES
        CONFIG_ASYMMETRIC_PUBLIC_KEY_SUBTYPE
        CONFIG_CRYPTO_CMAC
        CONFIG_PKCS7_MESSAGE_PARSER
        CONFIG_CRYPTO_USER_API_HASH
        CONFIG_CRYPTO_USER_API_HASH
        CONFIG_CRYPTO_USER_API_HASH
        CONFIG_CRYPTO_HMAC
        CONFIG_CRYPTO_HMAC
        CONFIG_CRYPTO_HMAC
        CONFIG_CRYPTO_SHA512
        CONFIG_X509_CERTIFICATE_PARSER
        CONFIG_PKCS8_PRIVATE_KEY_PARSER
        CONFIG_CRYPTO_SHA1
The following optimized implementations might be available:
        CONFIG_CRYPTO_SHA1_SSSE3
        CONFIG_CRYPTO_AES_NI_INTEL
        CONFIG_CRYPTO_SHA512_SSSE3
        CONFIG_CRYPTO_AES_X86_64
        CONFIG_CRYPTO_SHA256_SSSE3
```

6by9, 2020-04-15:
Surely
```
Kernel 4.20+ is required for this feature.
```
is a more fundamental issue if you're running 4.19.114.

pelwell, 2020-04-15:
It's not clear how many of those features are required.

iamdavidcz, 2020-07-22:
@pelwell I have prepared a [patch](https://github.com/raspberrypi/linux/files/4961446/0001-configs-Add-several-CRYPTO-options-required-by-iwd.patch.txt) that adds kernel options [required by iwd](https://iwd.wiki.kernel.org/gettingstarted#kernel_dependencies). It would be great if these modules were available in 5.4.53 👍

pelwell, 2020-07-22:
Could you go the extra mile and turn that into a Pull Request? It provides a place for discussion, avoids errors and saves time.

MagicMuscleMan, 2020-08-30:
Thanks for adding the kernel modules! I can confirm, that iwd is working now (however I had to remove wpasupplicant an reboot).
The only thing not supported yet is using WPS (Wi-Fi Protected Setup) with iwd. I am not sure, if that's actually a good thing.

For me, this is fine, but if anyone wants to add support for that, this would be the patch of the .config file:

```
7001c7001
< # CONFIG_KEY_DH_OPERATIONS is not set
---
> CONFIG_KEY_DH_OPERATIONS=y
7051c7051
< CONFIG_CRYPTO_KPP=m
---
> CONFIG_CRYPTO_KPP=y
7070c7070
< # CONFIG_CRYPTO_DH is not set
---
> CONFIG_CRYPTO_DH=y
```

closed 2020-08-30 by MagicMuscleMan