raspberrypi/linux #3548: Who should register the raspberrypi-clk driver?

pelwell, 2020-04-16:
The `raspberrypi-clk` driver was originally declared in and instantiated from Device Tree. Then, after 91f2cf4a6b213, the firmware driver (itself a platform driver with) explicitly registers it. Now, since eae14a5f097a6, it once again relies on DT initialisation, but the explicit initialisation from the firmware driver is still there and failing, resulting in an alarming error:
```
raspberrypi-clk raspberrypi-clk: Missing firmware node
```
@mripard and @vianpl - can you decide between you which it's going to be? I'm in favour of DT-based instantiation - why do it in code when you can do it in data? - so without a good counter-argument I'll rip out the (now useless) instantiation code in the firmware driver.

pelwell, 2020-04-16:
@lategoodbye You may have an opinion on this.

mripard, 2020-04-16:
Yeah, that was a mistake on my side on the first iteration, sorry.

I'm not sure the discussion is entirely over yet, but what seems to be the trend at the moment is to have the clocks relationship described in DT, but without a separate node like I did in the first version.

Given that we merged an intermediate version between v1 and (the upcoming) v2, I guess simply removing the platform_device instantiation like you suggested is the least intrusive fix.

lategoodbye, 2020-04-16:
No really strong opinion about this. Since this all has been accepted in mainline, i fine with removing the platform_device instantiation. I hope this doesn't break any old DT + new kernel combination.

pelwell, 2020-04-17:
Thanks, everyone - https://github.com/raspberrypi/linux/commit/a142142bdf77a83cdae20a1f32873cbd34558efb (until the next rebase) reverts "firmware: raspberrypi: register clk device".

closed 2020-04-17 by pelwell