raspberrypi/linux #3550: Wireguard hasn't been backported to 5.6.y yet

AlwaysLivid, 2020-04-17:
**Describe the bug**
I've been struggling with getting Wireguard to work on a semi-vanilla Debian `arm64` system running on the Raspberry Pi 4 for a bit now, as the `wireguard-dkms` packages contain code that detect systems running on `>=5.6.0` and refuse to install the `dkms` modules. That's because the `5.6.y` branch does not contain the drivers in the `/drivers/net/wireguard` folder, in contrast to the upstream [kernel tree](https://github.com/torvalds/linux/tree/master/drivers/net/wireguard).

**To reproduce**
* Install Debian with a freshly-built `5.6.y arm64` kernel.
* Attempt to use Wireguard or use the `wireguard-dkms` module. 

**System**

* Which model of Raspberry Pi?

```
Raspberry Pi 4B
```

* Which OS and version (`cat /etc/rpi-issue`)?

```
Debian GNU/Linux 10
```

* Which firmware version (`vcgencmd version`)?

```
Apr 16 2020 12:31:32 
Copyright (c) 2012 Broadcom
version 30355345c483638d3f8b013f58f0ee17094e7de2 (clean) (release) (start)
```

* Which kernel version (`uname -a`)?

```
Linux pi 5.6.4-v8 #1 SMP PREEMPT Thu Apr 16 23:20:04 EEST 2020 aarch64 GNU/Linux
```

AlwaysLivid, 2020-04-17:
I'll note that maintaining a modified up-to-date package would be fairly hard and time-consuming on my end for a small Raspberry Pi system that I'd like to keep mostly unmaintained, but as a temporary workaround, I'll use the `5.5.y` kernel, personally.

I'd like to cherrypick/merge the changes myself and open a pull request, but I've noticed a handful of pull requests that haven't received attention so far, so I assumed that it'd be better to leave it up to any maintainers that have stumbled upon this issue.

pelwell, 2020-04-17:
The rpi-5.6.y kernel tree includes the wireguard sources - there is nothing to back-port.

If the kernel configuration you (or anyone) are using doesn't enable those drivers then that's a problem for whoever gave you the config file. If you are actually asking for those drivers to be enabled in the standard Raspberry Pi defconfigs on 5.6 then that's yet another different issue.

AlwaysLivid, 2020-04-17:
> The rpi-5.6.y kernel tree includes the wireguard sources - there is nothing to back-port.

> The rpi-5.6.y kernel tree includes the wireguard sources - there is nothing to back-port.

Oh, sorry, I apologize for that one then. Could I ask where I could find the Wireguard sources? I looked at the folder I mentioned earlier and tried to run `find` on a local copy of the repository, but I still didn't manage to spot them.

closed 2020-04-17 by AlwaysLivid

pelwell, 2020-04-17:
Are you looking at the right branch? In the GitHub GUI you need to find the Branch drop-down and switch to rpi-5.6.y (it currently defaults to rpi-4.19.y). On the command line working with a clone of the raspberrypi/linux repo you need to run `git checkout rpi-5.6.y`.

AlwaysLivid, 2020-04-17:
> Are you looking at the right branch? In the GitHub GUI you need to find the Branch drop-down and switch to rpi-5.6.y (it currently defaults to rpi-4.19.y). On the command line working with a clone of the raspberrypi/linux repo you need to run `git checkout rpi-5.6.y`.

Oh, I'm so sorry, this was it-- I did [spot the drivers](https://github.com/raspberrypi/linux/tree/rpi-5.6.y/drivers/net/wireguard).

Thanks a lot for your help.