raspberrypi/linux #3555: Enable Wireguard on 5.6.y by default

AlwaysLivid, 2020-04-17:
After I opened an issue (#3550) out of confusion, due to the fact that I couldn't figure out what was up with Wireguard, I spent my day reading up more about Linux kernel development and experimenting with it, with the help of the folk over at the #raspberrypi IRC channel on Freenode.

I figured that maybe I should perhaps ask whether enabling Wireguard by default on the `5.6.y` tree could be considered.

AlwaysLivid, 2020-04-17:
I've prepared [my own tree](https://github.com/AlwaysLivid/linux-rpi-wg/tree/rpi-5.6.y-wg) containing my own changes that are meant for personal use, but I'd like to open a pull request with those rather simple changes if that's alright.

pelwell, 2020-04-18:
That's very polite of you, but it's less email to just open the pull request (politely, of course). I see you've enabled it on 64-bit only, which ought to make any overhead in size or performance less noticeable.

closed 2020-04-18 by pelwell