raspberrypi/linux #3556: USB video capture devices unable to use full resolution

ferment-it, 2020-04-18:
I am encountering an issue that prevents the Raspberry Pi 4 from capturing full resolution NTSC video (720x480) using a variety of low cost USB video capture devices. The devices work as expected with Ubuntu Linux running on a Mac laptop.

The issue impacts USB video capture devices that use both the cx231xx and uvcvideo drivers but it presents slightly differently (details below).

I am using a Pi 4 running Raspbian Buster, full system details are at the bottom. Could this possibly be some kind of issue / limitation with the Pi’s USB hardware? Is there anything I can to do resolve it? Thank you for looking into it. 

**Issue description using uvcvideo driver devices:**
(534d:0021 “AV TO USB2.0” - sold as easyday DC60 on Amazon)

Attempting to preview or record at 720x480 resolution yields an extremely choppy stream, as if only a small number of the frames are actually making it through once a second or so.

This problem is reproduced using every software package attempted:

```
qv4l2
ffplay/ffmpeg ($ ffplay -f v4l2 -i /dev/video0)
gst-launch-1.0
```
Turning on the debug options for gst-launch-1.0 presents a long stream of “lost frame detected” and “failed to allocate buffer” which you can see here: [](https://pastebin.com/u2VVLrzz
)
Reducing the resolution of the capture to 320x240 makes the video record/play normally (I can share example output if helpful to see). The gst-launch-1.0 debug output is reduced to just three lines for a clip of the same length as above.

The device works as expected and runs at full resolution on an Ubuntu 18.04.3 ($ uname -r output reads 5.0.0-23-generic)

**Issue description using cx231xx driver devices:**
(1f4d:0102 “G-Tek Electronics Group” - the August VGB100 on Amazon) 

No output at all is seen in the same software options - qv4l2, ffmpeg/ffplay, gst-launch-1.0 when using at 720x480.  $ cat /dev/video0 shows no characters at all in the terminal.

Turning on the isoc_debug parameter on the cx231xx driver module reveals dmseg errors that look like similar to this one in a continuous stream:

**[94211.419521] cx231xx #0 print_err_status :URB packet 3, status -18 [Unknown].**

The gst-launch debug log is very short and indicates some kind of problem with the v4l  buffer: https://pastebin.com/N2TWgqxP

Reducing the screen resolution sufficiently (e.g. to 300x200) causes video to appear in all of the software packages (and characters in the terminal for cat /dev/video0). The kernel error messages stop as well.

This device also functions entirely normally on the same Ubuntu setup as above.

I have also tried other similar cx231xx devices and had the same result of no output (I but returned these ones before I figured out how to try the reduced screen resolution):

- Hauppauge USB 2Live (2040:c200 Hauppauge)
- Diamond VC500 USB 2.0 (also shows up the same identifier as the Austons device: 1f4d:0102 G-Tek Electronics Group)

Thanks much for your attention, would love to be able to get this working.

**Some final notes:**
I’ve tried each of the devices in all four USB ports (every 2.0 and 3.0)
There are no other USB devices plugged into the Pi at the same time - since I am accessing it over VNC
I am using rsync to grab the video to my main PC for review - I know that screen streaming over VNC can cause video to be choppy so this is not a cause here

**System details** 
```
Raspberry Pi 4 - 4GB ram

$cat /etc/rpi-issue
Raspberry Pi reference 2019-09-26
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 80d486687ea77d31fc3fc13cf3a2f8b464e129be, stage5

$vcgencmd version
Sep 24 2019 17:34:30 
Copyright (c) 2012 Broadcom
version cd3add54955f8fa065b414d8fc07c525e7ddffc8 (clean) (release) (start)

$uname -a
Linux raspberrypi 4.19.75-v7l+ #1270 SMP Tue Sep 24 18:51:41 BST 2019 armv7l GNU/Linux
```

P33M, 2020-04-18:
What is the output of "sudo rpi-eeprom-update"?
What happens if you upgrade the VLI firmware as detailed here: https://www.raspberrypi.org/forums/viewtopic.php?f=29&t=260879

ferment-it, 2020-04-18:
Thanks so much for the quick reply. So here was the output before the firmware upgrade:

`$ sudo rpi-eeprom-update"?
BOOTLOADER: up-to-date
CURRENT: Tue 10 Sep 10:41:50 UTC 2019 (1568112110)
 LATEST: Tue 10 Sep 10:41:50 UTC 2019 (1568112110)
VL805: up-to-date
CURRENT: 000137ab
 LATEST: 000137ab`

After the following the firmware instructions linked and rebooting the device the issue does not seem to have been resolved unfortunately. The output of rpi-eeprom-update now is below mentions that an update is needed? Any suggestions?

`$ sudo rpi-eeprom-update
BCM2711 detected
*** UPDATE REQUIRED ***
BOOTLOADER: update required
CURRENT: Tue 10 Sep 10:41:50 UTC 2019 (1568112110)
 LATEST: Mon 18 Nov 11:06:55 UTC 2019 (1574075215)
 FW DIR: /lib/firmware/raspberrypi/bootloader/beta
VL805: up-to-date
CURRENT: 000137ab
 LATEST: 000137ab
`

P33M, 2020-04-19:
Have you rebooted the Pi since running the update command?

ferment-it, 2020-04-19:
Hi P33M thanks for the help - I had rebooted but not shut it down / plugged and unplugged it

After doing that sudo repi-eeprom-update includes the below output but unfortunately I just tried with the cx231xx device and I'm still seeing the **cx231xx #0 print_err_status :URB packet 3, status -18 [Unknown]** messages in isoc_debug and the picture does not appear. Any thoughts on what to try next? Truly appreciate it!

`pi@raspberrypi:~ $ sudo rpi-eeprom-update
BCM2711 detected
BOOTLOADER: up-to-date
CURRENT: Mon 18 Nov 11:06:55 UTC 2019 (1574075215)
 LATEST: Mon 18 Nov 11:06:55 UTC 2019 (1574075215)
 FW DIR: /lib/firmware/raspberrypi/bootloader/beta
VL805: up-to-date
CURRENT: 000137ab
 LATEST: 000137ab`

timg236, 2020-04-20:
If the LATEST version is being reported as 137ab then it either the package is out of date or it's corrupted, files modified etc. You can use the following commands to 

sudo apt update
sudo apt upgrade

dpkg --verify rpi-eeprom
It might note that /etc/default/rpi-eeprom-update has been modified to beta

Alternatively, you can use the Pi Imager from the downloads page to create a rescue image which will also update the VL805 firmware https://www.raspberrypi.org/downloads/

JamesH65, 2020-04-20:
We now recommend using 

`sudo apt full-upgrade`

for reasons I cannot remember!

ferment-it, 2020-04-20:
Hi both thanks much! I used update and full-upgrade as suggested, here’s what I see now:

pi@raspberrypi:~ $ dpkg --verify rpi-eeprom
??5?????? c /etc/default/rpi-eeprom-update

Unfortunately the problem seems to persist with the device 

Any thoughts? I don’t like those question marks in that output - is that expected ? Thanks

timg236, 2020-04-21:
It's telling you that you have modified a file has been installed by a package. Unless you specifically want to install beta versions use the Raspberry Pi Imager from the downloads page to create an EEPROM rescue image which will program your VL805 SPI EEPROM

ferment-it, 2020-04-21:
Thanks - based on a forum thread linked here I updated a config to the beta version which might explain that (the instructions I followed are below) - should I try rolling that back first? Or proceed to the EEPROM rescue step you mention directly ? Thank you 

“To update to the new firmware image, edit /etc/default/rpi-eeprom-update as root and replace the parameter FIRMWARE_RELEASE_STATUS="critical" with FIRMWARE_RELEASE_STATUS="beta" and reboot.”

ferment-it, 2020-04-22:
OK so I went ahead and used the EEPROM rescue image.

The LED began to blink quickly - does that mean it worked? I'm operating the Pi headless (and unfortunately do not even own a monitor) so if there was some kind of on screen prompt to follow I did not see that.

Rpi-eeprom update is showing a date of September 2019 which has changed from earlier, but 137ab is the current and latest version which seems to be wrong based on earlier comments.

I went back and tried the EEPROM rescue image again after setting /etc/default/rpi-eeprom-update back to "critical" in case that was necessary, same result.  Did I do this right? Thanks

`BCM2711 detected
BOOTLOADER: up-to-date
CURRENT: Tue 10 Sep 10:41:50 UTC 2019 (1568112110)
 LATEST: Tue 10 Sep 10:41:50 UTC 2019 (1568112110)
 FW DIR: /lib/firmware/raspberrypi/bootloader/critical
VL805: update required
CURRENT: 000137ab
 LATEST: 000137ad
`

timg236, 2020-04-22:
It turns out the Raspberry Pi Imager release was at the older version. You should be able to do

sudo rpi-eeprom-update -a
sudo reboot

to apply the imagers.

Out of interest, please can you list the contents of /lib/firmware/raspberrypi/bootloader/beta. rpi-eeprom-update is just a script which finds the latest release by listing / sorting files and it sounds like something is out of place in that directory

ferment-it, 2020-04-22:
Thanks for the suggestion - it looks like the version is still showing as 000137ab after that? Output of what I did is included below (yesterday I set the updates back to "critical" so included the contents of that folder as well in case that's useful - thank you

pi@raspberrypi:~ $ cat /etc/default/rpi-eeprom-update
FIRMWARE_RELEASE_STATUS="critical"

pi@raspberrypi:~ $ ls -l /lib/firmware/raspberrypi/bootloader/beta
total 8064
-rw-r--r-- 1 root root 524288 Jan 22 09:28 pieeprom-2019-09-06.bin
-rw-r--r-- 1 root root 524288 Jan 22 09:28 pieeprom-2019-09-10.bin
-rw-r--r-- 1 root root 524288 Jan 22 09:28 pieeprom-2019-09-23.bin
-rw-r--r-- 1 root root 524288 Jan 22 09:28 pieeprom-2019-09-25.bin
-rw-r--r-- 1 root root 524288 Jan 22 09:28 pieeprom-2019-10-08.bin
-rw-r--r-- 1 root root 524288 Jan 22 09:28 pieeprom-2019-10-16.bin
-rw-r--r-- 1 root root 524288 Jan 22 09:28 pieeprom-2019-11-18.bin
-rw-r--r-- 1 root root 524288 Jan 22 09:28 pieeprom-2019-12-03.bin
-rw-r--r-- 1 root root 524288 Jan 22 09:28 pieeprom-2020-01-09.bin
-rw-r--r-- 1 root root 524288 Jan 22 09:28 pieeprom-2020-01-17.bin
-rw-r--r-- 1 root root 524288 Mar 13 10:16 pieeprom-2020-03-04.bin
-rw-r--r-- 1 root root 524288 Mar 20 12:18 pieeprom-2020-03-16.bin
-rw-r--r-- 1 root root 524288 Mar 20 12:18 pieeprom-2020-03-19.bin
-rw-r--r-- 1 root root 524288 Apr  9 14:12 pieeprom-2020-04-09.bin
-rw-r--r-- 1 root root 524288 Apr 17 06:50 pieeprom-2020-04-16.bin
-rw-r--r-- 1 root root  87784 Apr 12 18:27 recovery.bin
-rw-r--r-- 1 root root  96984 Jan 22 09:28 vl805-00013701.bin
-rw-r--r-- 1 root root  98712 Jan 22 09:28 vl805-000137ab.bin
-rw-r--r-- 1 root root  98904 Jan 22 09:28 vl805-000137ad.bin

pi@raspberrypi:~ $ ls -l /lib/firmware/raspberrypi/bootloader/critical
total 1380
-rw-r--r-- 1 root root 524288 Jan 22 09:28 pieeprom-2019-07-15.bin
-rw-r--r-- 1 root root 524288 Jan 22 09:28 pieeprom-2019-09-10.bin
-rw-r--r-- 1 root root  60820 Jan 22 09:28 recovery.bin
-rw-r--r-- 1 root root  96984 Jan 22 09:28 vl805-00013701.bin
-rw-r--r-- 1 root root  98712 Jan 22 09:28 vl805-000137ab.bin
-rw-r--r-- 1 root root  98904 Jan 22 09:28 vl805-000137ad.bin

pi@raspberrypi:~ $ sudo rpi-eeprom-update -a
BCM2711 detected
*** INSTALLING EEPROM UPDATES ***
BOOTLOADER: up-to-date
CURRENT: Tue 10 Sep 10:41:50 UTC 2019 (1568112110)
 LATEST: Tue 10 Sep 10:41:50 UTC 2019 (1568112110)
 FW DIR: /lib/firmware/raspberrypi/bootloader/critical
VL805: update required
CURRENT: 000137ab
 LATEST: 000137ad
EEPROM updates pending. Please reboot to apply the update.

<REBOOT + POWER CYCLE>

pi@raspberrypi:~ $ sudo rpi-eeprom-update
BCM2711 detected
BOOTLOADER: up-to-date
CURRENT: Tue 10 Sep 10:41:50 UTC 2019 (1568112110)
 LATEST: Tue 10 Sep 10:41:50 UTC 2019 (1568112110)
 FW DIR: /lib/firmware/raspberrypi/bootloader/critical
VL805: up-to-date
CURRENT: 000137ad
 LATEST: 000137ad

timg236, 2020-04-22:
Looks ok to me "LATEST: 000137ad" so it's now running the latest production release of the VL805 firmware.

ferment-it, 2020-04-22:
Unfortunately I am still experiencing the issue I described at the top, even after several reboots and power cycles. Looks like I should still be up to date on firmware if 000137ad is the latest? Would this be expected to resolve the issue or do I need a different version? Thanks

pi@raspberrypi:~ $ sudo rpi-eeprom-update
BCM2711 detected
BOOTLOADER: up-to-date
CURRENT: Tue 10 Sep 10:41:50 UTC 2019 (1568112110)
 LATEST: Tue 10 Sep 10:41:50 UTC 2019 (1568112110)
 FW DIR: /lib/firmware/raspberrypi/bootloader/critical
VL805: up-to-date
CURRENT: 000137ad
 LATEST: 000137ad

P33M, 2020-04-23:
Using the latest firmware removes several classes of bugs that could cause frame drops.
- Please post a full dmesg output from boot up to and including when you start using the device and experience frame drops.
- Please post the output of `sudo lsusb -v` with the device plugged in.

ferment-it, 2020-04-23:
Here you go:

1. Dmesg at Pi Startup - https://pastebin.com/LJPdsivY

2. Dmesg after inserting the device into the USB port - https://pastebin.com/zT9YQrBk

3. Dmesg output when using "$ ffplay -standard ntsc v4l2 -i /dev/video0"  (which works fine on ubuntu) - but on RPi ffplay does not even open a window, and the information about incoming data does not increment up at all as expected
https://pastebin.com/n9M5Myfe
(this is after setting /sys/class/video4linux/video0/dev_debug to 2, and /sys/module/cx231xx/parameters/isoc_debug to 1)

4. sudo lsusb -v for device - https://pastebin.com/5ctM0Y9N

Thanks for your input! Finally getting this working would get me unblocked from a project to digitize family VHS video with the Pi I've been banging my head against for quite some time now

P33M, 2020-04-23:
When you're running a capture (with or without errors), what's the output from `sudo cat /sys/kernel/debug/usb/devices`? There's multiple interfaces with multiple alternate settings, so I need to see how the driver is using the device.

ferment-it, 2020-04-23:
Including that output below - it's been a lot of fun learning about all these logs in nooks and crannies using Raspberry pi for this project! (my first linux experience - and really my first time spending a lot of time focused on computer science at all!)

pi@raspberrypi:~ $ sudo cat /sys/kernel/debug/usb/devices

T:  Bus=01 Lev=00 Prnt=00 Port=00 Cnt=00 Dev#=  1 Spd=480  MxCh= 1
B:  Alloc=  0/800 us ( 0%), #Int=  0, #Iso=  8
D:  Ver= 2.00 Cls=09(hub  ) Sub=00 Prot=01 MxPS=64 #Cfgs=  1
P:  Vendor=1d6b ProdID=0002 Rev= 4.19
S:  Manufacturer=Linux 4.19.97-v7l+ xhci-hcd
S:  Product=xHCI Host Controller
S:  SerialNumber=0000:01:00.0
C:* #Ifs= 1 Cfg#= 1 Atr=e0 MxPwr=  0mA
I:* If#= 0 Alt= 0 #EPs= 1 Cls=09(hub  ) Sub=00 Prot=00 Driver=hub
E:  Ad=81(I) Atr=03(Int.) MxPS=   4 Ivl=256ms

T:  Bus=01 Lev=01 Prnt=01 Port=00 Cnt=01 Dev#=  2 Spd=480  MxCh= 4
D:  Ver= 2.10 Cls=09(hub  ) Sub=00 Prot=01 MxPS=64 #Cfgs=  1
P:  Vendor=2109 ProdID=3431 Rev= 4.21
S:  Product=USB2.0 Hub
C:* #Ifs= 1 Cfg#= 1 Atr=e0 MxPwr=100mA
I:* If#= 0 Alt= 0 #EPs= 1 Cls=09(hub  ) Sub=00 Prot=00 Driver=hub
E:  Ad=81(I) Atr=03(Int.) MxPS=   1 Ivl=256ms

T:  Bus=01 Lev=02 Prnt=02 Port=00 Cnt=01 Dev#=  3 Spd=480  MxCh= 0
D:  Ver= 2.00 Cls=ef(misc ) Sub=02 Prot=01 MxPS=64 #Cfgs=  1
P:  Vendor=1f4d ProdID=0102 Rev=40.02
S:  Manufacturer=Geniatech Inc.
S:  Product=Video Capture
S:  SerialNumber=20090456
C:* #Ifs= 6 Cfg#= 1 Atr=80 MxPwr=500mA
A:  FirstIf#= 1 IfCount= 5 Cls=ff(vend.) Sub=ff Prot=ff
I:* If#= 0 Alt= 0 #EPs= 2 Cls=ff(vend.) Sub=ff Prot=ff Driver=(none)
E:  Ad=8e(I) Atr=03(Int.) MxPS=  32 Ivl=1ms
E:  Ad=0e(O) Atr=03(Int.) MxPS=  32 Ivl=1ms
I:* If#= 1 Alt= 0 #EPs= 1 Cls=ff(vend.) Sub=ff Prot=ff Driver=cx231xx
E:  Ad=8f(I) Atr=03(Int.) MxPS=   8 Ivl=8ms
I:* If#= 2 Alt= 0 #EPs= 1 Cls=ff(vend.) Sub=ff Prot=ff Driver=(none)
E:  Ad=83(I) Atr=02(Bulk) MxPS= 512 Ivl=0ms
I:  If#= 2 Alt= 1 #EPs= 1 Cls=ff(vend.) Sub=ff Prot=ff Driver=(none)
E:  Ad=83(I) Atr=05(Isoc) MxPS=  28 Ivl=125us
I:  If#= 2 Alt= 2 #EPs= 1 Cls=ff(vend.) Sub=ff Prot=ff Driver=(none)
E:  Ad=83(I) Atr=05(Isoc) MxPS=  52 Ivl=125us
I:  If#= 3 Alt= 0 #EPs= 1 Cls=ff(vend.) Sub=ff Prot=ff Driver=(none)
E:  Ad=84(I) Atr=02(Bulk) MxPS= 512 Ivl=0ms
I:  If#= 3 Alt= 1 #EPs= 1 Cls=ff(vend.) Sub=ff Prot=ff Driver=(none)
E:  Ad=84(I) Atr=05(Isoc) MxPS= 184 Ivl=125us
I:  If#= 3 Alt= 2 #EPs= 1 Cls=ff(vend.) Sub=ff Prot=ff Driver=(none)
E:  Ad=84(I) Atr=05(Isoc) MxPS= 728 Ivl=125us
I:* If#= 3 Alt= 3 #EPs= 1 Cls=ff(vend.) Sub=ff Prot=ff Driver=(none)
E:  Ad=84(I) Atr=05(Isoc) MxPS=2892 Ivl=125us
I:  If#= 3 Alt= 4 #EPs= 1 Cls=ff(vend.) Sub=ff Prot=ff Driver=(none)
E:  Ad=84(I) Atr=05(Isoc) MxPS=1800 Ivl=125us
I:* If#= 4 Alt= 0 #EPs= 1 Cls=ff(vend.) Sub=ff Prot=ff Driver=(none)
E:  Ad=85(I) Atr=02(Bulk) MxPS= 512 Ivl=0ms
I:  If#= 4 Alt= 1 #EPs= 1 Cls=ff(vend.) Sub=ff Prot=ff Driver=(none)
E:  Ad=05(O) Atr=02(Bulk) MxPS= 512 Ivl=125us
I:* If#= 5 Alt= 0 #EPs= 1 Cls=ff(vend.) Sub=ff Prot=ff Driver=(none)
E:  Ad=86(I) Atr=02(Bulk) MxPS= 512 Ivl=0ms
I:  If#= 5 Alt= 1 #EPs= 1 Cls=ff(vend.) Sub=ff Prot=ff Driver=(none)
E:  Ad=86(I) Atr=05(Isoc) MxPS= 576 Ivl=125us

T:  Bus=02 Lev=00 Prnt=00 Port=00 Cnt=00 Dev#=  1 Spd=5000 MxCh= 4
B:  Alloc=  0/800 us ( 0%), #Int=  0, #Iso=  0
D:  Ver= 3.00 Cls=09(hub  ) Sub=00 Prot=03 MxPS= 9 #Cfgs=  1
P:  Vendor=1d6b ProdID=0003 Rev= 4.19
S:  Manufacturer=Linux 4.19.97-v7l+ xhci-hcd
S:  Product=xHCI Host Controller
S:  SerialNumber=0000:01:00.0
C:* #Ifs= 1 Cfg#= 1 Atr=e0 MxPwr=  0mA
I:* If#= 0 Alt= 0 #EPs= 1 Cls=09(hub  ) Sub=00 Prot=00 Driver=hub
E:  Ad=81(I) Atr=03(Int.) MxPS=   4 Ivl=256ms

P33M, 2020-04-23:
Hmm this is a very busy device. I would guess (from crossreferencing the dmesg with the active endpoints) that we have:
1) Video on this endpoint (isochronous)
```
I:* If#= 3 Alt= 3 #EPs= 1 Cls=ff(vend.) Sub=ff Prot=ff Driver=(none)
E: Ad=84(I) Atr=05(Isoc) MxPS=2892 Ivl=125us
```

2) Vertical blanking data on this endpoint (bulk)
```
I:* If#= 4 Alt= 0 #EPs= 1 Cls=ff(vend.) Sub=ff Prot=ff Driver=(none)
E: Ad=85(I) Atr=02(Bulk) MxPS= 512 Ivl=0ms
```
3) "sliced CC", whatever that is - closed captions?  (bulk)
```
I:* If#= 5 Alt= 0 #EPs= 1 Cls=ff(vend.) Sub=ff Prot=ff Driver=(none)
E: Ad=86(I) Atr=02(Bulk) MxPS= 512 Ivl=0ms
```
4) Audio input (bulk, wierdly)
```
I:* If#= 2 Alt= 0 #EPs= 1 Cls=ff(vend.) Sub=ff Prot=ff Driver=(none)
E: Ad=83(I) Atr=02(Bulk) MxPS= 512 Ivl=0ms
```

The video endpoint also has a bulk alternate setting. 
Here are some things to try:

- What is the output of `sudo lspci -xxx -s 01:00.0` with the capture card plugged in, but idle
- What is the output of the same command again, but when you are attempting to capture video data?

The driver supports a bulk capture mode, which may be worth a go.
- Unplug the capture card, then remove the associated driver with `sudo rmmod cx231xx`.
- Manually install the driver but with the bulk option set: `sudo modprobe cx231xx transfer_mode=0`
- Plug the capture card back in

Then can you do the two lspci steps above - and see if the bulk capture mode behaves differently to the default. 

The peak bandwidth of the isochronous endpoint is almost as large as the USB specification allows (max is 3072 bytes per transfer), so I wonder if any of these bulk transfers are causing bandwidth contention.

ferment-it, 2020-04-23:
Thank you! Ok so for the first part in isoc mode - it looks as though the output of "sudo lspci -xxx -s 01:00.0" is identical in either case (idle or capturing) based on a diff i did - it looks like this https://pastebin.com/AYiTxRBk

After switching to bulk mode, I am actually able to get a video to appear where none did earlier (improvement!). It plays normally for a few seconds, but then video stalls entirely while audio continues, and then it begins again, and stalls out. (I can try to upload a clip somewhere if that would help)

The output of the bulk mode lspci commands is here - the left is idle, the right is capturing - https://www.diffchecker.com/JXvRqYmX

Is there other debug output that would be helpful to look at to understand what might be happening? Thanks much for your continued help!

P33M, 2020-04-23:
So the byte in the lspci dump at offset 0xd4 that changes is the link control register. The VLI firmware is supposed to disable PCIe link power management (aka ASPM L0s and L1) when there is significant USB2.0 bulk or isochronous traffic, but in your case it looks like 
a) the available bulk throughput isn't high enough or the device bursts too much data, even with the firmware correctly disabling ASPM
b) the firmware isn't detecting your device's isochronous traffic for some reason.

Can you try the isochronous case but before you start capturing, can you do `sudo setpci -s 01:00.0 0xD4.B=0x41` to disable L1?

ferment-it, 2020-04-23:
Thanks again - I am seeing the same issue (video freezes while audio continues) but it's maybe slightly improved from the earlier attempt

I noticed on the isoc_debug output that there are some intermittent flips back and forth between streams of the following two messages (sample of this below) that I think correspond to the times when the video freezes

[  281.418222] cx231xx #0 buffer_filled :[68a2588b/1] wakeup
[  281.452631] cx231xx #0 buffer_filled :[88a21b9c/2] wakeup
[  281.487136] cx231xx #0 buffer_filled :[3bb48236/3] wakeup
[  281.518135] cx231xx #0 buffer_filled :[30fc8283/0] wakeup
[  281.552612] cx231xx #0 buffer_filled :[68a2588b/1] wakeup
[  281.552692] cx231xx #0 get_next_buf :No active queue to serve
[  281.571783] cx231xx #0 get_next_buf :No active queue to serve
[  281.587534] cx231xx #0 get_next_buf :No active queue to serve
[  281.603629] cx231xx #0 get_next_buf :No active queue to serve
[  281.621781] cx231xx #0 get_next_buf :No active queue to serve
[  281.637631] cx231xx #0 get_next_buf :No active queue to serve

P33M, 2020-04-24:
Do you see an improvement if you disable L1 as above and capture video with no audio?

ferment-it, 2020-04-24:
Good idea- there is an improvement - the freezes are shorter and fewer in-between, but they still happen every 10 seconds or so. Does that reveal something useful?

P33M, 2020-04-24:
Are you saving the video stream to a file, and if you are, is that file on SD card?

ferment-it, 2020-04-24:
Yes I am saving it to an mkv file (in raw video form, which my goal is  to preserve to process later) and yes that file is on the SD card. Nothing else is plugged into USB but the capture device :)

P33M, 2020-04-24:
Can you compare performance if you save the output video to /dev/null?

ferment-it, 2020-04-24:
OK - so this is tough for me to be 100% sure about, since I am running the Pi headless - and I've learned that the VNC frame rate is much too low to watch video remotely (I pull all the capture videos to my laptop for review).

However I think that outputting to /dev/null may have eliminated the issue - my observation this: this time I am not seeing anything but log messages like the one below for the entire time of the capture. In the earlier attempts these were punctuated with "No active queue to serve" messages that seemed to correspond to the periods of the video dropping (not certain about though since I wasn't watching live)

What do we make of that? 

[14543.705279] cx231xx #0 buffer_filled :[e8e490e1/0] wakeup

JamesH65, 2020-04-24:
It implies that the data rate you are sending to the SD card is too high for the card to cope with. Have you worked out what data rate you are throwing at it?

ferment-it, 2020-04-24:
I don't know exactly but yes it must be very high (720x480 video in raw format). The clip I recorded earlier without the sound was ~575 MB for 28 seconds worth (so something like ~20.5MB/s perhaps) 

I just looked up the write speed for SD cards (12.5MB/s?) and it looks like that might actually exceed it? (I have the Pi with 4 gigs of ram in case that's meaningful)

ferment-it, 2020-04-24:
Actually scratch that - the 12.5MB/s was just the first thing I saw on Google, my specific card looks like it's much faster (98MB/s) so should be fine?  Here's the info from my Amazon purchase:

SanDisk 16GB Ultra MicroSDHC UHS-I Memory Card with Adapter - 98MB/s, C10, U1, Full HD, A1, Micro SD Card - SDSQUAR-016G-GN6MA

(Of course I'll need a lot more than 16GB capacity to use the Pi for these raw videos - I bought the card before coming up with the use case and I'm trying to get a proof of concept before I invest in a much larger one)

P33M, 2020-04-24:
Now we're cooking with gas - your card's 98MB/s specification is for "transfer speed" which is not a substitute for write speed. In fact the only guaranteed minimum write speed is 10MB/s because that card has an Apps Class A1 rating.

You really need a USB3.0 storage device to use your capture card effectively - or, stream the video to a file on a remote network share. On a Pi 4, USB3.0 and the Ethernet link do not share bandwidth with the USB2.0 ports.

ferment-it, 2020-04-24:
OK! So if I am following correctly then I should:

1) Insert the capture device into one of the USB 2.0 ports  
2) Connect a USB 3.0 hard drive to the Pi (one of the blue ports)
3) Disable the L1 per the instructions above ("sudo setpci -s 01:00.0 0xD4.B=0x41")

And then things should work as I would expected?

Dumb question - is the write speed of a hard disk meaningful bottleneck to consider here? Or can I choose whichever USB 3.0 hard drive provides good value? (Google seems to suggest that 7200RPM would be more than fast enough, but I'm not really seeing RPM figures on Amazon external HD listings)

Also - do I have to worry if a drive is indicated as NTFS format? I assume I can reformat into ext3? (and that would be cross-compatible on a Mac right?)

Thanks much for helping run this to ground, really excited to get it going. Thanks so much

JamesH65, 2020-04-25:
The write speed of current HDD's with USB3.0 would almost certainly be adequate, but always worthwhile checking!

It's quite possible a fast enough storage medium will make all the other problems go away - could just be plug and play.

ferment-it, 2020-04-25:
Got my hands on a USB 3 drive - I still see some of the errors about “no active queue” though. It’s just just barely noticeable on the video but would like to eliminate it entirely if possible 

I discovered the answer to my Q above and that the Rpi can write to NTFS but it seems to take up significant CPU looking at top.

I wonder if another HD format might have faster write speeds and allow the error messages go away entirely? Or is that wishful thinking 

It looks like FAT32 doesn’t permit the size of file I’m going to end up with... but ext3/4 are not cross compatible with Mac for me to actually view the output on my PC. Could exfat be worth trying?

Thanks !