raspberrypi/linux #3560: bcm2835_mmal_vchiq: buffer_from_host: msg_context not allocated

mjasny, 2020-04-20:
**Describe the bug**
I'm working with a Raspberry Pi 4 on a simple gstreamer pipeline to decode a h264 stream using v4l2h264enc and v4l2h264dec.
Everything works perfectly fine until a certain amount of time (20 seconds - 15 minutes) where suddenly the gst-pipeline crashes, caused by v4l2 allocator/bufferpool (see kernel trace).

**To reproduce**
- `gst-launch-1.0 rtspsrc location='rtsp://address' ! rtph264depay ! h264parse ! v4l2h264dec ! queue ! v4l2h264enc ! h264parse ! flvmux ! filesink location=output.flv`

- Wait a few minutes

**Expected behaviour**
- Continue encoding without failure.

**Actual behaviour**
- The output produced until the failure is perfectly fine. After a certain amount of time 20sec-15min the gstreamer pipeline crashes and produces the following kernel trace.

**Additional context**
More details:
https://gitlab.freedesktop.org/gstreamer/gstreamer/-/issues/538

**System**
 Copy and paste the results of the raspinfo command in to this section. Alternatively, copy and paste a pastebin link, or add answers to the following questions:

* Which model of Raspberry Pi? e.g. Pi3B+, PiZeroW
Raspberry Pi 4 2GB

* Which OS and version (`cat /etc/rpi-issue`)?
```
Raspberry Pi reference 2020-02-13
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 5f884374b6ac6e155330c58caa1fb7249b8badf1, stage2
```
* Which firmware version (`vcgencmd version`)?
```
Feb 12 2020 12:36:21 
Copyright (c) 2012 Broadcom
version c3c8dbdf147686fb0c3f32aece709d0653368810 (clean) (release) (start)
```
* Which kernel version (`uname -a`)?
```
Linux raspberrypi 4.19.97-v7l+ #1294 SMP Thu Jan 30 13:21:14 GMT 2020 armv7l GNU/Linux
```

**Logs**
When the gstreamer pipeline fails:
```
[13922.410921] bcm2835-codec bcm2835-codec: bcm2835_codec_stop_streaming: Timeout waiting for buffers to be returned - 40 outstanding                                                                                                   
[13922.411070] ------------[ cut here ]------------
[13922.411101] WARNING: CPU: 1 PID: 2449 at drivers/media/common/videobuf2/videobuf2-core.c:1688 __vb2_queue_cancel+0x158/0x1c8 [videobuf2_common]
[13922.411107] Modules linked in: bnep hci_uart btbcm serdev bluetooth ecdh_generic 8021q garp stp llc ftdi_sio usbserial brcmfmac vc4 brcmutil v3d gpu_sched drm_kms_helper sha256_generic drm snd_soc_core drm_panel_orientation_quirks snd_compress snd_pcm_dmaengine syscopyarea sysfillrect sysimgblt fb_sys_fops cfg80211 rfkill snd_bcm2835(C) bcm2835_codec(C) bcm2835_v4l2(C) v4l2_mem2mem bcm2835_mmal_vchiq(C) snd_pcm v4l2_common raspberrypi_hwmon videobuf2_dma_contig videobuf2_vmalloc hwmon videobuf2_memops snd_timer videobuf2_v4l2 videobuf2_common snd videodev media vc_sm_cma(C) rpivid_mem uio_pdrv_genirq uio ip_tables x_tables ipv6
[13922.411219] CPU: 1 PID: 2449 Comm: rtpjitterbuffer Tainted: G        WC        4.19.97-v7l+ #1294
[13922.411223] Hardware name: BCM2835
[13922.411246] [<c0212e04>] (unwind_backtrace) from [<c020d5e0>] (show_stack+0x20/0x24)
[13922.411258] [<c020d5e0>] (show_stack) from [<c09b15c8>] (dump_stack+0xe0/0x124)
[13922.411272] [<c09b15c8>] (dump_stack) from [<c0222544>] (__warn+0x104/0x11c)
[13922.411283] [<c0222544>] (__warn) from [<c0222694>] (warn_slowpath_null+0x50/0x58)
[13922.411303] [<c0222694>] (warn_slowpath_null) from [<bf265104>] (__vb2_queue_cancel+0x158/0x1c8 [videobuf2_common])
[13922.411334] [<bf265104>] (__vb2_queue_cancel [videobuf2_common]) from [<bf265ac0>] (vb2_core_streamoff+0x44/0xb4 [videobuf2_common])
[13922.411359] [<bf265ac0>] (vb2_core_streamoff [videobuf2_common]) from [<bf28895c>] (vb2_streamoff+0x40/0x60 [videobuf2_v4l2])
[13922.411380] [<bf28895c>] (vb2_streamoff [videobuf2_v4l2]) from [<bf321c9c>] (v4l2_m2m_streamoff+0x3c/0xf4 [v4l2_mem2mem])
[13922.411401] [<bf321c9c>] (v4l2_m2m_streamoff [v4l2_mem2mem]) from [<bf321d74>] (v4l2_m2m_ioctl_streamoff+0x20/0x24 [v4l2_mem2mem])
[13922.411452] [<bf321d74>] (v4l2_m2m_ioctl_streamoff [v4l2_mem2mem]) from [<bf1bee68>] (v4l_streamoff+0x28/0x2c [videodev])
[13922.411531] [<bf1bee68>] (v4l_streamoff [videodev]) from [<bf1c1a38>] (__video_do_ioctl+0x268/0x528 [videodev])
[13922.411607] [<bf1c1a38>] (__video_do_ioctl [videodev]) from [<bf1c5470>] (video_usercopy+0x218/0x600 [videodev])
[13922.411683] [<bf1c5470>] (video_usercopy [videodev]) from [<bf1c5878>] (video_ioctl2+0x20/0x24 [videodev])
[13922.411756] [<bf1c5878>] (video_ioctl2 [videodev]) from [<bf1bd144>] (v4l2_ioctl+0x4c/0x60 [videodev])
[13922.411801] [<bf1bd144>] (v4l2_ioctl [videodev]) from [<c03e98a4>] (do_vfs_ioctl+0xbc/0x804)
[13922.411812] [<c03e98a4>] (do_vfs_ioctl) from [<c03ea030>] (ksys_ioctl+0x44/0x6c)
[13922.411820] [<c03ea030>] (ksys_ioctl) from [<c03ea070>] (sys_ioctl+0x18/0x1c)
[13922.411830] [<c03ea070>] (sys_ioctl) from [<c0201000>] (ret_fast_syscall+0x0/0x28)
[13922.411836] Exception stack(0xdbd21fa8 to 0xdbd21ff0)
[13922.411843] 1fa0:                   b151a530 b5d72000 00000024 40045613 0130faa8 b6bc5510
[... 43 lines cut ...]
[13958.572657] [<c03ea030>] (ksys_ioctl) from [<c03ea070>] (sys_ioctl+0x18/0x1c)
[13958.572672] [<c03ea070>] (sys_ioctl) from [<c0201000>] (ret_fast_syscall+0x0/0x28)
[13958.572681] Exception stack(0xd3a15fa8 to 0xd3a15ff0)
[13958.572694] 5fa0:                   b151a530 b5d72000 00000024 40045613 0130faa8 b6bc5510
[13958.572707] 5fc0: b151a530 b5d72000 b5d727d0 00000036 b1515020 00000001 011fe550 013101c4
[13958.572717] 5fe0: b6dd9c60 bee8ae3c b5d3a954 b6bc551c
[13958.572728] ---[ end trace 54b296e9f80b3042 ]---
[13958.572742] videobuf2_common: driver bug: stop_streaming operation is leaving buf 12b8ea04 in active state
[13958.572757] videobuf2_common: driver bug: stop_streaming operation is leaving buf ea367892 in active state
[13958.572769] videobuf2_common: driver bug: stop_streaming operation is leaving buf 4b0a2ea3 in active state
[13958.572782] videobuf2_common: driver bug: stop_streaming operation is leaving buf 21a87d63 in active state
[13958.572795] videobuf2_common: driver bug: stop_streaming operation is leaving buf 21137ac9 in active state
[13958.572808] videobuf2_common: driver bug: stop_streaming operation is leaving buf 161c98a9 in active state
[13958.572820] videobuf2_common: driver bug: stop_streaming operation is leaving buf 3047fb3b in active state
[13958.572833] videobuf2_common: driver bug: stop_streaming operation is leaving buf c5c17d44 in active state
[13958.572846] videobuf2_common: driver bug: stop_streaming operation is leaving buf 3160d087 in active state
[13958.572859] videobuf2_common: driver bug: stop_streaming operation is leaving buf 316cfd22 in active state
[13958.572871] videobuf2_common: driver bug: stop_streaming operation is leaving buf df235343 in active state
[13958.572884] videobuf2_common: driver bug: stop_streaming operation is leaving buf 55e9c981 in active state
[13958.572896] videobuf2_common: driver bug: stop_streaming operation is leaving buf 621fc030 in active state
[13958.572909] videobuf2_common: driver bug: stop_streaming operation is leaving buf fb56d9cc in active state
[13958.572922] videobuf2_common: driver bug: stop_streaming operation is leaving buf d245f944 in active state
[13958.572934] videobuf2_common: driver bug: stop_streaming operation is leaving buf 408747d9 in active state
[13958.572947] videobuf2_common: driver bug: stop_streaming operation is leaving buf eaf71e16 in active state
```

ndufresne, 2020-04-20:
Could it be that this driver should clear V4L2_BUF_CAP_SUPPORTS_ORPHANED_BUFS on it's VB2 Queue ?

6by9, 2020-04-21:
4.19.97 may be the latest kernel released through apt but isn't the latest here (4.19.115).
Most of our effort is now on the rpi-5.4.y branch to bring 5.4 up. ```BRANCH=next rpi-update``` would give you that.

Yes we do support orphaned buffers, at least with 5.4. Sorry, not going back to 4.19 to check, but I seem to remember backporting it.
Orphaned buffers are a feature of videobuf2 rather than the driver. It allows userspace to retain buffers when the driver has finished with them.

You've provided the snippet of log from when it goes wrong, but why has the kernel been told to stop_streaming if you're still mid stream? What is this RTSP stream that you are transcoding?
Please provide full kernel traces, preferably with the ```debug``` module parameter on bcm2835_codec set to 5.

mjasny, 2020-04-21:
I updated the RPi to the next-branch: `Linux raspberrypi 5.4.32-v7l+ #1308 SMP Fri Apr 17 20:10:43 BST 2020 armv7l GNU/Linux` and also tested with both gstreamer 1.14.4 (from official repositories) and gstreamer 1.17.0 (from gst-master).

The RTSP stream is a 1920x1080p / 25fps 4500KB CBR stream from a surveillance camera. I didn't give the kernel stop_stream, I just ctrl+c after gstreamer showed this buffer-error.

I did the `modprobe -r bcm2835_codec` and `modprobe bcm2835_codec debug=5`. After a fresh restart it seems to run longer but still runs into the issue.

FYI: The kernel trace has everything including ctrl+c of the gstreamer process.
Full dmesg is attached with debug=5 for bcm2835_codec, got pretty big (37 MiB unpacked), I hope it is still usable...
[dmesg.log.gz](https://github.com/raspberrypi/linux/files/4509237/dmesg.log.gz)

6by9, 2020-04-21:
I need to look at the logging syntax as it isn't identified which instance of bcm2835-codec is being used - decoder, encoder, or conversion. Context gives us enough information in this instance.

For whatever reason we are being asked to stop streaming via 
```[  706.091985] bcm2835-codec bcm2835-codec: vidioc_decoder_cmd, cmd 1```
cmd 1 = V4L2_DEC_CMD_STOP, as per https://linuxtv.org/downloads/v4l-dvb-apis-new/userspace-api/v4l/vidioc-decoder-cmd.html

The decoder is therefore flushing, and sends an EOS event once all the CAPTURE frames have been generated.
```
[  706.157688] bcm2835-codec bcm2835-codec: op_buffer_cb: length 0, flags 1, idx 8
[  706.157701] bcm2835-codec bcm2835-codec: op_buffer_cb: Empty buffer - flags 0001
[  706.157706] bcm2835-codec bcm2835-codec: Sending EOS event
[  706.157728] bcm2835-codec bcm2835-codec: op_buffer_cb: done 13399 output buffers
```
Stop streaming on the OUTPUT_MPLANE (10) queue completes
```
[  706.158489] bcm2835-codec bcm2835-codec: bcm2835_codec_stop_streaming: type: 10 - return buffers
[  706.160860] bcm2835-codec bcm2835-codec: bcm2835_codec_stop_streaming: done
```
But the CAPTURE_MPLANE (9) queue has retained some buffers.
```
[  706.160988] bcm2835-codec bcm2835-codec: bcm2835_codec_stop_streaming: type: 9 - return buffers
```
I have seen under some circumstances that we aren't returning all the buffers from MMAL, but videobuf2 cleans up on our behalf and all should be good. I need to investigate the whys and hows, but I don't see this as being fatal.

We then see what is presumably the encoder being stopped without issue
```
[  769.742637] bcm2835-codec bcm2835-codec: bcm2835_codec_stop_streaming: type: 10 - return buffers
[  769.744845] bcm2835-codec bcm2835-codec: bcm2835_codec_stop_streaming: done
[  769.745057] bcm2835-codec bcm2835-codec: bcm2835_codec_buffer_cleanup: ctx:6782917e, vb 88622ccc
[  769.745064] bcm2835-codec bcm2835-codec: bcm2835_codec_buffer_cleanup: ctx:6782917e, vb 7c16c7ab
[  769.745191] bcm2835-codec bcm2835-codec: bcm2835_codec_stop_streaming: type: 9 - return buffers
...
[  769.745517] bcm2835-codec bcm2835-codec: bcm2835_codec_stop_streaming: done
```

So I think at 769.748889 we should be back to all disabled.

We then get things starting up again on CAPTURE_MPLANE (buffer size 3133440, so presumably 1080p decode), but almost immediately cleaning up.
```
[  769.748889] bcm2835-codec bcm2835-codec: bcm2835_codec_buf_queue: type: 9 ptr dcfccf41 vbuf->flags 0, seq 0, bytesused 3133440
[  769.748898] bcm2835-codec bcm2835-codec: bcm2835_codec_buf_queue: type: 9 ptr 3a1ae8a0 vbuf->flags 0, seq 0, bytesused 3133440
[  769.748906] bcm2835-codec bcm2835-codec: bcm2835_codec_buf_queue: type: 9 ptr 4f29c369 vbuf->flags 0, seq 0, bytesused 3133440
[  769.748913] bcm2835-codec bcm2835-codec: bcm2835_codec_buf_queue: type: 9 ptr ca256223 vbuf->flags 0, seq 0, bytesused 3133440
[  769.748920] bcm2835-codec bcm2835-codec: bcm2835_codec_buf_queue: type: 9 ptr eddd4257 vbuf->flags 0, seq 0, bytesused 3133440
[  769.748927] bcm2835-codec bcm2835-codec: bcm2835_codec_buf_queue: type: 9 ptr a21c2118 vbuf->flags 0, seq 0, bytesused 3133440
[  769.748934] bcm2835-codec bcm2835-codec: bcm2835_codec_buf_queue: type: 9 ptr da2d3630 vbuf->flags 0, seq 0, bytesused 3133440
[  769.748941] bcm2835-codec bcm2835-codec: bcm2835_codec_buf_queue: type: 9 ptr 720597f8 vbuf->flags 0, seq 0, bytesused 3133440
[  769.748948] bcm2835-codec bcm2835-codec: bcm2835_codec_buf_queue: type: 9 ptr 11212198 vbuf->flags 0, seq 0, bytesused 3133440

[  769.748956] bcm2835-codec bcm2835-codec: bcm2835_codec_start_streaming: type: 9 count 9

[  769.749284] bcm2835-codec bcm2835-codec: bcm2835_codec_buffer_cleanup: ctx:0e86de0b, vb 4820bc5e
[  769.749293] bcm2835-codec bcm2835-codec: bcm2835_codec_buffer_cleanup: ctx:0e86de0b, vb 3bb0c541
[  769.751722] bcm2835-codec bcm2835-codec: bcm2835_codec_stop_streaming: type: 9 - return buffers
```
I've got no errors logged there, so can't tell what or why things have given up. start_streaming would have logged an error if it failed in any way.

@ndufresne Any suggestions as to why the pipeline may be spontaneously draining? Reconfiguring for some reason? Without having this RTSP stream it's going to be partially guesswork here.

mjasny, 2020-04-21:
To clarify a bit, I reran the pipeline and watched the timestamps. The cleanup after timestamp `[  769.742637]` and all actions afterwards happened after the termination of gstreamer (by me). After the gstreamer pipeline fails with ERROR messages, no further output in the kernel trace is generated until Ctrl+C. Thus the gap of ~60 seconds.

If it would help, I could try to give you private access to the rtsp stream @6by9 .

ndufresne, 2020-04-21:
@mjasny can you attach a gst from env GST_DEBUG="*v4l*:7", I'll need that to explain why this happens. It looks like a renegotiation, we use to have drain() triggered on discount (packet lost), but this is fixed in 1.16. Additional information can be optain from the video node trace and vb2 traces.

mjasny, 2020-04-21:
@ndufresne Here you go: [gst.log.gz](https://github.com/raspberrypi/linux/files/4509913/gst.log.gz)

I setup access through the internet for the rtsp stream for testing. When I access it through the internet instead of local network I noticed that the error occurs earlier, already after a few seconds running.

6by9, 2020-04-21:
Ah, I hadn't noticed the gap in the timestamps.

So 706-ish we're shutting down the decoder due to V4L2_DEC_CMD_STOP requested by GStreamer. Yes it throws a warning, but should have completed everything.
It is a little odd that there is a 3 second gap with no output buffers returned before the EOS.
```
[  703.974258] bcm2835-codec bcm2835-codec: device_run: Submitted src 6dcbde7e, dst 3a1ae8a0
[  706.091985] bcm2835-codec bcm2835-codec: vidioc_decoder_cmd, cmd 1
```
We've had a MMAL output / V4L2 CAPTURE buffer returned at 703.973995, so it wouldn't appear to be that the codec itself has stalled. I'd need to a bundle more analysis to ensure that it wasn't the encoder that has stalled somewhere.

Don't worry about providing access to the stream at present - the first question is why GStreamer is stopping the stream, and why isn't it happy that everything has stopped.

Is this the standard GStreamer install from the Raspbian repos? Those are still 1.14 as they're the same as Debian Buster. If there's a fix in 1.16 then that would be worth trying.

mjasny, 2020-04-21:
I tested both gstreamer from raspbian repository (1.14.4) and compiled gstreamer-master branch (1.17.0) which runs in a separated environment. The recent gst.log is from the gst-1.17.0.

ndufresne, 2020-04-21:
This looks like a 1.14 log, what I see is that it runs for 2s, a packet is dropped (decude from Drain being called). But draining a V4L2 Statefull codec causes stalls. We no longer do that since 1.16, so it's surprising you get the same result with 1.17.  Adding videodecoder:7 category could help, maybe rtph264depay:7 to get some info about the packets/network.

Can you do a simple test like this:

```
gst-launch-1.0 videotestsrc ! v4l2h264enc ! h264parse ! identity drop-probability=0.01 ! v4l2h264dec ! ...
```
This will simulate packet lost without the network bits.

For completeness, the unrelated warning are:
 - VIDIOC_CROPCAP return EINVAL, EINVAL should only be used for userspace error, I doubt Gst is miss-using this API, so it's likey suppose to be ENOTTY (aka not implement, in which case gst stay quite)
- GStreamer does not know the framerate, so it cannot guess the latency (not a big deal and no driver issues there)
- Driver didn't update the v4l2_buffer.feild to something else then NONE, driver bug, but not a big deal
- Encoder warn when enabling copy on threshold, this is gst debug trace bug, sorry
- Encoder is producing more bitstream buffer then stream, on that, the spec is not clear, use h264parse afterward, it will try and fix it, timestamps might be a little off

mjasny, 2020-04-21:
This is my environment for the 1.17.0 version:

```
[gst-master] pi@raspberrypi:~/gst-build $ gst-launch-1.0 --version
gst-launch-1.0 version 1.17.0
GStreamer 1.17.0 (GIT)
Unknown package origin
```
I ran this command for both 1.14 and 1.17 gstreamer:

`GST_DEBUG_NO_COLOR=1 GST_DEBUG="*v4l*:7,*videodecoder*:7,*rtph264depay*:7" gst-launch-1.0 videotestsrc ! v4l2h264enc ! h264parse ! identity drop-probability=0.01 ! v4l2h264dec ! fakesink |& tee gst.1.14.log`

Both runs terminated (with Error) Here are the logs:

[dmesg.1.14.log](https://github.com/raspberrypi/linux/files/4510869/dmesg.1.14.log)
[dmesg.1.17.log](https://github.com/raspberrypi/linux/files/4510871/dmesg.1.17.log)
[gst.1.14.log](https://github.com/raspberrypi/linux/files/4510872/gst.1.14.log)
[gst.1.17.log](https://github.com/raspberrypi/linux/files/4510873/gst.1.17.log)

6by9, 2020-04-21:
Thanks @ndufresne for the analysis.

VIDIOC_CROPCAP. A vaguely recall looking at this one, and I wasn't clear on what the correct behaviour was.
The driver doesn't implement ```vidioc_g_pixelaspect``` as the data isn't immediately to hand. That's of little consequence as it just skips that and leaves pixelaspect as 1/1.
 ```v4l_cropcap``` then calls g_selection with ```V4L2_SEL_TGT_COMPOSE_BOUNDS``` and ```V4L2_SEL_TGT_COMPOSE_DEFAULT``` on an OUTPUT or OUTPUT_MPLANE queue, and ```V4L2_SEL_TGT_CROP_BOUNDS``` and ```V4L2_SEL_TGT_CROP_DEFAULT``` on a CAPTURE or CAPTURE_MPLANE queue.

We support g_selection on _CAPTURE queues for decode, returning EINVAL if any other selection is requested. We then support V4L2_SEL_TGT_COMPOSE_DEFAULT, V4L2_SEL_TGT_COMPOSE, and V4L2_SEL_TGT_COMPOSE_BOUNDS, which was copying mtk_vcodec_dec. 
So querying g_selection for queue VIDEO_CAPTURE on a decoder v4l_cropcap will request V4L2_SEL_TGT_CROP_BOUNDS and V4L2_SEL_TGT_CROP_DEFAULT, neither of which are considered to be valid, and result in -EINVAL. 
I seem to recall checking the other mainline codec drivers, and none of them seemed to reply with answers that v4l_cropcap will be happy with, so I was a little bemused as to whether CROPCAP worked against any mainline decoder.

v4l2_buffer.field - yup, ought to sort that. Ideally I need to support interlaced decoding, but the field order info isn't currently easily exposed through MMAL. The hardware spits out the equivalent of either ```V4L2_FIELD_INTERLACED_TB``` or ```V4L2_FIELD_INTERLACED_BT```, but I don't know the detail of whether it is always one or the other.

Bitstream buffers vs pictures - I brought this up at ELCE Media Summit that the spec isn't clear as to where header bytes fitted into the stream and buffers. I'd forgotten that it had been clarified that there is an expectation that it would be in the buffer with the first encoded frame, and haven't fixed that yet. Thanks for the reminder.

ndufresne, 2020-04-21:
@6by9 I think CROPCAP called by Gst is caused by shared code, CROPCAP is used for capture to determine the pixel aspect ratio (which isn't exposed in composition iirc). I don't think it make sense for CODEC as we usually read this from the bitstream, but someone would need to do large re-factoring or ugly hack to remove this call. I'd say let's ignore. Thanks for the update on the rest, indeed, I think the output of the encoder should be 1 AU (frame, field or layer).

@mjasny sorry, this seems non-conclusive as streaming failed for other reason.

mjasny, 2020-04-22:
@ndufresne I'm sorry I can't get the videotestsrc command to run under gst 1.1.17 (from master). I did three complete rebuilds and reinstallation. My Raspberry Pi has 512MB GPU memory and over 1.4GB free CPU memory so I doubt it's a memory issue:

```
Setting pipeline to PAUSED ...
Pipeline is PREROLLING ...
Redistribute latency...
0:00:00.125983492  3858  0x160f580 ERROR         v4l2bufferpool gstv4l2bufferpool.c:679:gst_v4l2_buffer_pool_streamon:<v4l2h264enc0:pool0:sink> error with STREAMON 3 (No such process)                                                
0:00:00.126055159  3858  0x160f580 ERROR         v4l2bufferpool gstv4l2bufferpool.c:2099:gst_v4l2_buffer_pool_process:<v4l2h264enc0:pool0:sink> failed to start streaming
ERROR: from element /GstPipeline:pipeline0/v4l2h264enc:v4l2h264enc0: Failed to process frame.                      
Additional debug info:
../subprojects/gst-plugins-good/sys/v4l2/gstv4l2videoenc.c(817): gst_v4l2_video_enc_handle_frame (): /GstPipeline:pipeline0/v4l2h264enc:v4l2h264enc0:
Maybe be due to not enough memory or failing driver
ERROR: pipeline doesn't want to preroll.
Setting pipeline to NULL ...
ERROR: from element /GstPipeline:pipeline0/GstVideoTestSrc:videotestsrc0: Internal data stream error.              
Additional debug info:
../subprojects/gstreamer/libs/gst/base/gstbasesrc.c(3075): gst_base_src_loop (): /GstPipeline:pipeline0/GstVideoTestSrc:videotestsrc0:
streaming stopped, reason error (-5)
ERROR: pipeline doesn't want to preroll.
Freeing pipeline ...
```
Here are the full logs:
[dmesg.1.17.log](https://github.com/raspberrypi/linux/files/4514084/dmesg.1.17.log)
[gst.1.17.log](https://github.com/raspberrypi/linux/files/4514085/gst.1.17.log)

---

Instead I ran it on a fresh gst-1.16 which fails with: `gst_video_frame_map_id: assertion 'GST_IS_BUFFER (buffer)' failed`:

`gst-launch-1.0 videotestsrc ! v4l2h264enc ! h264parse ! identity drop-probability=0.01 ! v4l2h264dec ! fakesink`

```
Setting pipeline to PAUSED ...
Pipeline is PREROLLING ...
Redistribute latency...
Pipeline is PREROLLED ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock

** (gst-launch-1.0:4515): CRITICAL **: 07:19:09.780: gst_video_frame_map_id: assertion 'GST_IS_BUFFER (buffer)' fail
ed

(gst-launch-1.0:4515): GStreamer-CRITICAL **: 07:19:09.781: gst_pad_push: assertion 'GST_IS_BUFFER (buffer)' failed
ERROR: from element /GstPipeline:pipeline0/GstVideoTestSrc:videotestsrc0: Internal data stream error.
Additional debug info:
../subprojects/gstreamer/libs/gst/base/gstbasesrc.c(3072): gst_base_src_loop (): /GstPipeline:pipeline0/GstVideoTest
Src:videotestsrc0:
streaming stopped, reason error (-5)
Execution ended after 0:03:10.133883208
Setting pipeline to PAUSED ...
Setting pipeline to READY ...

(gst-launch-1.0:4515): GStreamer-CRITICAL **: 07:19:09.792: gst_mini_object_unref: assertion 'GST_MINI_OBJECT_REFCOU
NT_VALUE (mini_object) > 0' failed
Setting pipeline to NULL ...
Freeing pipeline ...
```

[dmesg.1.16.log](https://github.com/raspberrypi/linux/files/4514112/dmesg.1.16.log)
[gst.1.16.log](https://github.com/raspberrypi/linux/files/4514113/gst.1.16.log)

**If these logs are still not sufficient please come back to me here or on IRC.**

mjasny, 2020-05-03:
It turned out that some older gstreamer libraries got mixed into the environment and were causing this issue. Identifying and replacing them fixed the crashes. Interestingly I did not see any version complains in the logs. My program is now running for a few days without any crashes.
Thanks for your time and help on this issue!

closed 2020-05-03 by mjasny

miguelwon, 2020-05-18:
How did you solve the problem? I'm having the same error and I'm using updated GStreamer libraries (an updated gst-build):

```
0:00:01.090893246  1872  0x1541520 ERROR         v4l2bufferpool gstv4l2bufferpool.c:679:gst_v4l2_buffer_pool_streamon:<v4l2h264enc0:pool0:sink> error with STREAMON 3 (No such process)
0:00:01.091070078  1872  0x1541520 ERROR         v4l2bufferpool gstv4l2bufferpool.c:2099:gst_v4l2_buffer_pool_process:<v4l2h264enc0:pool0:sink> failed to start streaming
```

mjasny, 2020-05-18:
I used the 1.16 tag from github and made sure that I use a fresh and clean raspbian image (without any gstreamer installation from the repos). You'll get while building some dependencies which are missing. There is a nice script which prepares your 1.16 worktree. The STREAMON although is still present in the master code, but I didn't investigate more because all is working fine now in the stable 1.16 version.

barneyman, 2020-09-19:
just a comment to help others find this

ges-launch-1,0 v4l2h264dec, hanging, pause, filesink async

i had problems with a gstream app i wrote that used ges, that i could replicate with 

`ges-launch-1.0 +clip ./rendered.mkv duration=25.0 i=1.0 --outputuri=file:///home/pi/source/out.orb`

updating the kernel with `BRANCH=next rpi-update` *and* doing a clean build of 1.16.2 resolved my issues; i still have a segfault in tmp-orc, in my code, that doesn't happen with ges-launch, so i'm pretty sure that's a 'my code' problem

rma153, 2021-02-27:
**mjasny**, I'm seeing this same error blow. Can you please specify the steps you took to solve it, i.e., list which "older gstreamer libraries got mixed into the environment and were causing this issue. Identifying and replacing them fixed the crashes."

ERROR: from element /GstPipeline:pipeline0/v4l2h264enc:v4l2h264enc0: Failed to process frame.                      
Additional debug info:
../subprojects/gst-plugins-good/sys/v4l2/gstv4l2videoenc.c(817): gst_v4l2_video_enc_handle_frame (): /GstPipeline:pipeline0/v4l2h264enc:v4l2h264enc0:
Maybe be due to not enough memory or failing driver

rma153, 2021-03-01:
I gather then that the solution is to update the kernel with BRANCH=next rpi-update and do a clean build of 1.16.2

pelwell, 2021-03-02:
BRANCH=next is not being updated at that moment - use the master branch (the default), i.e. just `sudo rpi-update`.

mjasny, 2021-03-02:
> I used the 1.16 tag from github and made sure that I use a fresh and clean raspbian image (without any gstreamer installation from the repos). You'll get while building some dependencies which are [...]

As I wrote above, I started with a clean Raspbian-Lite image and installed gstreamer 1.16 directly from the official git-repository (using meson, ninja etc...). Worked without problems after that.